dingo35/SmartEVSE-3.5 #112: Question / Feature Request - Solar C2 Power Share balancing

vdweegen, 2024-07-30:
Currently this issue can be treated as either a question or a feature request. I currently have a Dual SmartEVSE Setup and have power share configured.

My MAINS is 25A
MAX CIRCUIT is 16A
Mains Meter is a Sensor Box

**Current Setup**

```
                       N    L1   L2   L3                  
                       |    |    |    |
                   -------------------------
                   |      16A Breaker      |
                   -------------------------
                       |    |    |    |
    -----------------------------------------------------------------
    |                       Distribution Box                        |
    -----------------------------------------------------------------
        |    |    |    |                   |    |    |    |
        N   L1   L2   L3                   N   L1   L2   L3
        |    |    |    |                   |    |    |    |
---------------   |    |           ---------------   |    |
| SmartEVSE1   |  |    |           | SmartEVSE2  |   |    |
---------------   |    |           ---------------   |    |
        |    |    |    |                   |    |    |    |
      --------------------               --------------------
      |     Eastron3P    |               |     Eastron3P    |
      --------------------               --------------------
        |    |    |    |                   |    |    |    |
      --------------------               --------------------
      | 4-p contactor C1 |               | 4-p contactor C1 |
      --------------------               --------------------
        |    |    |    |                   |    |        
        |    |    |    |                   |    |
      --------------------               --------------------
      |    EV-cable      |               |    EV-cable      |
      --------------------               --------------------                  
```
At the moment I'm using SMART-mode with Power Share (SmartEVSE1 is Master, SmartEVSE2 is Slave1) which is working perfectly.

Both SmartEVSE's have a fixed cable installed. SmartEVSE1 has a type2 Cable (Citroen eC4), SmartEVSE2 has a type1 cable (Mitsubishi Outlander PHEV)

I'm planning on using Solar Mode to leverage the surplus energy generated my 8.2KwP solar panel setup, which means adding a C2 contactor to SmartEVSE1 

**Plannet Setup**
```
                       N    L1   L2   L3                  
                       |    |    |    |
                   -------------------------
                   |      16A Breaker      |
                   -------------------------
                       |    |    |    |
    -----------------------------------------------------------------
    |                       Distribution Box                        |
    -----------------------------------------------------------------
        |    |    |    |                   |    |    |    |
        N   L1   L2   L3                   N   L1   L2   L3
        |    |    |    |                   |    |    |    |
---------------   |    |           ---------------   |    |
| SmartEVSE1   |  |    |           | SmartEVSE2  |   |    |
---------------   |    |           ---------------   |    |
        |    |    |    |                   |    |    |    |
      --------------------               --------------------
      |     Eastron3P    |               |     Eastron3P    |
      --------------------               --------------------
        |    |    |    |                   |    |    |    |
      --------------------               --------------------
      | 4-p contactor C1 |               | 4-p contactor C1 |
      --------------------               --------------------
        |    |    |    |                   |    |
        |    | ------------------          |    |
        |    | |2-p contactor C2|          |    |              
        |    | ------------------          |    |          
        |    |    |    |                   |    |
      --------------------               --------------------
      |    EV-cable      |               |    EV-cable      |
      --------------------               --------------------                  
```

**Questions**
1. When activating Solar Mode, would using <Solar Off> for C2 mean that both SmartEVSE's could share the available capacity 6A-16A if available (if enough surplus is available)?
2. Would Slaves be aware of C2 settings from the Master Node, or would settings be replicated?
3. Would SmartEVSE2 behave as <Always Off> for C2?
4. Would adding a C2 contactor on SmartEVSE2 in the future be possible (lets say I get a type2 fixed cable in the future)?
5. Would setting <Solar Off> for C2 on SmartEVSE2 then be possible?

**Device Configs (current situation)**
```
SmartEVSE1

{
  "version": "v3.6.0-RC2",
  "serialnr": 6678,
  "mode": "SMART",
  "mode_id": 3,
  "car_connected": false,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "XXXXXXXXXX",
    "rssi": -54,
    "bssid": "XX:XX:XX:XX:XX:XX"
  },
  "evse": {
    "temp": 43,
    "temp_max": 65,
    "connected": false,
    "access": true,
    "mode": 1,
    "loadbl": 1,
    "pwm": 1024,
    "solar_stop_timer": 0,
    "state": "Ready to Charge",
    "state_id": 0,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 160,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 25,
    "current_max_circuit": 16,
    "current_max_sum_mains": 600,
    "solar_max_import": 2,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Not present",
    "modem": "Not present",
    "mains_meter": "Sensorbox",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0
  },
  "mqtt": {
    "host": "10.12.40.32",
    "port": 1883,
    "topic_prefix": "SmartEVSE-15494",
    "username": "",
    "password_set": false,
    "status": "Connected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Eastron3P",
    "address": 101,
    "import_active_power": 0,
    "total_kwh": 1449.800049,
    "charged_kwh": 3.400000095,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 1449.800049,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 196,
    "L1": 178,
    "L2": 3,
    "L3": 15,
    "last_data_update": 1722362781,
    "original_data": {
      "TOTAL": 196,
      "L1": 178,
      "L2": 3,
      "L3": 15
    }
  },
  "backlight": {
    "timer": 0,
    "status": "OFF"
  }
}
```

```
SmartEVSE2

{
  "version": "v3.6.0-RC2",
  "serialnr": 7130,
  "mode": "SMART",
  "mode_id": 3,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "XXXXXXXXXX",
    "rssi": -73,
    "bssid": "XX:XX:XX:XX:XX:XX"
  },
  "evse": {
    "temp": 50,
    "temp_max": 65,
    "connected": true,
    "access": true,
    "mode": 1,
    "loadbl": 2,
    "pwm": 221,
    "solar_stop_timer": 0,
    "state": "Charging",
    "state_id": 2,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 130,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 25,
    "current_max_circuit": 16,
    "current_max_sum_mains": 600,
    "solar_max_import": 2,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Not present",
    "modem": "Not present",
    "mains_meter": "Sensorbox",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0
  },
  "mqtt": {
    "host": "10.12.40.32",
    "port": 1883,
    "topic_prefix": "SmartEVSE-46809",
    "username": "",
    "password_set": false,
    "status": "Connected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Eastron3P",
    "address": 110,
    "import_active_power": 0.600000024,
    "total_kwh": 514.5999756,
    "charged_kwh": 2.299999952,
    "currents": {
      "TOTAL": 26,
      "L1": 26,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 514.5999756,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 89,
    "L1": 71,
    "L2": 4,
    "L3": 14,
    "last_data_update": 1722364203,
    "original_data": {
      "TOTAL": 89,
      "L1": 71,
      "L2": 4,
      "L3": 14
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  }
}
```

vdweegen, 2024-07-30:
Just updated both SmartEVSE units to v.3.6.6

dingo36, 2024-07-31:
1. Yes, you would set the Solar settings (StartCurrent, ImportCurrent) on the Master and if enough power is available, it would be shared with the slave. However, current version v3.6.6 would switch off BOTH SmartEVSEs if there is not enough current to feed both of them with the minimum current of 6A.
I have had a fix for this for a long time but nobody seems to want to test this fix. So if you are a volunteer?
2. Currently, you can not enable/disable C2 on the slave devices. Of course you can hardwire them to 1 phase or 3 phase. I will be happy to enable software settings for C2 on the slave if you would test it, although Im pretty sure the Auto setting would not work; but the Always_On, Always_Off, Solar_Off would....
3. I don't understand the question. Slave would charge as expected, only at single phase, so some calculations will be suboptimal since the Master will assume possible currents through phases 2 and 3. But this could easily be avoided if I enabling C2 settings on the slave, and you set it to Always_Off; that way the Master is aware you will be single phase charging on the Slave.
4.  Yes but see 2.
5.  See 2.

As you can probably sense, AFAIK you are on uncharted territories here; most Master/Slave users use it in Smart Mode, or, if in Solar, switch on the Master first, and later on the Slave.
But I 'm happy to work with you to see if we can get this config going?

vdweegen, 2024-07-31:
> 1. Yes, you would set the Solar settings (StartCurrent, ImportCurrent) on the Master and if enough power is available, it would be shared with the slave. However, current version v3.6.6 would switch [...]

Yes, I would be more than willing to test. Will be receiving my C2 contactor in roughly two weeks (after mstegen's holiday).
> 2. Currently, you can not enable/disable C2 on the slave devices. Of course you can hardwire them to 1 phase or 3 phase. I will be happy to enable software settings for C2 on the slave if you would [...]

I was thinking about this. It would theoretically be possible using my setup to charge both cars over a single phase (so 2x16A=32A) in solar mode using the following setup (not the type1 fixed cable is now wired to L3):

```
                       N    L1   L2   L3                  
                       |    |    |    |
                   -------------------------
                   |      16A Breaker      |
                   -------------------------
                       |    |    |    |
    -----------------------------------------------------------------
    |                       Distribution Box                        |
    -----------------------------------------------------------------
        |    |    |    |                   |    |    |    |
        N   L1   L2   L3                   N   L1   L2   L3
        |    |    |    |                   |    |    |    |
---------------   |    |           ---------------   |    |
| SmartEVSE1   |  |    |           | SmartEVSE2  |   |    |
---------------   |    |           ---------------   |    |
        |    |    |    |                   |    |    |    |
      --------------------               --------------------
      |     Eastron3P    |               |     Eastron3P    |
      --------------------               --------------------
        |    |    |    |                   |    |    |    |
      --------------------               --------------------
      | 4-p contactor C1 |               | 4-p contactor C1 |
      --------------------               --------------------
        |    |    |    |                   |              |
        |    | ------------------          |              |
        |    | |2-p contactor C2|          |              |              
        |    | ------------------          |              |          
        |    |    |    |                   |              |
      --------------------               --------------------
      |    EV-cable      |               |    EV-cable      |
      --------------------               --------------------                  
```
If SmartEVSE1 toggles C2 to Off (SmartEVSE2 would be set to <Always Off>) it would be possible to charge both cars at 16A without tripping the 16A breakers (assuming enough energy would be available). If would, however, require settings some value to let the SmartEVSE's know which phase is the 'main/active phase' when C2 is switched off. This would also be possible using a C2 contactor on SmartEVSE2 to disconnect two phases, which should work as long as the active phase as seen from the mains is wired to L1 on the fixed cable.
> 3. I don't understand the question. Slave would charge as expected, only at single phase, so some calculations will be suboptimal since the Master will assume possible currents through phases 2 and [...]

Exactly, would be interested in testing setting the C2 setting to <Always Off> on the slave node.
> 4. Yes but see 2.
> 5. See 2.
> 
> As you can probably sense, AFAIK you are on uncharted territories here; most Master/Slave users use it in Smart Mode, or, if in Solar, switch on the Master first, and later on the Slave. But I 'm happy to work with you to see if we can get this config going?

It will be interesting to see if we can get this config to work. I could test setting the C2 contactor to <Always off> right away. When I receive the contactor for SmartEVSE1 I can do further testing on the completed setup.

vdweegen, 2024-07-31:
> "If SmartEVSE1 toggles C2 to Off (SmartEVSE2 would be set to ) it would be possible to charge both cars at 16A without tripping the 16A breakers (assuming enough energy would be available). If would, however, require settings some value to let the SmartEVSE's know which phase is the 'main/active phase' when C2 is switched off. This would also be possible using a C2 contactor on SmartEVSE2 to disconnect two phases, which should work as long as the active phase as seen from the mains is wired to L1 on the fixed cable."
> 
> I understand your reasoning, but I am not keen on implementing such settings. The problem is that you are now configuring your EVSE to two specific EV's that are connected to their specific SmartEVSE's, while in practice, any EV could be connected at any EVSE at any time. SmartEVSE is designed to be a normal EVSE that can be connected to any EV, so lets keep it that way :-) Of course you can always add your own quirks to the software, that is why it is open source!

I wholeheartedly agree with this. I think you misunderstand my intention here. I'm proposing configuring the setup such that when in solar mode the C2 contactors on both the SmartEVSEs would  configure the setup similarly as having balancing two single phase SmartEVSEs (where one would be wired to L1, and one would be wired to L2/L3).

> 
> Let me know when you are ready for testing, the fix mentioned in 1. is called the "calcbalancedcurrent" fix, which is the function that is mainly touched.

Would be able to test this right away. Without the contactor on SmartEVSE1 this unit (master) would always charge 3phase, Node1 (SmartEVSE2) would charge on L1 (see if C2 contactor to <Always Off> works).

dingo35, 2024-07-31:
From the manual page:

"Note that it is important that you actually DO NOT switch the L1 pin of the CCS plug with the C2 contactor; some cars (e.g. Tesla Model 3) will go into error;
  they expect the charging phase to be on the L1 pin when single-phase charging..."

dingo35, 2024-07-31:
Also I reserved the right to assume that C2 is always connected to L2 and L3, when doing calculations in Solar mode :-)

vdweegen, 2024-07-31:
> From the manual page:
> 
> "Note that it is important that you actually DO NOT switch the L1 pin of the CCS plug with the C2 contactor; some cars (e.g. Tesla Model 3) will go into error; they expect the charging phase to be on the L1 pin when single-phase charging..."

Yes, I am aware. This could be mitigated/solved by rotating the wiring (in such a way that the phase sine would still be intact). For example:

```
SmartEVSE1    Type 2 Fixed Cable
L1 ---------------- L1
L2 ---------------- L2 (wired through C2 contactor)
L3 ---------------- L3 (wired through C2 contactor)

SmartEVSE2    Type 2 Fixed Cable
L1 ---------------- L2 (wired through C2 contactor)
L2 ---------------- L3 (wired through C2 contactor)
L3 ---------------- L1

```

Then both C2 contactor are switched to off, both cars will charge on their respective L1 whilst keeping the phase differences intact when both (or either) C2 contactors are switched on.

Please consider this as a braindump and an attempt to think of new ideas to improve both the software and the possibilities of the solution.

vdweegen, 2024-07-31:
> Also I reserved the right to assume that C2 is always connected to L2 and L3, when doing calculations in Solar mode :-)

Right, fair point. If and when I decide to get a new car forcing me to change the second SmartEVSE to a fixed type2 cable I'll investigate this route myself. For now let's see if we can get the current setup working.

Currently building the v3.6.6 tag with the `calcbalancedcurrent3` branched merged in.

dingo35, 2024-07-31:
Ok, that branch is a few weeks old, there are some known bugs in there.

I can update the branch tomorrow so its uptodate with master branch .....

vdweegen, 2024-07-31:
That might actually be easier, cheers.

dingo35, 2024-08-01:
Test version:
[f7c6869c.zip](https://github.com/user-attachments/files/16452143/f7c6869c.zip)

Previous versions (<= v.3.6.7) would Wait for Power/Solar when it was impossible to deliver MinCurrent (6A) to all charging EVSE's; in this version it disconnects one or more EVSE's until the MinCurrent restraint can be fullfilled.

Prioritization strategy can be set in the LCD menu:

```
PRIORITY (only appears when PWR SHARE set to <Master>)
        Determines the priority when multiple EVSEs are asking for power and there is
        not enough current to provide them all with MinCurrent.
  <NodeNr>      Priority is given to already charging EV's, with highest priority to Master,
                then to Node1, then to Node2 etc.
  <FirstConn>   Priority is given to already charging EV's, with highest priority to the EV
                that was first connected to its EVSE.
                The thought is: "First come, first serve"
  <LastConn>    Priority is given to already charging EV's, with highest priority to the EV
                that was last connected to its EVSE.
                The thought is: the EV that is connected the longest, has already the most
                charge; a charge from 5% -> 15% is more valuable to that EV
                than a charge from 80% -> 90% .

```

vdweegen, 2024-08-01:
Testing now:

```
SmartEVSE1

{
  "version": "09:56:12 @Aug  1 2024",
  "serialnr": 6678,
  "mode": "SOLAR",
  "mode_id": 2,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "XXXXXXXXXX",
    "rssi": -62,
    "bssid": "XX:XX:XX:XX:XX:XX"
  },
  "evse": {
    "temp": 37,
    "temp_max": 65,
    "connected": true,
    "access": true,
    "mode": 2,
    "loadbl": 2,
    "pwm": 102,
    "solar_stop_timer": 541,
    "state": "Charging",
    "state_id": 2,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 60,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 25,
    "current_max_circuit": 16,
    "current_max_sum_mains": 600,
    "max_sum_mains_time": 0,
    "solar_max_import": 2,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Not present",
    "mains_meter": "Sensorbox",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0
  },
  "mqtt": {
    "host": "10.12.40.32",
    "port": 1883,
    "topic_prefix": "SmartEVSE-15494",
    "username": "",
    "password_set": false,
    "status": "Connected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Eastron3P",
    "address": 101,
    "import_active_power": 3.099999905,
    "total_kwh": 1471.199951,
    "charged_kwh": 0.100000001,
    "currents": {
      "TOTAL": 150,
      "L1": 52,
      "L2": 51,
      "L3": 47
    },
    "import_active_energy": 1471.199951,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 78,
    "L1": 36,
    "L2": 24,
    "L3": 18,
    "last_data_update": 1722519782,
    "original_data": {
      "TOTAL": -50,
      "L1": -8,
      "L2": -21,
      "L3": -21
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  }
}

```

```
SmartEVSE2

{
  "version": "09:56:12 @Aug  1 2024",
  "serialnr": 7130,
  "mode": "SOLAR",
  "mode_id": 2,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "XXXXXXXXXX",
    "rssi": -58,
    "bssid": "XX:XX:XX:XX:XX:XX"
  },
  "evse": {
    "temp": 38,
    "temp_max": 65,
    "connected": true,
    "access": true,
    "mode": 2,
    "loadbl": 1,
    "pwm": 1024,
    "solar_stop_timer": 544,
    "state": "Charging Stopped - Waiting for Solar",
    "state_id": 9,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 0,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 25,
    "current_max_circuit": 16,
    "current_max_sum_mains": 600,
    "max_sum_mains_time": 0,
    "solar_max_import": 2,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Not present",
    "mains_meter": "Sensorbox",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0
  },
  "mqtt": {
    "host": "10.12.40.32",
    "port": 1883,
    "topic_prefix": "SmartEVSE-46809",
    "username": "",
    "password_set": false,
    "status": "Connected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Eastron3P",
    "address": 110,
    "import_active_power": 0,
    "total_kwh": 521.7000122,
    "charged_kwh": 0,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 521.7000122,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 80,
    "L1": 37,
    "L2": 24,
    "L3": 19,
    "last_data_update": 1722520048,
    "original_data": {
      "TOTAL": 80,
      "L1": 37,
      "L2": 24,
      "L3": 19
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  }
}
```

vdweegen, 2024-08-19:
I've been testing for the past weeks (without 2nd contactor) and as far as I can tell charging using priority by NodeNr works. 

I just installed the 2nd contactor (Solar Off) so will starting more thoroughly the next few weeks.

A quick observation I made last weekend while charging is Solar mode (will need to verify this further): It appears that when the Node1 unit is charging while the Master unit is disconnected the units don't reset/re-negotiate state when a car is connected to the master Unit. The Node1 unit was charging 6A resulting in a 1A sum on MAINS, when I connected a car to the master unit it also starting charging at 6A drawing a sum of 7A on the MAINS. I would expect the session to stop and power/charging to be re-negatiated.

dingo35, 2024-08-19:
The expected behaviour would be, that Node1 kept on charging and the Master would be going to "Waiting for Solar".

The priorities are set among the charging EV's (state C), not among connected EV's...

dingo35, 2024-08-26:
No, even the manufacturers 3.0.1 nor any other version have actually worked with different modes in Master/Slave config; they may have appeared so on the LCD screen, but it never worked in different modes.

If you really need two different modes, dont use Master/Slave config but the subpanel config, with an EVMeter to guard your MaxCircuit

closed 2024-08-26 by dingo35

vdweegen, 2024-08-26:
> Test version: [f7c6869c.zip](https://github.com/user-attachments/files/16452143/f7c6869c.zip)
>
> Previous versions (<= v.3.6.7) would Wait for Power/Solar when it was impossible to deliver MinCurrent [...]

Just to verify, have these changes been merged to the main branch? If not it would keep me from updating. Thanks

dingo35, 2024-08-26:
They are in the main branch, but not in any of the releases yet.

vdweegen, 2024-10-21:
> They are in the main branch, but not in any of the releases yet.

Testing has yielded good results so far. Any ideas on when these changes can be merged into the releases?

dingo35, 2024-10-21:
I had to take this commit out of the master branch, it is sidelined to calcbalancedcurrent5 branch.

I am very sorry, but there was not enough testing in the community, so I have to consider it unstable. Also there is too much going on (OCPP, new Sensorbox firmware, v4 firmware) to keep this commit updated....

If the dust settles I will re-merge this into the main branch, but it will take a while!