dingo35/SmartEVSE-3.5 #113: Solar countdown timer published on MQTT

Imaginous, 2024-08-09:
**Describe the solution you'd like**
Is it possible to publish the solar countdown timer on MQTT.
I want to use this to lower the changing level of my Tesla just before the timer expires. This way it won't nag me with unnesecary _"Charging stopped unexpectedly"_ messages.

**Describe alternatives you've considered**
I now use my own timer in parallel with an offset, but I would be nice if I could rely on the timer of the SmartEVSE.

**Additional context**
Keep up the good work!

dingo35, 2024-08-10:
Please test:

[39d875b4052f.zip](https://github.com/user-attachments/files/16568817/39d875b4052f.zip)

Please do not only test the new SolarStopTimer topic, but also check the other ones; I changed the calls to MQTT.publish, it should have no effect but you never know !

Imaginous, 2024-08-10:
MQTT is posting correctly. 

Maybe only post the timer when it has changed since the last update. It's now continously updating when it's zero. 

Can't test the timer yet, because the car is fully charged.

Imaginous, 2024-08-13:
There might be a logic problem with the count down timer.
It now is 0 when the timer is inactive. I think it should be the set MAX TIMEOUT (or -1) when it is not counting and the car is charging in solar mode.

Now we (I) can't see the difference between an expired timer and a non triggered timer.

I don't know what you think of it.

dingo35, 2024-08-13:
You can tell the difference by checking solar_stop_time, that tells you if the timer has been set.

To keep the logic simple there is no difference between an expired timer and an unset timer.

Imaginous, 2024-08-20:
After testing I think this feature works fine and can be in the next official release.

closed 2024-08-20 by Imaginous