dingo35/SmartEVSE-3.5 #114: Current blocked at 32A on the smart EVSE and 16A on the car.

legeantbleu, 2024-08-10:
In version 3.6.7 (identical with the previous versions) I regularly have a bug (but I don't know how to reproduce it).

In solar mode the smart EVSE suddenly goes to 32A but the car charges in 16A (on tesla interface it is indicated 16/32A) and the regulation no longer works...
[image]
[image]

We can see the problem regularly on the 32A peaks (I only have 6 kWp of photovoltaics...), on the second screen I switched to the mode of solar > off and then solar.

`{"version":"v3.6.7","serialnr":7340,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"UBNT","rssi":-48,"bssid":"24:5A:4C:63:DE:7C"},"evse":{"temp":51,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":1,"pwm":353,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":207,"override_current":0,"current_min":6,"current_max":32,"current_main":32,"current_max_circuit":32,"current_max_sum_mains":60,"max_sum_mains_time":1,"solar_max_import":6,"solar_start_current":0,"solar_stop_time":15,"enable_C2":"Auto","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.1.28","port":1883,"topic_prefix":"SmartEVSE-7340","username":"mqtt_adm","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":62,"L1":62,"L2":0,"L3":0,"last_data_update":1723280034,"original_data":{"TOTAL":62,"L1":62,"L2":0,"L3":0}},"backlight":{"timer":120,"status":"ON"}} `

dingo35, 2024-08-10:
"In solar mode the smart EVSE suddenly goes to 32A but the car charges in 16A (on tesla interface it is indicated 16/32A) and the regulation no longer works..."

I dont know what it means "tesla interface it is indicated 16/32A" ; how do you know it is actually charging 16A? Is the 32A on the interface means it is charging 2x16A ?

Also I notice you have configured Pwr Share=Master, you have Slaves connected?
If you change to Pwr Share Disabled, still the same peaks on charge_current?

legeantbleu, 2024-08-10:
On the tesla application the first digit indicates the actual load (16 A) and the second the maximum load limit (32 A).

It is strange that the car is limited to 16A (controlled on the power meter) while the limit is 32A.

I activated this option to access the "current maxcircuit" parameter which was limited to 16A, I will try to modify the pwr share and see if it starts again.

Thank you :)

dingo35, 2024-08-10:
If you keep getting the peaks, please flash v3.6.7 debug version and upload the part of the telnet log just before and after the peak....

dingo35, 2024-08-10:
By the way, it is silly to enable the "MaxCircuit" option and expect it to work if you dont have an EVmeter installed.

If you have a single SmartEVSE and you want to limit it to 16A, just set Max to 16A....

legeantbleu, 2024-08-10:
> By the way, it is silly to enable the "MaxCircuit" option and expect it to work if you dont have an EVmeter installed.
> 
> If you have a single SmartEVSE and you want to limit it to 16A, just set Max to 16A....

Yet before doing this I could not exceed 16A in solar mode, I was forced to switch the maxcircuit to 32A by activating the share function.

I didn't have a bug today after disabling the pwr share for the moment.

dingo35, 2024-08-12:
"Yet before doing this I could not exceed 16A in solar mode, I was forced to switch the maxcircuit to 32A by activating the share function."

We might have had a bug in MaxCircuit guarding, but in v3.6.7 there are no known bugs in this.

From the manual:
```
CIRCUIT
        If PWR SHARE set to <Disabled>:
        Only appears when an EV METER is configured, in Smart or Solar mode.
        Set the max current the EVSE circuit can handle (load balancing): 10-200A
        Not obeyed in Normal mode.
        (see also subpanel wiring)

        If PWR SHARE set to <Master>:
        Set the max current the EVSE circuit can handle (load balancing): 10-200A
        Obeyed in all modes!```

So in your configuration, with PwrShare disabled and no EVMeter installed, Circuit will be ignored.

closed 2024-08-22 by dingo35