dingo35/SmartEVSE-3.5 #115: SmartEVSE randomly goes to unavailable in HA when connected via MQTT

hmmbob, 2024-08-12:
**Describe the bug**
As discussed on Tweakers: https://gathering.tweakers.net/forum/list_message/79791476#79791476 and further

I had EMQX do a trace log for the SmartEVSE client (and setup an automation in HA to warn me when the SmartEVSE would go to unavailable), and at 15:45:19 I had a hit. Uploading 2 'cycles' of logging: 15:45:19 and 15:45:22 including 2 lines before and 2 lines after. The issue starts at `2024-08-12T15:45:19.693458+02:00`

```
2024-08-12T15:45:08.603968+02:00 [AUTHZ] SmartEVSE-6360@192.168.88.216:59592 msg: authorization_module_nomatch, action: PUBLISH(Q0,R0), module: emqx_authz_client_info, topic: SmartEVSE-6360/EVTotalEnergyCharged, username: smartevse
2024-08-12T15:45:08.604134+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/EVTotalEnergyCharged, username: smartevse, payload: 3040835
2024-08-12T15:45:19.583082+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL1, PacketId=undefined, Payload=-10), username: smartevse
2024-08-12T15:45:19.583374+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/MainsCurrentL1, username: smartevse, payload: -10
2024-08-12T15:45:19.583491+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL2, PacketId=undefined, Payload=-40), username: smartevse
2024-08-12T15:45:19.583882+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/MainsCurrentL2, username: smartevse, payload: -40
2024-08-12T15:45:19.583967+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL3, PacketId=undefined, Payload=-40), username: smartevse
2024-08-12T15:45:19.584117+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/MainsCurrentL3, username: smartevse, payload: -40
2024-08-12T15:45:19.584221+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVCurrentL1, PacketId=undefined, Payload=0), username: smartevse
2024-08-12T15:45:19.584352+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/EVCurrentL1, username: smartevse, payload: 0
2024-08-12T15:45:19.584465+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVCurrentL2, PacketId=undefined, Payload=0), username: smartevse
2024-08-12T15:45:19.584579+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/EVCurrentL2, username: smartevse, payload: 0
2024-08-12T15:45:19.584630+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVCurrentL3, PacketId=undefined, Payload=0), username: smartevse
2024-08-12T15:45:19.584735+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/EVCurrentL3, username: smartevse, payload: 0
2024-08-12T15:45:19.584842+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/ESPUptime, PacketId=undefined, Payload=971093), username: smartevse
2024-08-12T15:45:19.584959+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/ESPUptime, username: smartevse, payload: 971093
2024-08-12T15:45:19.585188+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/ESPTemp, PacketId=undefined, Payload=42), username: smartevse
2024-08-12T15:45:19.585310+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/ESPTemp, username: smartevse, payload: 42
2024-08-12T15:45:19.585526+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/Mode, PacketId=undefined, Payload=Off), username: smartevse
2024-08-12T15:45:19.585667+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/Mode, username: smartevse, payload: Off
2024-08-12T15:45:19.586037+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/MaxCurrent, PacketId=undefined, Payload=250), username: smartevse
2024-08-12T15:45:19.586149+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/MaxCurrent, username: smartevse, payload: 250
2024-08-12T15:45:19.586357+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/ChargeCurrent, PacketId=undefined, Payload=60), username: smartevse
2024-08-12T15:45:19.586447+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/ChargeCurrent, username: smartevse, payload: 60
2024-08-12T15:45:19.586581+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/ChargeCurrentOverride, PacketId=undefined, Payload=0), username: smartevse
2024-08-12T15:45:19.586675+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/ChargeCurrentOverride, username: smartevse, payload: 0
2024-08-12T15:45:19.586786+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/Access, PacketId=undefined, Payload=Deny), username: smartevse
2024-08-12T15:45:19.586879+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/Access, username: smartevse, payload: Deny
2024-08-12T15:45:19.693458+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: {frame_error,#{cause => zero_remaining_len,header_type => 0}}, username: smartevse
2024-08-12T15:45:19.693719+02:00 [SOCKET] SmartEVSE-6360@192.168.88.216:59592 msg: socket_force_closed, reason: [cause: zero_remaining_len, header_type: 0], username: smartevse
2024-08-12T15:45:19.694037+02:00 [AUTHZ] SmartEVSE-6360@192.168.88.216:59592 msg: authorization_module_nomatch, action: PUBLISH(Q0,R1), module: emqx_authz_client_info, topic: SmartEVSE-6360/connected, username: smartevse
2024-08-12T15:45:19.694169+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/connected, username: smartevse, payload: offline
2024-08-12T15:45:22.653472+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59593 msg: mqtt_packet_received, packet: CONNECT(Q0, R0, D0, ClientId=SmartEVSE-6360, ProtoName=MQTT, ProtoVsn=4, CleanStart=false, KeepAlive=15, Username=smartevse, Password=******, Will(Q0, R1, Topic=SmartEVSE-6360/connected Payload=offline)), username: smartevse
2024-08-12T15:45:22.654006+02:00 [SUBSCRIBE] SmartEVSE-6360@192.168.88.216:59593 msg: subscribe, sub_id: SmartEVSE-6360, sub_opts: [nl: 0, qos: 1, rap: 0, rh: 0, sub_props: []], topic: SmartEVSE-6360/Set/#, username: smartevse
2024-08-12T15:45:22.654310+02:00 [UNSUBSCRIBE] SmartEVSE-6360@192.168.88.216:59592 msg: unsubscribe, sub_opts: [nl: 0, qos: 1, subid: SmartEVSE-6360, rap: 0, rh: 0, sub_props: []], topic: SmartEVSE-6360/Set/#, username: smartevse
2024-08-12T15:45:22.654831+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59593 msg: mqtt_packet_sent, packet: CONNACK(Q0, R0, D0, AckFlags=1, ReasonCode=0), username: smartevse
2024-08-12T15:45:22.654474+02:00 [SOCKET] SmartEVSE-6360@192.168.88.216:59592 msg: emqx_connection_terminated, reason: {shutdown,takenover}, username: smartevse
2024-08-12T15:45:22.689734+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59593 msg: mqtt_packet_received, packet: SUBSCRIBE(Q1, R0, D0, PacketId=36 TopicFilters=[SmartEVSE-6360/Set/#(#{nl => 0,qos => 1,rap => 0,rh => 0})]), username: smartevse
2024-08-12T15:45:22.690216+02:00 [AUTHZ] SmartEVSE-6360@192.168.88.216:59593 msg: authorization_module_nomatch, action: SUBSCRIBE(Q1), module: emqx_authz_client_info, topic: SmartEVSE-6360/Set/#, username: smartevse
2024-08-12T15:45:22.690377+02:00 [SUBSCRIBE] SmartEVSE-6360@192.168.88.216:59593 msg: subscribe, sub_id: SmartEVSE-6360, sub_opts: [nl: 0, qos: 1, rap: 0, rh: 0, sub_props: []], topic: SmartEVSE-6360/Set/#, username: smartevse
[... cut ...]  "command_template" : "{{ value | int * 10 }}"}
2024-08-12T15:45:30.660619+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:59593 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL1, PacketId=undefined, Payload=-10), username: smartevse
2024-08-12T15:45:30.660874+02:00 [AUTHZ] SmartEVSE-6360@192.168.88.216:59593 msg: authorization_module_nomatch, action: PUBLISH(Q0,R0), module: emqx_authz_client_info, topic: SmartEVSE-6360/MainsCurrentL1, username: smartevse

```

hmmbob, 2024-08-12:
Looks like the existing client connection runs into an error, and then the SmartEVSE opens a new connection. 

See these lines, and the difference between `SmartEVSE-6360@192.168.88.216:59592` and `SmartEVSE-6360@192.168.88.216:59593`

```
[MQTT] SmartEVSE-6360@192.168.88.216:59592 msg: mqtt_packet_received, packet: {frame_error,#{cause => zero_remaining_len,header_type => 0}}, username: smartevse
[SOCKET] SmartEVSE-6360@192.168.88.216:59592 msg: socket_force_closed, reason: [cause: zero_remaining_len, header_type: 0], username: smartevse
[PUBLISH] SmartEVSE-6360@192.168.88.216:59592 msg: publish_to, topic: SmartEVSE-6360/connected, username: smartevse, payload: offline
[MQTT] SmartEVSE-6360@192.168.88.216:59593 msg: mqtt_packet_received, packet: CONNECT(Q0, R0, D0, ClientId=SmartEVSE-6360, ProtoName=MQTT, ProtoVsn=4, CleanStart=false, KeepAlive=15, Username=smartevse, Password=******, Will(Q0, R1, Topic=SmartEVSE-6360/connected Payload=offline)), username: smartevse
[SUBSCRIBE] SmartEVSE-6360@192.168.88.216:59593 msg: subscribe, sub_id: SmartEVSE-6360, sub_opts: [nl: 0, qos: 1, rap: 0, rh: 0, sub_props: []], topic: SmartEVSE-6360/Set/#, username: smartevse
[UNSUBSCRIBE] SmartEVSE-6360@192.168.88.216:59592 msg: unsubscribe, sub_opts: [nl: 0, qos: 1, subid: SmartEVSE-6360, rap: 0, rh: 0, sub_props: []], topic: SmartEVSE-6360/Set/#, username: smartevse
[MQTT] SmartEVSE-6360@192.168.88.216:59593 msg: mqtt_packet_sent, packet: CONNACK(Q0, R0, D0, AckFlags=1, ReasonCode=0), username: smartevse
[SOCKET] SmartEVSE-6360@192.168.88.216:59592 msg: emqx_connection_terminated, reason: {shutdown,takenover}, username: smartevse
[MQTT] SmartEVSE-6360@192.168.88.216:59593 msg: mqtt_packet_received, packet: SUBSCRIBE(Q1, R0, D0, PacketId=36 TopicFilters=[SmartEVSE-6360/Set/#(#{nl => 0,qos => 1,rap => 0,rh => 0})]), username: smartevse
[SUBSCRIBE] SmartEVSE-6360@192.168.88.216:59593 msg: subscribe, sub_id: SmartEVSE-6360, sub_opts: [nl: 0, qos: 1, rap: 0, rh: 0, sub_props: []], topic: SmartEVSE-6360/Set/#, username: smartevse
```

hmmbob, 2024-08-12:
Euh, then I'll need to rewire everything. Like, I have 37 devices connected to HA through MQTT.....

I've updated the setting `max_packet_size` of EMQX from the default 1MB to 10MB - even though I can hardly imagine that the SmartEVSE sends >1MB packets....

dingo35, 2024-08-12:
?!? You just install mosquitto on your favorite platform, since you dont do authorization that will cost you 5 minutes, and you bring down HA, then you bring down EMQX, and you bring up HA.

It has nothing to do with the hardware ?!?

dingo35, 2024-08-12:
An alternative would be to get EMQX having a more detailed log; if we can get the actual MQTT message that EMQX chokes on, I can determine which publish call in SmartEVSE is the culpritt (if any)...

hmmbob, 2024-08-12:
> ?!? You just install mosquitto on your favorite platform, since you dont do authorization that will cost you 5 minutes, and you bring down HA, then you bring down EMQX, bring up mosquitto, and you bring up HA.
> 
> It has nothing to do with the hardware ?!?

I see what you're saying, just afraid of all entities names changing....

> An alternative would be to get EMQX having a more detailed log; if we can get the actual MQTT message that EMQX chokes on, I can determine which publish call in SmartEVSE is the culpritt (if any)...

I tried googling it, but did not find the option yet. I'll try and dig deeper.

> Another alternative: v3.6.5 is still on mongoose 7.13, in v3.6.6 we switched to mongoose 7.14; that might have something to do with it ?!?

This is probably the easiest step to do. Is a downgrade safe to do?

dingo35, 2024-08-12:
It must be possible to log in more detail, look here:

https://forum.emqx.io/t/emqx-gateway-connection-error-frame-error-malformed-message-len/188

Not sure if mongoose puts multiple mqtt messages in a single UDP packet, but that might be the problem..

dingo35, 2024-08-12:
As to topics and payloads, they are determined by SmartEVSE, so I wouldnt expect changes in HA entities when switching to mosquitto, unless you configured some renaming in EMQX.

You could even keep HA down and monitor traffic through mosquitto_sub.

hmmbob, 2024-08-13:
It just happened again (SmartEVSE 3.6.5!). Leaving this here; I also started debug logs but they weren't running yet when this happened.

```
2024-08-13T07:29:29.711424+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:63336 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/Access, PacketId=undefined, Payload=Deny), username: smartevse
2024-08-13T07:29:29.711497+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:63336 msg: publish_to, topic: SmartEVSE-6360/Access, username: smartevse, payload: Deny
2024-08-13T07:29:29.711622+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:63336 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/RFID, PacketId=undefined, Payload=Ready to read card), username: smartevse
2024-08-13T07:29:29.711714+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:63336 msg: publish_to, topic: SmartEVSE-6360/RFID, username: smartevse, payload: Ready to read card
2024-08-13T07:29:29.711861+02:00 [debug] tag: MQTT, clientid: SmartEVSE-6360, msg: raw_bin_received, peername: 192.168.88.216:63336, username: smartevse, size: 609, type: hex, bin: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003129001B536D617274455653452D363336302F524649444C6173745265616430303030303030303030303031250014536D617274455653452D363336302F5374617465526561647920746F20436861726765311A0014536D617274455653452D363336302F4572726F724E6F6E653125001A536D617274455653452D363336302F4556506C75675374617465436F6E6E6563746564
2024-08-13T07:29:29.712352+02:00 [info] tag: MQTT, clientid: SmartEVSE-6360, peername: 192.168.88.216:63336, username: smartevse, reason: #{cause => zero_remaining_len,header_type => 0}, at_state: <<"clean">>, input_bytes: <<0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,49,41,0,27,83,109,97,114,116,69,86,83,69,45,54,51,54,48,47,82,70,73,68,76,97,115,116,82,101,97,100,48,48,48,48,48,48,48,48,48,48,48,48,49,37,0,20,83,109,97,114,116,69,86,83,69,45,54,51,54,48,47,83,116,97,116,101,82,101,97,100,121,32,116,111,32,67,104,97,114,103,101,49,26,0,20,83,109,97,114,116,69,86,83,69,45,54,51,54,48,47,69,114,114,111,114,78,111,110,101,49,37,0,26,83,109,97,114,116,69,86,83,69,45,54,51,54,48,47,69,86,80,108,117,103,83,116,97,116,101,67,111,110,110,101,99,116,101,100>>, parsed_packets: []
2024-08-13T07:29:29.712974+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:63336 msg: mqtt_packet_received, packet: {frame_error,#{cause => zero_remaining_len,header_type => 0}}, username: smartevse
2024-08-13T07:29:29.713058+02:00 [SOCKET] SmartEVSE-6360@192.168.88.216:63336 msg: socket_force_closed, reason: [cause: zero_remaining_len, header_type: 0], username: smartevse
2024-08-13T07:29:29.713251+02:00 [debug] clientid: SmartEVSE-6360, msg: maybe_publish_will_msg_other_publish, peername: 192.168.88.216:63336, username: smartevse, reason: sock_closed
2024-08-13T07:29:29.713529+02:00 [AUTHZ] SmartEVSE-6360@192.168.88.216:63336 msg: authorization_module_nomatch, action: PUBLISH(Q0,R1), module: emqx_authz_client_info, topic: SmartEVSE-6360/connected, username: smartevse
2024-08-13T07:29:29.713595+02:00 [info] tag: AUTHZ, clientid: SmartEVSE-6360, msg: authorization_permission_allowed, peername: 192.168.88.216:63336, username: smartevse, topic: SmartEVSE-6360/connected, source: file, action: PUBLISH(Q0,R1)
2024-08-13T07:29:29.713874+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:63336 msg: publish_to, topic: SmartEVSE-6360/connected, username: smartevse, payload: offline
2024-08-13T07:29:32.827000+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:63337 msg: mqtt_packet_received, packet: CONNECT(Q0, R0, D0, ClientId=SmartEVSE-6360, ProtoName=MQTT, ProtoVsn=4, CleanStart=false, KeepAlive=15, Username=smartevse, Password=******, Will(Q0, R1, Topic=SmartEVSE-6360/connected Payload=offline)), username: smartevse
[... cut ...] tEVSE-6360-MainsCurrentL1", "unique_id" : "SmartEVSE-6360-MainsCurrentL1", "state_topic" : "SmartEVSE-6360/MainsCurrentL1", "availability_topic" : "SmartEVSE-6360/connected", "device": {"model" : "SmartEVSE v3", "identifiers" : "SmartEVSE-6360", "name" : "SmartEVSE-6360", "manufacturer" : "Stegen", "configuration_url" : "http://192.168.88.216", "sw_version" : "v3.6.5"}, "device_class" : "current", "unit_of_measurement" : "A", "value_template" : "{{ value | int / 10 }}"}), username: smartevse
```

Imaginous, 2024-08-13:
I'm using my own broker, but recently noticed topics with non ASCII characters (with MQTT Explorer). I assumed it was "my bad" since it's my own code.

But reading this it might be an issue.
BTW running the beta with the "WaitingForSolar" timer I haven't seen it yet.

Just my 2 cents.

hmmbob, 2024-08-13:
> But reading this it might be an issue. BTW running the beta with the "WaitingForSolar" timer I haven't seen it yet.

That's interesting, because of [this comment](https://github.com/dingo35/SmartEVSE-3.5/issues/113#issuecomment-2279973874)
[image]

I would suspect that [this commit](https://github.com/dingo35/SmartEVSE-3.5/commit/39d875b4052f4a192cb9545428a87c4510e157dc) might be the key here?

adjego, 2024-08-13:
I've also installed the beta from [https://github.com/dingo35/SmartEVSE-3.5/issues/113#issuecomment-2279973874](url) so will run this for a day and come back to you all what i see.

dingo35, 2024-08-13:
Thanks all, seems we are on to something!

The version you are testing publishes SolarStopTime approx. every second, even if it doesn't change. If that annoys you / your MQTT server, this version only publishes on change:
[95f4bbe3.zip](https://github.com/user-attachments/files/16595606/95f4bbe3.zip)

hmmbob, 2024-08-13:
> Thanks all, seems we are on to something!
>
> The version you are testing publishes SolarStopTime approx. every second, even if it doesn't change. If that annoys you / your MQTT server, this version [...]

Installed, just for the sake of trying 😉

adjego, 2024-08-13:
> Installed, just for the sake of trying 😉

Me too!

adjego, 2024-08-13:
It seems i already have the same issue still when using this version:

1723536123: Client SmartEVSE disconnected due to protocol error.
1723536124: New connection from 192.168.1.147:49571 on port 1883.
1723536124: New client connected from 192.168.1.147:49571 as SmartEVSE (p2, c0, k15).

dingo35, 2024-08-13:
Ok, could you perhaps try the "publish SolarStopTimer every second" version, so we have a complete picture?

hmmbob, 2024-08-13:
Just happened again, I am on the not-every-second version. Will flash the every-second version now.

```
2024-08-13T11:40:51.602845+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVTotalEnergyCharged, PacketId=undefined, Payload=26179), username: smartevse
2024-08-13T11:40:51.603055+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/EVTotalEnergyCharged, username: smartevse, payload: 26179
2024-08-13T11:41:02.546523+02:00 [debug] tag: MQTT, clientid: SmartEVSE-6360, msg: raw_bin_received, peername: 192.168.88.216:57194, username: smartevse, size: 643, type: hex, bin: 3021001D536D617274455653452D363336302F4D61696E7343757272656E744C3133303022001D536D617274455653452D363336302F4D61696E7343757272656E744C322D34303022001D536D617274455653452D363336302F4D61696E7343757272656E744C332D3330301D001A536D617274455653452D363336302F455643757272656E744C3130301D001A536D617274455653452D363336302F455643757272656E744C3230301D001A536D617274455653452D363336302F455643757272656E744C3330301E0018536D617274455653452D363336302F455350557074696D6539303330301A0016536D617274455653452D363336302F45535054656D70333831180013536D617274455653452D363336302F4D6F64654F6666311E0019536D617274455653452D363336302F4D617843757272656E74323530311F001C536D617274455653452D363336302F43686172676543757272656E743031270024536D617274455653452D363336302F43686172676543757272656E744F7665727269646530311B0015536D617274455653452D363336302F41636365737344656E7931270013536D617274455653452D363336302F52464944526561647920746F207265616420636172643129001B536D617274455653452D363336302F524649444C6173745265616430303030303030303030303031250014536D617274455653452D363336302F5374617465526561647920746F20436861726765311A0014536D617274455653452D363336302F4572726F724E6F6E653125001A536D617274455653452D363336302F4556506C75675374617465436F6E6E6563746564311E0017536D617274455653452D363336302F576946695353494442424D544C
2024-08-13T11:41:02.547505+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL1, PacketId=undefined, Payload=30), username: smartevse
2024-08-13T11:41:02.548252+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/MainsCurrentL1, username: smartevse, payload: 30
2024-08-13T11:41:02.548540+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL2, PacketId=undefined, Payload=-40), username: smartevse
2024-08-13T11:41:02.548803+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/MainsCurrentL2, username: smartevse, payload: -40
2024-08-13T11:41:02.549030+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL3, PacketId=undefined, Payload=-30), username: smartevse
2024-08-13T11:41:02.549268+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/MainsCurrentL3, username: smartevse, payload: -30
2024-08-13T11:41:02.549514+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVCurrentL1, PacketId=undefined, Payload=0), username: smartevse
2024-08-13T11:41:02.549779+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/EVCurrentL1, username: smartevse, payload: 0
2024-08-13T11:41:02.550021+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVCurrentL2, PacketId=undefined, Payload=0), username: smartevse
2024-08-13T11:41:02.550268+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/EVCurrentL2, username: smartevse, payload: 0
2024-08-13T11:41:02.550449+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVCurrentL3, PacketId=undefined, Payload=0), username: smartevse
2024-08-13T11:41:02.550654+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/EVCurrentL3, username: smartevse, payload: 0
2024-08-13T11:41:02.550876+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/ESPUptime, PacketId=undefined, Payload=9030), username: smartevse
2024-08-13T11:41:02.551083+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/ESPUptime, username: smartevse, payload: 9030
2024-08-13T11:41:02.551330+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/ESPTemp, PacketId=undefined, Payload=38), username: smartevse
2024-08-13T11:41:02.551570+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/ESPTemp, username: smartevse, payload: 38
2024-08-13T11:41:02.551747+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57194 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/Mode, PacketId=undefined, Payload=Off), username: smartevse
2024-08-13T11:41:02.552119+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:57194 msg: publish_to, topic: SmartEVSE-6360/Mode, username: smartevse, payload: Off
[... cut ...] 3452D363336302F57694669525353492D3530301F001C536D617274455653452D363336302F4556436861726765506F7765723031280025536D617274455653452D363336302F455645564D65746572456E657267794368617267656430302A0023536D617274455653452D363336302F4556546F74616C456E65726779436861726765643236313739
2024-08-13T11:41:13.657288+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:57195 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL1, PacketId=undefined, Payload=30), username: smartevse
```

dingo35, 2024-08-13:
I just happened to capture a malformed packet:

[image]

Note that:
1. I am missing the protocol version, that should be an easy fix
2. A LOT of publish messages are combined into a single TCP packet
3. The malformed packet tries to publish EVTotalEnergyCharged, and for some reason the topic has a huge length: 13105 bytes !?!?!

Because of the multiple tcp transmissions from my MQTT server 10.0.0.28 back to my SmartEVSE 10.0.0.76 this looks like the TCP packet was crippled?!?

hmmbob, 2024-08-13:
I'm no C expert, but does this make sense?

https://github.com/dingo35/SmartEVSE-3.5/blob/743d0e379f94521d633f6f35d17bc0a18b07c683/SmartEVSE-3/src/evse.cpp#L2905-L2906

dingo35, 2024-08-13:
Yes its called function overloading.... was not there in 3.6.7 so that is not the problem...

dingo35, 2024-08-13:
Ok the connect version error message on wireshark has to do with the fact that I did not capture the CONNECT packet, if I capture that then the message disappears.

It seems to go wrong when the TCP packet gets crippled, all the previous versions worked on MQTT 3.1.1, here is a version with MQTT 5, perhaps that protocol has a better error handling ?
[mqtt5.zip](https://github.com/user-attachments/files/16598540/mqtt5.zip)

hmmbob, 2024-08-13:
Installed, let's see.

The broker recognizes the MQTTv5 connection at least. 
The change also changed the "session expiry interval" setting 🤔 (was 2 hrs before, see https://gathering.tweakers.net/forum/list_message/79816884#79816884)

[image]

hmmbob, 2024-08-13:
Jajajajajaja lekker bezig 

I'm running the MQTTv5 version now, waiting for an error. Then I'll proceed in order with the above options.

adjego, 2024-08-13:
> Ok the connect version error message on wireshark has to do with the fact that I did not capture the CONNECT packet, if I capture that then the message disappears.
>
> It seems to go wrong when the TCP [...]

Already have the issue again:

1723551057: Client SmartEVSE disconnected due to protocol error.
1723551057: New connection from 192.168.1.147:50199 on port 1883.
1723551057: New client connected from 192.168.1.147:50199 as SmartEVSE (p5, c0, k15).

Will run the next build in line :)

adjego, 2024-08-13:
> ...and here I'm trying another direction: MQTT version 3.1.1, but now compiled with -DMG_ENABLE_LWIP=1 which forces mongoose to use the ESP32's LWIP network stack.....
> 
> [MG_ENABLE_LWIP.zip](https://github.com/user-attachments/files/16599064/MG_ENABLE_LWIP.zip) Enjoy!

And also this one still has the issue:

1723554928: Client SmartEVSE disconnected due to protocol error.
1723554929: New connection from 192.168.1.147:55609 on port 1883.
1723554929: New client connected from 192.168.1.147:55609 as SmartEVSE (p2, c0, k15).

Will go for the last version available at this moment.

hmmbob, 2024-08-13:
Whoa, you have way quicker occurrences then I have. Mine is still running fine on the MQTTv5 version

dingo35, 2024-08-13:
Im guessing it has to do with the reliability of your wifi signal.
If I press "Raw" my production unit gives RSSI -70, perhaps @adjego 's is lower (=worse), and yours @hmmbob is higher?

hmmbob, 2024-08-13:
> Im guessing it has to do with the reliability of your wifi signal. If I press "Raw" my production unit gives RSSI -70, perhaps @adjego 's is lower (=worse), and yours @hmmbob is higher?

Mine is higher/better indeed. No need for `raw`, we publish that to MQTT as well 😉 

[image]

adjego, 2024-08-13:
> ....and here the opposite, forcing mongoose to use its builtin network stack, by compiling wit -DMG_ENABLE_TCPIP=1
> 
> [MG_ENABLE_TCPIP.zip](https://github.com/user-attachments/files/16599106/MG_ENABLE_TCPIP.zip)
> 
> .... uses slightly more flash ram...
> 
> EDIT: still on MQTT version 3.1.1

This version causes a bootloop on my Smartevse.

> Mine is higher/better indeed. No need for `raw`, we publish that to MQTT as well 😉 [...]

Well, mine is even higher as the Wifi AC is almost on top of it.

"wifi": {
"status": "WL_CONNECTED",
"ssid": "Laadpaal",
"rssi": -27,
"bssid": "18:E8:29:A1:83:85"
},

hmmbob, 2024-08-13:
I'm also seeing the error with MQTTv5.

```
2024-08-13T19:15:57.078351+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:63858 msg: mqtt_packet_received, packet: {frame_error,#{cause => zero_remaining_len,header_type => 0}}, username: smartevse
2024-08-13T19:15:57.078510+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:63858 msg: mqtt_packet_sent, packet: DISCONNECT(Q0, R0, D0, ReasonCode=129), username: smartevse
2024-08-13T19:15:57.078771+02:00 [SOCKET] SmartEVSE-6360@192.168.88.216:63858 msg: socket_force_closed, reason: [cause: zero_remaining_len, header_type: 0], username: smartevse
2024-08-13T19:15:57.079045+02:00 [AUTHZ] SmartEVSE-6360@192.168.88.216:63858 msg: authorization_module_nomatch, action: PUBLISH(Q0,R1), module: emqx_authz_client_info, topic: SmartEVSE-6360/connected, username: smartevse
2024-08-13T19:15:57.079164+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:63858 msg: publish_to, topic: SmartEVSE-6360/connected, username: smartevse, payload: offline

```

I am not sending mains data via MQTT, but via the web-api (http_post) and I'm sending it every 2 seconds.

Could it be something threading?

dingo35, 2024-08-14:
Ok guys here we go again; this version has double size send/rcv buffers for mongoose:

[MG_IO_SIZE=4096.zip](https://github.com/user-attachments/files/16610707/MG_IO_SIZE.4096.zip)

dingo35, 2024-08-14:
And this version has a 100ms delay in the middle of all the publish commands, so now the TCP packets are 400bytes instead of 800 bytes. More of a workaround then a solution, but I want to see if it helps:

[100ms_delay.zip](https://github.com/user-attachments/files/16610844/100ms_delay.zip)

Glad you guys are helping hunting this bug I'm only getting this error once every 24 hours or so ....

Imaginous, 2024-08-14:
I noticed that messages seem to be send when they change. Maybe some/all messages can be grouped and send in bulk. 

This can be done on any change of a value. 

Maybe this alleviates the complexity and makes it more robust.

dingo35, 2024-08-14:
No if you look at the code they are sent in bulk approx every 10s . Look at PublishData().

hmmbob, 2024-08-14:
> Ok guys here we go again; this version has double size send/rcv buffers for mongoose:
> 
> [MG_IO_SIZE=4096.zip](https://github.com/user-attachments/files/16610707/MG_IO_SIZE.4096.zip)

I'll start with this one, as this seems to be the 'nicest' solution.

> I noticed that messages seem to be send when they change. Maybe some/all messages can be grouped and send in bulk.

I can confirm that they're sent in bulk, I see them coming in at my broker every 10sec

adjego, 2024-08-14:
> Ok guys here we go again; this version has double size send/rcv buffers for mongoose:
> 
> [MG_IO_SIZE=4096.zip](https://github.com/user-attachments/files/16610707/MG_IO_SIZE.4096.zip)

I will also install this one! And start testing again!

adjego, 2024-08-14:
> > Ok guys here we go again; this version has double size send/rcv buffers for mongoose:
> > [MG_IO_SIZE=4096.zip](https://github.com/user-attachments/files/16610707/MG_IO_SIZE.4096.zip)
> 
> I will also install this one! And start testing again!

This was a fast one:

1723635869: Client SmartEVSE disconnected due to malformed packet.
1723635870: New connection from 192.168.1.147:64531 on port 1883.
1723635870: New client connected from 192.168.1.147:64531 as SmartEVSE (p2, c0, k15).

Please note the error did change from "protocol error" to "malformed packet".

Will try the 100ms delay version now.

adjego, 2024-08-14:
> And this version has a 100ms delay in the middle of all the publish commands, so now the TCP packets are 400bytes instead of 800 bytes. More of a workaround then a solution, but I want to see if it [...]

And this one shows "protocol error" again.

1723638009: Client SmartEVSE disconnected due to protocol error.
1723638010: New connection from 192.168.1.147:49946 on port 1883.
1723638010: New client connected from 192.168.1.147:49946 as SmartEVSE (p2, c0, k15).

dingo35, 2024-08-14:
@adjego thx for testing! Since you seem to have the most frequent issues, can you describe your infra here: upload your SmartEVSE config, wifi AP brand/model, do you use VLAN's, is your LAN heavily used or normal SoHo setup ?

Trying to find a pattern here....

adjego, 2024-08-14:
Config:

`{
"version": "v3.6.7",
"serialnr": 6850,
"mode": "SOLAR",
"mode_id": 2,
"car_connected": false,
"wifi": {
"status": "WL_CONNECTED",
"ssid": "Laadpaal",
"rssi": -35,
"bssid": "18:E8:29:A1:83:85"
},
"evse": {
"temp": 34,
"temp_max": 65,
"connected": false,
"access": true,
"mode": 2,
"loadbl": 0,
"pwm": 1024,
"solar_stop_timer": 0,
"state": "Ready to Charge",
"state_id": 0,
"error": "None",
"error_id": 0,
"rfid": "Not Installed"
},
"settings": {
"charge_current": 0,
"override_current": 0,
"current_min": 6,
"current_max": 16,
"current_main": 25,
"current_max_circuit": 16,
"current_max_sum_mains": 600,
"max_sum_mains_time": 0,
"solar_max_import": 0,
"solar_start_current": 5,
"solar_stop_time": 10,
"enable_C2": "Solar Off",
"mains_meter": "API",
"starttime": 0,
"stoptime": 0,
"repeat": 0
},
"mqtt": {
"host": "192.168.1.8",
"port": 1883,
"topic_prefix": "SmartEVSE",
"username": "",
"password_set": false,
"status": "Connected"
},
"home_battery": {
"current": 0,
"last_update": 0
},
"ev_meter": {
"description": "Disabled",
"address": 12,
"import_active_power": 0,
"total_kwh": 0,
"charged_kwh": 0,
"currents": {
"TOTAL": 0,
"L1": 0,
"L2": 0,
"L3": 0
},
"import_active_energy": 0,
"export_active_energy": 0
},
"mains_meter": {
"import_active_energy": 0,
"export_active_energy": 0
},
"phase_currents": {
"TOTAL": -8,
"L1": -9,
"L2": 0,
"L3": 1,
"last_data_update": 1723639745,
"original_data": {
"TOTAL": -8,
"L1": -9,
"L2": 0,
"L3": 1
}
},
"backlight": {
"timer": 0,
"status": "OFF"
}
}`

I have Unifi Mesh AC where the Smartevse is connected to, no Vlans in use and my Home Assistant and MQTT broker is running on a Raspberry Pi4 with 4GB ram. 

Next to this i also run some other docker containers:
- Unifi Controller
- Teslamate
- Zwave JS (uses MQTT)
- DSMR reader

Also have a seperate kWh meter to measure house consumption due to having a too old smartmeter. Running Python script with RS485 to retrieve the info from the meter and sent it using MQTT to Home Assistant and SmartEvse.

If more info is needed, please let me know :)

adjego, 2024-08-14:
What i also noticed is because i sent my mains every second, also the smartevse is sending this data + all the other data als every second. I just stopped my process regarding mains, and now the interval from the SmartEvse is 10 seconds.

Is it usefull to even sent the data from the Smartevse more often when for example mains data is updated? Or might it be smart to also have some limit here?

dingo35, 2024-08-14:
If you /SetMainsMeter by MQTT, the publishing is triggered immediately.

Things you can try:
-update less frequently, e.g. once per 2 sec or once per 3 sec
-update via the Http REST interface

These are of course no solutions, but trying to isolate the problem....

adjego, 2024-08-14:
I think i found something, which is related to the protocol error

Publish of "smartevse/Error" and some other topics:
[image]

Misformed publish of "smartevse/Error" and some other topics:
[image]

Misformed packet i will also try to capture

dingo35, 2024-08-14:
Yes didnt we already know that from the mosquitto logs? Malformed packet causes the broker to reconnect, as per protocol spec.
The problem is: what causes the malformed packets? We only do relatively high level calls to mongoose...

hmmbob, 2024-08-14:
My thoughts go to "threading" and "buffers", but it's a complicated one.

One workaround could be to keep using MQTT v5 and use the will delay: as the client is quickly reconnecting one can "delay" the offline message https://docs.oasis-open.org/mqtt/mqtt/v5.0/os/mqtt-v5.0-os.html#_Toc3901062

hmmbob, 2024-08-14:
> > Ok guys here we go again; this version has double size send/rcv buffers for mongoose:
> > [MG_IO_SIZE=4096.zip](https://github.com/user-attachments/files/16610707/MG_IO_SIZE.4096.zip)
> 
> I'll start with this one, as this seems to be the 'nicest' solution. Flashed, keep you posted.
> 
> > I noticed that messages seem to be send when they change. Maybe some/all messages can be grouped and send in bulk.

Yeah, also seeing this now. But, this was the first time today. The raw packets are in the log.

```
2024-08-14T19:43:31.871935+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:53925 msg: publish_to, topic: SmartEVSE-6360/WiFiBSSID, username: smartevse, payload: D8:47:32:F4:08:AC
2024-08-14T19:43:31.872389+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/WiFiRSSI, PacketId=undefined, Payload=-54), username: smartevse
2024-08-14T19:43:31.872911+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:53925 msg: publish_to, topic: SmartEVSE-6360/WiFiRSSI, username: smartevse, payload: -54
2024-08-14T19:43:31.873108+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVChargePower, PacketId=undefined, Payload=0), username: smartevse
2024-08-14T19:43:31.873350+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:53925 msg: publish_to, topic: SmartEVSE-6360/EVChargePower, username: smartevse, payload: 0
2024-08-14T19:43:31.873586+02:00 [debug] tag: MQTT, clientid: SmartEVSE-6360, msg: raw_bin_received, peername: 192.168.88.216:53925, username: smartevse, size: 332, type: hex, bin: 31280025000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000536D617274455653452D363336302F455645564D65746572456E657267794368617267656430302A0023536D617274455653452D363336302F4556546F74616C456E65726779436861726765643236313835
2024-08-14T19:43:31.874098+02:00 [info] tag: MQTT, clientid: SmartEVSE-6360, peername: 192.168.88.216:53925, username: smartevse, reason: #{cause => zero_remaining_len,header_type => 0}, at_state: <<"clean">>, input_bytes: <<0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,83,109,97,114,116,69,86,83,69,45,54,51,54,48,47,69,86,69,86,77,101,116,101,114,69,110,101,114,103,121,67,104,97,114,103,101,100,48,48,42,0,35,83,109,97,114,116,69,86,83,69,45,54,51,54,48,47,69,86,84,111,116,97,108,69,110,101,114,103,121,67,104,97,114,103,101,100,50,54,49,56,53>>, parsed_packets: [{mqtt_packet,{mqtt_packet_header,3,false,0,true},{mqtt_packet_publish,<<0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0>>,undefined,#{}},<<0>>}]
2024-08-14T19:43:31.874933+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=                                     , PacketId=undefined, Payload= ), username: smartevse
2024-08-14T19:43:31.875216+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: {frame_error,#{cause => zero_remaining_len,header_type => 0}}, username: smartevse
2024-08-14T19:43:31.875425+02:00 [SOCKET] SmartEVSE-6360@192.168.88.216:53925 msg: socket_force_closed, reason: topic_name_invalid, username: smartevse
2024-08-14T19:43:31.875943+02:00 [debug] clientid: SmartEVSE-6360, msg: maybe_publish_will_msg_other_publish, peername: 192.168.88.216:53925, username: smartevse, reason: sock_closed
2024-08-14T19:43:31.876887+02:00 [PUBLISH] SmartEVSE-6360@192.168.88.216:53925 msg: publish_to, topic: SmartEVSE-6360/connected, username: smartevse, payload: offline
2024-08-14T19:43:31.877479+02:00 [SOCKET] SmartEVSE-6360@192.168.88.216:53925 msg: socket_force_closed, reason: [cause: zero_remaining_len, header_type: 0], username: smartevse
2024-08-14T19:43:31.996219+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53926 msg: mqtt_packet_received, packet: CONNECT(Q0, R0, D0, ClientId=SmartEVSE-6360, ProtoName=MQTT, ProtoVsn=4, CleanStart=false, KeepAlive=15, Username=smartevse, Password=******, Will(Q0, R1, Topic=SmartEVSE-6360/connected Payload=offline)), username: smartevse
2024-08-14T19:43:31.996952+02:00 [SUBSCRIBE] SmartEVSE-6360@192.168.88.216:53926 msg: subscribe, sub_id: SmartEVSE-6360, sub_opts: [nl: 0, qos: 1, sub_props: [], rap: 0, rh: 0], topic: SmartEVSE-6360/Set/#, username: smartevse
2024-08-14T19:43:31.997441+02:00 [UNSUBSCRIBE] SmartEVSE-6360@192.168.88.216:53925 msg: unsubscribe, sub_opts: [nl: 0, qos: 1, sub_props: [], rap: 0, subid: SmartEVSE-6360, rh: 0], topic: SmartEVSE-6360/Set/#, username: smartevse
2024-08-14T19:43:31.997663+02:00 [debug] clientid: SmartEVSE-6360, msg: emqx_channel_takeover_end, peername: 192.168.88.216:53925, username: smartevse
[... cut ...]  "SmartEVSE-6360-MainsCurrentL2", "unique_id" : "SmartEVSE-6360-MainsCurrentL2", "state_topic" : "SmartEVSE-6360/MainsCurrentL2", "availability_topic" : "SmartEVSE-6360/connected", "device": {"model" : "SmartEVSE v3", "identifiers" : "SmartEVSE-6360", "name" : "SmartEVSE-6360", "manufacturer" : "Stegen", "configuration_url" : "http://192.168.88.216", "sw_version" : "11:17:56 @Aug 14 2024"}, "device_class" : "current", "unit_of_measurement" : "A", "value_template" : "{{ value | int / 10 }}"}

```

dingo35, 2024-08-14:
Yes my mangled packet went wrong on that same variable. 

Another test, this is the normal version BUT with three variables NOT published:
[skip3pub.zip](https://github.com/user-attachments/files/16617240/skip3pub.zip)

[image]

hmmbob, 2024-08-14:
> Yes my mangled packet went wrong on that same variable.
> 
> Another test, this is the normal version BUT with three variables NOT published: [skip3pub.zip](https://github.com/user-attachments/files/16617240/skip3pub.zip)

I'll try this version. 

Meanwhile, I've been comparing a successful 10-second round of publishing at 19:43:20 versus the broken one at 19:43:31.

The 19:43:20 correctly receives and publishes the topics [in the order listed in the code](https://github.com/dingo35/SmartEVSE-3.5/blob/master/SmartEVSE-3/src/evse.cpp#L3066-L3122):

```
2024-08-14T19:43:20.927198+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL1, PacketId=undefined, Payload=10), username: smartevse
2024-08-14T19:43:20.928625+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL2, PacketId=undefined, Payload=-10), username: smartevse
2024-08-14T19:43:20.929455+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/MainsCurrentL3, PacketId=undefined, Payload=10), username: smartevse
2024-08-14T19:43:20.930740+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVCurrentL1, PacketId=undefined, Payload=0), username: smartevse
2024-08-14T19:43:20.932221+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVCurrentL2, PacketId=undefined, Payload=0), username: smartevse
2024-08-14T19:43:20.933334+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVCurrentL3, PacketId=undefined, Payload=0), username: smartevse
2024-08-14T19:43:20.934498+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/ESPUptime, PacketId=undefined, Payload=24597), username: smartevse
2024-08-14T19:43:20.935782+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/ESPTemp, PacketId=undefined, Payload=31), username: smartevse
2024-08-14T19:43:20.937167+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/Mode, PacketId=undefined, Payload=Off), username: smartevse
2024-08-14T19:43:20.938576+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/MaxCurrent, PacketId=undefined, Payload=250), username: smartevse
2024-08-14T19:43:20.940174+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/ChargeCurrent, PacketId=undefined, Payload=0), username: smartevse
2024-08-14T19:43:20.941430+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/ChargeCurrentOverride, PacketId=undefined, Payload=0), username: smartevse
2024-08-14T19:43:20.942796+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/Access, PacketId=undefined, Payload=Deny), username: smartevse
2024-08-14T19:43:20.944066+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/RFID, PacketId=undefined, Payload=Ready to read card), username: smartevse
2024-08-14T19:43:20.945653+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/RFIDLastRead, PacketId=undefined, Payload=000000000000), username: smartevse
2024-08-14T19:43:20.946902+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/State, PacketId=undefined, Payload=Ready to Charge), username: smartevse
2024-08-14T19:43:20.948339+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/Error, PacketId=undefined, Payload=None), username: smartevse
2024-08-14T19:43:20.949689+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/EVPlugState, PacketId=undefined, Payload=Connected), username: smartevse
2024-08-14T19:43:20.951168+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/WiFiSSID, PacketId=undefined, Payload=BBMTL), username: smartevse
2024-08-14T19:43:20.952509+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R1, D0, Topic=SmartEVSE-6360/WiFiBSSID, PacketId=undefined, Payload=D8:47:32:F4:08:AC), username: smartevse
2024-08-14T19:43:20.953770+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/WiFiRSSI, PacketId=undefined, Payload=-54), username: smartevse
2024-08-14T19:43:20.954827+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVChargePower, PacketId=undefined, Payload=0), username: smartevse
[... cut ...]  username: smartevse
2024-08-14T19:43:31.997663+02:00 [debug] clientid: SmartEVSE-6360, msg: emqx_channel_takeover_end, peername: 192.168.88.216:53925, username: smartevse
2024-08-14T19:43:31.997908+02:00 [SOCKET] SmartEVSE-6360@192.168.88.216:53925 msg: emqx_connection_terminated, reason: {shutdown,takenover}, username: smartevse
2024-08-14T19:43:31.998121+02:00 [info] clientid: SmartEVSE-6360, msg: terminate, peername: 192.168.88.216:53925, username: smartevse, reason: {shutdown,takenover}
```

hmmbob, 2024-08-14:
I might be onto something.

Is there a difference in how you treat kWh meters between these versions? 

Because this is a current JSON from the SmartEVSE, with the package you just linked:

```json
{
   "version":"20:06:53 @Aug 14 2024",
   "serialnr":6360,
   "mode":"OFF",
   "mode_id":0,
   "car_connected":true,
   "wifi":{
      "status":"WL_CONNECTED",
      "ssid":"BBMTL",
      "rssi":-56,
      "bssid":"D8:47:32:F4:08:AC"
   },
   "evse":{
      "temp":31,
      "temp_max":65,
      "connected":true,
      "access":false,
      "mode":1,
      "loadbl":0,
      "pwm":1024,
      "solar_stop_timer":0,
      "state":"Ready to Charge",
      "state_id":0,
      "error":"None",
      "error_id":0,
      "rfid":"Ready to read card",
      "rfid_lastread":"000000000000"
   },
   "settings":{
      "charge_current":0,
      "override_current":0,
      "current_min":6,
      "current_max":25,
      "current_main":25,
      "current_max_circuit":25,
      "current_max_sum_mains":600,
      "max_sum_mains_time":0,
      "solar_max_import":0,
      "solar_start_current":5,
      "solar_stop_time":3,
      "enable_C2":"Not present",
      "mains_meter":"API",
      "starttime":1723683660,
      "stoptime":1723687140,
      "repeat":0
   },
   "mqtt":{
      "host":"192.168.88.31",
      "port":1883,
      "topic_prefix":"SmartEVSE-6360",
      "username":"smartevse",
      "password_set":true,
      "status":"Connected"
   },
   "home_battery":{
      "current":0,
      "last_update":0
   },
   "ev_meter":{
      "description":"Eastron3P",
      "address":11,
      "import_active_power":0,
      "total_kwh":3040.800049,
      "charged_kwh":0,
      "currents":{
         "TOTAL":0,
         "L1":0,
         "L2":0,
         "L3":0
      },
      "import_active_energy":3040.800049,
      "export_active_energy":0
   },
   "mains_meter":{
      "import_active_energy":0,
      "export_active_energy":0
   },
   "phase_currents":{
      "TOTAL":30,
      "L1":10,
      "L2":10,
      "L3":10,
      "last_data_update":1723660720,
      "original_data":{
         "TOTAL":30,
         "L1":10,
         "L2":10,
         "L3":10
      }
   },
   "backlight":{
      "timer":0,
      "status":"OFF"
   }
}
```

and the latest report of this value in the MQTTv5 version was.... way different

```
2024-08-14T19:43:20.957333+02:00 [MQTT] SmartEVSE-6360@192.168.88.216:53925 msg: mqtt_packet_received, packet: PUBLISH(Q0, R0, D0, Topic=SmartEVSE-6360/EVTotalEnergyCharged, PacketId=undefined, Payload=26185), username: smartevse
```

And that would possibly explain why I saw a weird drop in my stats in HA:

[image]

And as I keep track of my meter readings for invoicing purposes, July 31 23:59:59 had a meter reading of 2982 kWh

dingo35, 2024-08-14:
Yes that might be a cause: I reworked all scattered kWh meter variables and functions to a class, that might have introduced a problem.
BUT that was only recent (since v3.6.7), and my logs show packet errors as old as May .....

dingo35, 2024-08-14:
You can test v3.6.6 to make sure, that still has the old meter variables in it...

hmmbob, 2024-08-14:
Could #116 be related too?

hmmbob, 2024-08-14:
> You can test v3.6.6 to make sure, that still has the old meter variables in it...

3.6.5 was broken too: https://github.com/dingo35/SmartEVSE-3.5/issues/115#issuecomment-2285398981

adjego, 2024-08-15:
The thing is on my side and on the side from my colleague who also sees these drops, we don't use a seperate kWh meter which is conntected to the Smarteve. Also in my wireshark screenshots, it was the publish message related to "error" publish bundle.

I will run some more wireshark tests today, to see if i can capture a malformed packet.

dingo35, 2024-08-15:
@adjego Yes I saw the same thing, although _very_ spurious in my setup. 

@hmmbob As to the EnergyCharged topic, I found a bug in the EVMeter routine, which is solved in this version:
[578217f6e.zip](https://github.com/user-attachments/files/16622254/578217f6e.zip)

hmmbob, 2024-08-15:
I was also thinking a bit more, and for some reason I keep coming back to "threading". Could that be the issue?

The [docs](https://mongoose.ws/documentation/) specify that Mongoose is not thread-safe (see 2nd green note)

dingo35, 2024-08-15:
If thats the problem the only way to solve it is to move away from mqtt in mongoose... :-(

hmmbob, 2024-08-15:
That being said - did the problems start when you moved from that separate mqtt component to using mongoose mqtt? I don't remember when that was

dingo35, 2024-08-15:
That was April 20th, between v3.5.1 and v3.6.0-RC1 we moved from the MQTT library to mongoose.

hmmbob, 2024-08-15:
> BUT that was only recent (since v3.6.7), and my logs show packet errors as old as May .....

Don't know if your logs go back further, but this gets close....

[53 more replies]

closed 2024-10-25 by dingo35