dingo35/SmartEVSE-3.5 #116: Modbus meter bug (WAGO meter)

ps-nl, 2024-08-14:
**Describe the bug**
In 3.6.7 I still have a bug with readings from the Modbus EVMETER. When idle, it gives spurious readings, including a very high negative current on the L2.  See also the Raw log below

No issues with this meter in version 3.6.2

**Upload your config**
{"version":"v3.6.7","serialnr":###,"mode":"OFF","mode_id":0,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"####","rssi":-62,"bssid":"######"},"evse":{"temp":34,"temp_max":65,"connected":false,"access":false,"mode":1,"loadbl":0,"pwm":1024,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":6,"enable_C2":"Always Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"10.####","port":1883,"topic_prefix":"SmartEVSE-####","username":"smartevse","password_set":true,"status":"Connected"},"home_battery":{"current":11,"last_update":1723618610},"ev_meter":{"description":"WAGO","address":20,"import_active_power":0,"total_kwh":13839.40039,"charged_kwh":0,"currents":{"TOTAL":-20967,"L1":0,"L2":-20972,"L3":5},"import_active_energy":13839.40039,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":7,"L1":-1,"L2":8,"L3":0,"last_data_update":1723618610,"original_data":{"TOTAL":7,"L1":-1,"L2":8,"L3":0}},"backlight":{"timer":120,"status":"ON"}}

**Upload a debug log**
Flash the debug version of the firmware (see https://github.com/dingo35/SmartEVSE-3.5/blob/master/SmartEVSE-3/HowToFlash.txt), telnet to your device, capture the debug log, cut it so that it shows JUST BEFORE and JUST AFTER the problem arises, and upload it here.

**To Reproduce**
Bug is always present, when the EVSE is not charging 

**Expected behavior**
A clear and concise description of what you expected to happen; expected Charging Current?

**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

dingo35, 2024-08-14:
Could you upload a verbose telnet log so we can see what's happening on your modbus?

ps-nl, 2024-08-14:
@dingo35 see the telnet log below. Interestingly, there appears nothing wrong here. It retrieves the correct value of 0 for each phase, as the car is currently not plugged in. 

in the front end, however, I still have incorrect readings. See the attached screenshot

`(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x14, function: 0x03, reg: 0x600c, token:0x1403600c, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 14 03 04 46 58 3d ae
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x14 Function 0x03 Register 0x600c Response
(V) (ModbusSend8)(C0) Sent packet address: 0x14, function: 0x03, reg: 0x5012, token:0x14035012, data: 0x0002.
(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x14, function: 0x03, reg: 0x500c, token:0x1403500c, data: 0x0006.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 14 03 04 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x14 Function 0x03 Register 0x500c Response
(receiveCurrentMeasurement)(C1) DINGO: Meter=WAGO, Power[0]=0.
(receiveCurrentMeasurement)(C1) DINGO: Meter=WAGO, Power[1]=0.
(receiveCurrentMeasurement)(C1) DINGO: Meter=WAGO, Power[2]=0.
(V) (ModbusDecode)(C1) Received packet (15 bytes) 14 03 0c 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x14 Function 0x03 Register 0x0000 Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=15.2 A Imeasured=0.0 A MaxCircuit=16 Imeasured_EV=0.8 A, Battery Current = -1.1 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=15.2 A, Idifference=15.2, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=15.2 A.
(D) (CalcBalancedCurrent)(C0) Checkpoint c: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=15.2A, saveActiveEVSE=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=15.2 A.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 15.2 A, MainsMeter.Timeout=8, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: -2.7 A L2: 1.1 A L3: 0.0 A Isum: -1.6 A`

Screenshot frontend:

[image]

dingo35, 2024-08-14:
Ok I don't have a WAGO currently here, if I misconfigure my Eastron to a WAGO I get the exact same modbus response and the exact same screen; so I suspect the modbus response is incorrect, I would expect a much longer response.

Could you flash an older debug-firmware (the more recent the better!), where your WAGO was working fine, and when idle (so L1-L3 must be 0A), upload the same telnet log like you did before, so I can see what the correct modbus request & response would be?

ps-nl, 2024-08-14:
Here is a debug log from 3.6.2 (which does not have the issue)

(D) (Timer100ms)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x14, function: 0x03, reg: 0x600c, token:0x1403600c, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 14 03 04 46 58 3d ae
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x14 Function 0x03 Register 0x600c Response
(V) (ModbusSend8)(C0) Sent packet address: 0x14, function: 0x03, reg: 0x5012, token:0x14035012, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 14 03 04 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x14 Function 0x03 Register 0x5012 Response
(D) (Timer100ms)(C0) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C0) Sent packet address: 0x14, function: 0x03, reg: 0x500c, token:0x1403500c, data: 0x0006.
(V) (ModbusDecode)(C1) Received packet (15 bytes) 14 03 0c 00 00 00 00 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x14 Function 0x03 Register 0x500c Response
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=16.0 A Imeasured=1.9 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.7 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=16.0 A, Idifference=16.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=16.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=16.0 A.
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.9 A IsetBalanced: 16.0 A, MainsMeterTimeout=9, EVMeterTimeout=64.
(I) (printStatus)(C0) L1: -0.8 A L2: 1.9 A L3: 0.0 A Isum: 1.1 A

dingo35, 2024-08-15:
This should fix it, please test:

[578217f6e.zip](https://github.com/user-attachments/files/16622237/578217f6e.zip)

ps-nl, 2024-08-15:
Yes, looks like that solved it. Many thanks!

closed 2024-08-15 by dingo35