dingo35/SmartEVSE-3.5 #117: SmartEVSE doesn't lower charging current when mains is

x3e, 2024-08-18:
**Describe the bug**

In Smart mode, I have set my main current to 32A, and my circuit current to 31A. Despite this, when the current goes over as high as 39A, the charging current is not lowered or stopped.

**Upload your config**

{"version":"v3.6.8","serialnr":6362,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"[REDACTED]","rssi":-43,"bssid":"[REDACTED]"},"evse":{"temp":38,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":528,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":310,"override_current":0,"current_min":6,"current_max":31,"current_main":32,"current_max_circuit":32,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":2,"solar_start_current":2,"solar_stop_time":10,"enable_C2":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.2.174","port":1884,"topic_prefix":"SmartEVSE-15486","username":"smartevse","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":354,"L1":354,"L2":0,"L3":0,"last_data_update":1723982940,"original_data":{"TOTAL":354,"L1":354,"L2":0,"L3":0}},"backlight":{"timer":120,"status":"ON"}}

**Upload a debug log**

(V) (CalcBalancedCurrent)(C0) [L]Node 0 = 31.0 A(CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=31.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A
IsetBalanced: 31.0 A, MainsMeter.Timeout=9, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: 28.5 A L2: 0.0 A L3: 0.0 A Isum: 28.5 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=31.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery
 Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1723985031.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=39.0 A, Idifference=32.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=39.0 A.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=39.0A, ActiveEVSE=1
.
(V) (CalcBalancedCurrent)(C0) [L]Node 0 = 31.0 A(CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=31.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A
IsetBalanced: 31.0 A, MainsMeter.Timeout=10, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: 39.4 A L2: 0.0 A L3: 0.0 A Isum: 39.4 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=31.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery
 Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=31.0 A, Idifference=32.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=31.0 A.
(V) (CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=31.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A
IsetBalanced: 31.0 A, MainsMeter.Timeout=8, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: 39.4 A L2: 0.0 A L3: 0.0 A Isum: 39.4 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=31.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery
 Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C0) phaseLastUpdate=1723985034.
(V) (CalcBalancedCurrent)(C0) Checkpoint 2 Isetbalanced=39.0 A, Idifference=32.0, mod=0.
(V) (CalcBalancedCurrent)(C0) Checkpoint 4 Isetbalanced=39.0 A.
(D) (CalcBalancedCurrent)(C0) Checkpoint b: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=39.0A, ActiveEVSE=1
.
(V) (CalcBalancedCurrent)(C0) [L]Node 0 = 31.0 A(CalcBalancedCurrent)(C0) Checkpoint 5 Isetbalanced=31.0 A.
(I) (printStatus)(C0) STATE: C Error: 0 StartCurrent: -2 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A
IsetBalanced: 31.0 A, MainsMeter.Timeout=9, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: 39.4 A L2: 0.0 A L3: 0.0 A Isum: 39.4 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=31.0 A Imeasured=0.0 A MaxCircuit=32 Imeasured_EV=0.0 A, Battery
 Current = 0.0 A, mode=1.

**To Reproduce**

Steps to reproduce the behavior:
Get a single phase mains connection of 40A. 
Set mode to SMART. Set main current to 32A. 
Set circuit current to 31A.
Start charging your car. 
Turn on your oven, boiler, washing machine and/or hair dryer. 
Wait until your mains current breaker drops out.

**Expected behavior**

When mains current goes over 32A, the charging current should be (immediately) lowered to the point where the mains current does not exceed 32A. If lowering the current fails, charging should stop.

Happy to answer any questions if needed. Let me know if my settings are incorrect or if I'm misunderstanding some functionality. Thanks in advance for your help!

dingo35, 2024-08-19:
Thanks for your report, very, very serious bug, that turns out to have been there for a long time, but it has been masked by another bug that made the currents being guarded by MaxCircuit, even when MaxCircuit wasn't supposed to be guarded. 
So solving this "incorrect guarding of MaxCircuit" bug made this one visible.

Again, thank you for noticing, this should solve it:
[cdabee582f62.zip](https://github.com/user-attachments/files/16656988/cdabee582f62.zip)

x3e, 2024-08-19:
I can confirm that firmware fixes the bug. Working beautifully now for my setup.
Thank you for your quick reply and your ongoing effort in building this firmware.

dingo35, 2024-08-22:
Fixed with 736d610f45ade and 4072e9d032a in v3.6.9 .

closed 2024-08-22 by dingo35