dingo35/SmartEVSE-3.5 #118: C1 and C2 not switching while car is in charging state

suffe, 2024-08-19:
**Describe the bug**
After connecting the car to the socket the car is recognized and start the charching state, in de app of the car you can see that the car is charching, but the relay that is connected to C1 our C2 is not activated. So there is no current to the car, after a minute the car gives an error. 

**Upload a debug log**
See attachment, there looks something wrong with the information from the sensorbox. 

[image]

dingo35, 2024-08-19:
Does this version solve your problem: 
[cdabee582f62.zip](https://github.com/user-attachments/files/16664234/cdabee582f62.zip)

If not, please upload your /settings ?

dingo35, 2024-08-19:
By the way, you know you should connect the relay between C1 and null?

suffe, 2024-08-19:
> By the way, you know you should connect the relay between C1 and null?

Yes off course 😊 I will try the version hopefully tomorrow, thanks!

dingo35, 2024-08-22:
This looks like a wiring / hardware problem; note that your screenshot shows State B, which means the EV is connected. It is up to your vehicle to decide to go to State C, which is charging.
So if the screenshot is actually of the same moment that your car's app says its charging, I suspect a wiring problem in the CP/PP pin (although State B is detected, which suggests that wiring is ok).
If the screenshot is from a different moment, please check whether the SmartEVSE reaches State C. If it reaches state C, the log should say "Contactor C1 ON", if then the contactor does not switch you have either a wiring problem, a defect contactor or a defect SmartEVSE ....

dingo35, 2024-08-22:
Contactor = relais

suffe, 2024-08-22:
Ah, I checked that one, so that one is working.

suffe, 2024-08-26:
Solved with new hardware, thanks for the help!

closed 2024-08-27 by dingo35