dingo35/SmartEVSE-3.5 #121: Charging beyond current_main is not possible anymore.

daninden, 2024-08-28:
I charge the car beyond what the main can provide by using solar power.
This used to work, but since the last update I made (3.6.3 -> 3.6.9) this is not working anymore.

Raw data:
       "charge_current": 250,
        "override_current": 0,
        "current_min": 6,
        "current_max": 32,
        "current_main": 25,
        "current_max_circuit": 32,

Charge current would previously go up to 320 if enough solar power available.

    "phase_currents": {
        "TOTAL": -50,
        "L1": 144,
        "L2": -99,
        "L3": -95,
        "last_data_update": 1724834299,
        "original_data": {
            "TOTAL": -50,
            "L1": 144,
            "L2": -99,
            "L3": -95
        }
    },

dingo35, 2024-08-31:
This should solve it:
[70ba46e6b.zip](https://github.com/user-attachments/files/16824038/70ba46e6b.zip)

daninden, 2024-09-02:
Installed it, but need to wait till I charge the car again.

daninden, 2024-09-05:
Doing a solarcharge now, looks like it charges beyond maxmains again!

dingo35, 2024-09-07:
Fixed by 1f188304

closed 2024-09-07 by dingo35