dingo35/SmartEVSE-3.5 #122: EVEnergyCharged is no longer sent over MQTT

abstractionnl, 2024-08-28:
**Describe the bug**
The value for EVEnergyCharged is no longer being updated since upgrading to 3.6.9 from 3.6.1

**Upload your config**
{"version":"v3.6.9","serialnr":redacted,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":redacted,"rssi":-67,"bssid":redacted},"evse":{"temp":40,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":0,"pwm":120,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":71,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":5,"enable_C2":"Always Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":redacted,"port":1883,"topic_prefix":redacted,"username":"smartevse","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"API","address":12,"import_active_power":1.399999976,"total_kwh":6495.5,"charged_kwh":6.400000095,"currents":{"TOTAL":59,"L1":0,"L2":0,"L3":59},"import_active_energy":6495.5,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":3,"L1":0,"L2":10,"L3":-7,"last_data_update":1724853316,"original_data":{"TOTAL":3,"L1":0,"L2":10,"L3":-7}},"backlight":{"timer":120,"status":"ON"}}

**To Reproduce**
Upgrading from  3.6.1 to 3.6.9

**Expected behavior**
EVEnergyCharged on MQTT topic is filled with number that is shown on the dashboard. 

**Screenshots**
[image]

abstractionnl, 2024-08-28:
Found the issue in that the topic got wrongly renamed in a previous commit

dingo35, 2024-08-31:
Thanks for your PR, merged it into master.

closed 2024-08-31 by dingo35