dingo35/SmartEVSE-3.5 #125: Solar mode stops charging after a few seconds

ocaj-nl, 2024-08-30:
**Describe the bug**
In solar mode, the car does briefly start charging, but after a few seconds it stops.
After a minute it retries, pattern is repeating a couple of times, then completely stops (not sure if SmartEVSE gives up, or the car).

Problem occurs with both version 3.6.6 and 3.6.9. 
Problem is not present in 3.6.2.

Problem may be related to this line in the logs: 
(MBhandleError)(C1) Error response: E0 - Timeout, address: 09, function: 06, reg: 0001.
Which is strange. It has a timeout on address 9 (broadcast for load-balancing), while I have load-balancing/powershare disabled.

Debug-log:
[debug_log.txt](https://github.com/user-attachments/files/16820562/debug_log.txt)

Config:
[config.txt](https://github.com/user-attachments/files/16820566/config.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. Mode: Solar
2. Connect car, wait for enough solar power

**Expected behavior**
Car continues charging (I have solar-stop timer disabled)

thvd, 2024-08-30:
I can confirm this behaviour with the latest updates. In Solar it starts but very soon it stops again and this repeats itself in a loop.

dingo35, 2024-08-31:
I cannot reproduce this; I changed my test setup to yours (Mains -> API, EV -> Eastron 3P), if Import is exceeded the SolarStopTimer starts running, and when that expires it starts waiting for solar, until StartCurrent is reached and then it starts charging again.. so the behaviour I would expect.

What is the update frequency of your MainsMeter API?

ocaj-nl, 2024-08-31:
My solarstop-timer is disabled (0), so it should never expire?
Update-frequency for Mainsmeter API is approx once per second.

dingo35, 2024-08-31:
Ok I think this is an error in the manual, so that it should say if stoptimer is disabled it will stop charging immediately if a cloud arrives.... I just flashed the old, original manufacturers firmware v3.0.1, and that behaves like that.

I will check with Michael Stegen and report back ....

ocaj-nl, 2024-08-31:
Actually, the documented behaviour (solar-stop Disabled = "Never stop charging") is exactly how it has been working until recently. 

I have configured it like that because I only connect my car to the charger if it needs charging. I then connect it in the evening and next morning as soon as the sun gets up, it will start charging. 
It will charge slowly (3x6A) unless there is more power available, then it will increase the current. If there are clouds or the sun sets before the car is charged, I still want it to keep charging. That is why I have the timer disabled....

This is fine normally, so Solar mode is my default setting. Unless I'm in a hurry, then I just switch to smart-mode.

dingo35, 2024-08-31:
You are right, that is the intended behaviour. Do you happen to know what versions worked fine?

ocaj-nl, 2024-08-31:
3.6.2 is working fine.
3.6.6 and 3.6.9 have changed behaviour.
I did not test any other versions.

So I guess, this new behaviour was introduced in either 3.6.3, 3.6.4, 3.6.5 or 3.6.6?

dingo35, 2024-09-04:
This should solve it, please test:
[641910fa.zip](https://github.com/user-attachments/files/16861899/641910fa.zip)

ocaj-nl, 2024-09-04:
Thanks, I just flashed it in the SmartEVSE. Too late for solar charging, but I will connect my car this evening and inform you tomorrow if it is fixed.

ocaj-nl, 2024-09-05:
> This should solve it, please test: [641910fa.zip](https://github.com/user-attachments/files/16861899/641910fa.zip)

I can confirm that it does. Solar mode is working just fine today, thanks!

thvd, 2024-09-07:
I can also confirm the fix works. Tested it yesterday, it worked!

dingo35, 2024-09-07:
Fixed by 43e5ba36

closed 2024-09-07 by dingo35