dingo35/SmartEVSE-3.5 #126: Extra Max current measurement and setpoint

niekvanprooijen, 2024-09-14:
**Is your feature request related to a problem? Please describe.**
It is a local problem for me, and I guess it will be a problem for more people soon. I have a 1x40A mainsmeter fuse. A couple of weeks ago I installed more solar panels + victron battery system and to protect the fuse-box I installed a so called "REM automaat". Between the "REM" and mains fuse the solar power is added to the net. 
The 3x6mm2 cable to the SmartEVSE's (plurar, 1 is master, 1 is slave) is protected with a 32A fuse. The box which houses the SmartEVSE's has a Eastron 1-phase energy meter connected to the master with RS485. The slave-SmartEVSE has an additional 16A fuse.
So the electrical situation is:
40A mains <-> P1 meter <-> 40A REM <-> 32A Fuse <-> SmartEVSE box <-> kWh Meter <-> Master <-> EV1
......................................................^-> kWh meter <-> Solar + battery............................................................^-> 16A fuse <-> Slave <-> EV2

Before I installed the REM fuse the Eastron was used as EV-meter to protect the 32A fuse and measure charge power and energy charged. P1 was sent with MQTT via HA.

Current situation:
P1 is sent with MQTT via HA.
Current through REM is calculated by HA and sent with MQTT. Used as EV-meter in SmartEVSE as suggested by installation instructions here on github.
Eastron kWh meter is still installed, but not used.
HA sets SmartEVSE to smart mode when prices are low (charge EV's as fast as possible, without triggering REM) and to solar mode when prices are high to prevent feedback into the grid. 

**Describe the solution you'd like**
MainsMeter value is still neccesary to detect feedback into the grid.
I would like to be able to use the Eastron kWh meter in the smartEVSE box to measure the actual charge power and energy AND to keep loadbalancing functionality. Right now it is still connected but not in use. EVMeter functionality is used to protect the REM fuse. 
I think the way to solve this is to add a subpanel breaker to the software so we can keep the standard EVmeter functionality, but still make sure the current through the REM fuse is not exceeded.

dingo35, 2024-09-14:
I think MaxCircuit does what you want. For details look at the manual, hardware installation, subpanel configuration.

niekvanprooijen, 2024-09-14:
I checked the manuals before requesting this.
SmartEVSE has to have a max current for the main fuse, REM fuse and EV fuse. Therefor all 3 current measurements have to be sent to smartEVSE. 
If I set EVMeter to protect the REM, the eastron EV kWh meter wont work. I have to set Max circuit to 40A (REM value).
If I use EVMeter to protect the 32A EV fuse, the REM fuse can trigger if I have more loads active (boiler, heatpump, etc).
Mainsmeter needs to remain the same to detect backfeed for Solar Mode.

dingo35, 2024-09-14:
If you set MaxCircuit to 40A, the REM is protected; you MUST enable PWR SHARE so MaxCircuit is not taken off of the EVMeter, but calculated by Master + Slaves. 
Then you set Max to 32A on the master, the 32A fuse is protected.
Then you set Max to 16A on the slave, the 16A fuse is protected. 
Then you set Mains to 40A so solar feedback is taken into account.

I don't see the problem?

niekvanprooijen, 2024-09-14:
How does it know which current is flowing through the REM?

dingo35, 2024-09-14:
It would guard that all SmartEVSE's do not exceed 40A; I assumed your REM has a built in kWh meter, because you connected it to HA. If it hasn't you have to add an extra EVMeter and use "subpanel configuration". You would lose the EVMeter readings of the Master, because the readings now would be for all users. But the EVMeter is mainly used for display-only.....

dingo35, 2024-09-14:
Add meter = move existing EV meter to subpanel position....

dingo35, 2024-09-14:
So the REM has a kWh meter built in that you can configure as an EVMeter in SmartEVSE?

niekvanprooijen, 2024-09-14:
Sort of, the current through REM is calculated by HA (P1 + Solar + battery) and sent to smartEVSE as EVMeter.

dingo35, 2024-09-14:
Ok so if this is a display-only wish, why don't you buy a $5 RS485-USB stick and connect your EVmeter to HA directly, on a separate modbus? Then you are not straining your existing modbus, you have the info directly in HA, put it on MQTT if you want to... and you are using HA for what is it meant for (be the central switch point for all your home automation) and the SmartEVSE what it is meant for (charging your EV at optimal parameters)...

dingo35, 2025-04-01:
Closed due to inactivity

closed 2025-04-01 by dingo35