dingo35/SmartEVSE-3.5 #127: Node 1 will not start charging with power share enabled

JopGeraets, 2024-09-23:
When node 1 is set to smart/solar/normal and power share is enabled, then the charging does not start.
The following is shown in telnet (debug mode):
```
(I) (printStatus)(C0) STATE: A Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 25.0 A, MainsMeter.Timeout=10, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: -6.7 A L2: -10.2 A L3: -10.2 A Isum: -27.1 A
(ProximityPin)(C0) PP pin: 1023 (1887 mV) (warning: fixed cable configured so PP probably disconnected, making this reading void)
(I) (EVSEStates)(C1) Cable limit: 25A  Max: 25A
(setState)(C1) 16:36:13 STATE A -> COMM_B
(V) (ModbusDecode)(C0) Received packet (6 bytes) 02 04 00 00 00 08
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x04 Register 0x0000 Request
(D) (MBNodeRequest)(C0) Node answering NodeStatus request value[0]=4 value[1]=0 value[2]=0 value[3]=1 value[4]=0 value[5]=1 value[6]=1 value[7]=25
(I) (printStatus)(C0) STATE: COMM_B Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 25.0 A, MainsMeter.Timeout=9, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: -6.7 A L2: -10.2 A L3: -10.2 A Isum: -27.1 A
(V) (ModbusDecode)(C0) Received packet (29 bytes) 09 10 00 20 00 0b 16 00 a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff bc ff 9a ff 9a
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x09 Function 0x10 Register 0x0020 Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A, MainsMeter Irms L1=-6.8A,L2=-10.2A,L3=-10.2A,
(V) (fn_mqtt)(C1) 217 CREATED
(fn_mqtt)(C1) 217 ERROR  DNS lookup failed
(V) (fn_mqtt)(C1) 217 CLOSED
(I) (printStatus)(C0) STATE: COMM_B Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 25.0 A, MainsMeter.Timeout=10, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: -6.8 A L2: -10.2 A L3: -10.2 A Isum: -27.2 A
(V) (ModbusDecode)(C0) Received packet (6 bytes) 02 04 00 00 00 08
(V) (ModbusDecode)(C0) Received packet (6 bytes) 02 04 00 00 00 08
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x04 Register 0x0000 Request
(D) (MBNodeRequest)(C0) Node answering NodeStatus request value[0]=4 value[1]=0 value[2]=0 value[3]=1 value[4]=0 value[5]=1 value[6]=1 value[7]=25
(I) (printStatus)(C0) STATE: COMM_B Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 25.0 A, MainsMeter.Timeout=9, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: -6.8 A L2: -10.2 A L3: -10.2 A Isum: -27.2 A
(V) (ModbusDecode)(C0) Received packet (29 bytes) 09 10 00 20 00 0b 16 00 a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff bb ff 9a ff 99
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x09 Function 0x10 Register 0x0020 Request
(V) (MBbroadcast)(C0) Broadcast received, Node 0.0 A, MainsMeter Irms L1=-6.9A,L2=-10.2A,L3=-10.3A,
(I) (printStatus)(C0) STATE: COMM_B Error: 0 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 25.0 A, MainsMeter.Timeout=10, EVMeter.Timeout=64.
(I) (printStatus)(C0) L1: -6.9 A L2: -10.2 A L3: -10.3 A Isum: -27.4 A
(V) (ModbusDecode)(C0) Received packet (6 bytes) 02 04 00 00 00 08
(V) (ModbusDecode)(C0)  valid Modbus packet: Address 0x02 Function 0x04 Register 0x0000 Request
(D) (MBNodeRequest)(C0) Node answering NodeStatus request value[0]=4 value[1]=0 value[2]=0 value[3]=1 value[4]=0 value[5]=1 value[6]=1 value[7]=25
(V) (fn_mqtt)(C1) 218 CREATED
(fn_mqtt)(C1) 218 ERROR  DNS lookup failed
```

Only when the power sharing is disabled, then the EVSE will start charging.

This is my setup:
```
{
  "version": "v3.6.10",
  "serialnr": 6794,
  "mode": "SMART",
  "mode_id": 3,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "Garage",
    "rssi": -59,
    "bssid": "56:D9:E7:49:F1:55"
  },
  "evse": {
    "temp": 28,
    "temp_max": 65,
    "connected": true,
    "access": true,
    "mode": 1,
    "loadbl": 2,
    "pwm": 1024,
    "solar_stop_timer": 0,
    "state": "Request State B",
    "state_id": 4,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 0,
    "override_current": 0,
    "current_min": 6,
    "current_max": 25,
    "current_main": 25,
    "current_max_circuit": 32,
    "current_max_sum_mains": 600,
    "max_sum_mains_time": 0,
    "solar_max_import": 0,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Not present",
    "mains_meter": "Sensorbox",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0
  },
  "mqtt": {
    "host": "homeassistant.local",
    "port": 1883,
    "topic_prefix": "SmartEVSE-52371",
    "username": "mqqt",
    "password_set": true,
    "status": "Disconnected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Disabled",
    "address": 100,
    "import_active_power": 0,
    "total_kwh": 0,
    "charged_kwh": 0,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": -231,
    "L1": -55,
    "L2": -92,
    "L3": -84,
    "last_data_update": 0,
    "original_data": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    }
  },
  "backlight": {
    "timer": 0,
    "status": "OFF"
  }
}
```

The slave does get the current power data from the sensorbox, this is displayed on the device and shown on the webpage.

I checked the wiring from the EVSE to the charge cable (using fixed cable). Both the earth and CP are connected properly.
When the cable is disconnected, 12 V can be measured between earth and CP.

I'm running version 3.6.10 on both node 1 and master.

dingo35, 2024-09-24:
Ok I cannot reproduce your problem, so I need more info. 

1) please check your modbus wiring between Master and Slave and Sensorbox; I don't think the problem is there, but I just to rule wiring problems out.
2) could you put Master and Slave in Normal mode, have a telnet session running on both devices and start charging on the slave (or at least try to);
3) please post the /settings from that same session, for both devices .

4) reverse the role of the devices: make your Master temporarily Node 1, and Node 1 temporarily Master; is the behaviour the same?

JopGeraets, 2024-09-27:
The problem was indeed in the modbus wiring.
The B cable from the master to the slave was loose.
After fixing this everything works

Thanks a lot, for the help!

closed 2024-09-27 by JopGeraets