dingo35/SmartEVSE-3.5 #128: Master and Node on seperate modes (smart/solar)

JopGeraets, 2024-09-27:
At the moment when the master or node changes from smart to solar or from solar to smart, then the other EVSE will follow and also have this mode.
Thus as far as I'm aware it is not possible to have the master and node on different modes (only considering the smart and solar modes).

What I'd like to achieve is for 1 EVSE to be in Solar mode, while the other EVSE can be in smart mode.

A scenario that this can be usefull in is when:
One EVSE is in solar mode and the car connected does not yet need extra energy (Just excess solar energy is enough). While the other car does need to be charged immidiatelly and thus needs more energy then solar can provide.

dingo35, 2024-09-27:
Why dont you disable power sharing, and operate them in different modes?
You could either give both of them a separate fuse or have and EVmeter installed and have it guard MaxCircuit?

JopGeraets, 2024-11-04:
Sorry, I didn't think about the option of turning power sharing off.
Then this is of course possible

Thanks for the feedback!

closed 2024-11-04 by JopGeraets