dingo35/SmartEVSE-3.5 #129: Connect EV indicator (because cheap tariffs are ahead)

jonesPD, 2024-09-30:
**Is your feature request related to a problem? Please describe.**
We have a dynamic electricity contract, which means our charging tariffs vary significantly (usually between 15Ect/kWh and 30 Ect/kWh, sometimes beyond). Whenever tariffs are low (e.g. raw price around or below zero), I would like to charge our EV. It would be great if the SmartEVSE charger can indicate that cheap tariffs are ahead (in time) by showing a (for example) blue LED switch when we arrive home (and the EV is not yet connected and access is blocked). My home domotica system already knows the dynamic prices and can set a flag via MQTT (and possibly REST), so SmartEVSE code can stay simple. 

**Describe the solution you'd like**
I'd like the SmartEVSE Switch/LED to show a blue color when I'm arriving home and the CheapTariffIndicator has been set by my home domotica system. Afaik, blue is not used to indicate anything yet (but open to improvements). 
MQTT proposal

- smartEVSE/Set/CheapTariffIndicator: unix time
- smartEVSE/CheapTariffIndicator 

Whenever present time is before the _CheapTariffIndicator_, the LED turns blue when no EV is connected and the SmartEVSE is blocked. 

**Describe alternatives you've considered**
Look at the Electricity prices app (e.g. Tinder, Coolblue, etc) is a bit of a habit for me, but not for my spouse...

**Additional context**
[Check out my psuedo-code additions in my own fork of this repository](https://github.com/jonesPD/SmartEVSE-3.5/commit/056035e7020937c5aa54667842105b1614b9c9c9)

Note: I'm not a github expert, usually don't program in C++. The modified code in my repository compiles ok, but I haven't tried it, because I would really appreciate @dingo35 feedback and suggestions on my modifications.

Imaginous, 2024-09-30:
Maybe a MQTT command to set a custom RGB color to set the LED color in each possible mode: normal, smart, solar, off.

jonesPD, 2024-10-01:
I'm not sure if we should change/override colors while charging, because those are listed in the manual. But a generic MQTT/REST command to set the color in Off mode is also a possibility. 
What do you think about the time aspect I proposed above? The unix time which indicates the time until the color is set, before reverting back to the standard color (Off in my proposal).

Imaginous, 2024-10-01:
I think external software should set the color based on "whatever", time in your case. It shouldn't be logic in the SmartEVSE.

dingo35, 2024-10-01:
I dont understand why you would task the end user with this; if your domotica system signals low prices coming up, just let your wifi plug in the EV as always, and let the domotica system set a delayed charge through the API?

jonesPD, 2024-10-01:
My domotica is set up to always charge the EV when it is plugged in, using the lowest rates before the next time it will be used (usual departure time is 7am). However, it is not necessary to plug in and charge every day - the car has enough range to charge every few days. We don't want to charge when it is not needed and the prices are (relatively) high. However, we do want to charge when the prices are low. 
For our EV, there is no easy API to read the SOC from, and even if that were available, the domotica doesn't know what the required range is for the next day. Hence, the approach to always charge when plugged in. 
Example: This Monday the car had plenty of range left, but the overnight raw prices dropped to zero Ect/kWh - therefore it is advantageous to charge overnight. 

Does that clarify?

mstegen, 2024-10-01:
I'm afraid this is too specific. 
Although i do like the idea of customizable LED colours. Some people might want to mimic a public charging stations status led colors.

jonesPD, 2024-10-01:
OK, good feedback here. Let's generalize the proposal in a way that people can control the color of the LED (and I can control it in the way that I'm thinking off). Do you propose to make the color configurable during all modes? Or limit it to a controllable Off mode color?

jonesPD, 2024-10-01:
@devdems : I don't have access to the SOC and I can't configure the car to stop charging at a SOC %%, but you are right, there are more solutions to my problem. :-)

Anyway, I've implemented a more generic approach to set the color of the LED in Off mode in [my fork](https://github.com/jonesPD/SmartEVSE-3.5/commit/053ea99e6a776ed0ac2d689cb4bd8c2500230cae). You can send the RGB color in off mode via MQTT (and it will publish the setting via MQTT). It compiles, and it works. I left out the integration in the web UI and /settings json for the time being. 

Welcome to feedback and improvements.

devdems, 2024-10-01:
Setting up custom RGB sounds useful. I can see it useful to set it up so it notifies me if I'm over my max kW in the tariff block before I run any heavy equipment in garage.

We just need to be careful that the change has no impact on the "error" RED light which needs to be prioritized.

jonesPD, 2024-10-01:
Agreed, you can only modify the color of the 'Off' mode: (Access_bit == 0 || State == STATE_MODEM_DENIED)
These colors default to 0,0,0 (off). Since I only modify the off mode color, none of the other colors should have been affected.

dingo35, 2024-10-01:
@jonesPD I'm trying to keep the REST API and the MQTT API in sync; could you add your mod to the REST API and make a PR out of it?

devdems, 2024-10-01:
If you ask me I would leave the option available on all modes (Off, Normal, Solar and Smart). If it's only on "Off" then you are limiting the use-cases where this could be useful like in the example that @mstegen wrote. For example, I never have the charger in Off mode but always in Smart. 

But if it's "error" it should always blink red regardless of what you are sending to the charger via REST/MQTT.

jonesPD, 2024-10-02:
@devdems : I can prepare a proposal with similar endpoints to control the color in Normal, Solar and Smart modes.

jonesPD, 2024-10-03:
[Pull request added](https://github.com/dingo35/SmartEVSE-3.5/pull/131).

I'll create a proposal for similar endpoints for the normal/solar/smart modes. 
Suggestions/ideas welcome.

jonesPD, 2024-10-05:
[Additional pull request for the Normal/Smart/Solar modes](https://github.com/dingo35/SmartEVSE-3.5/pull/132).

Elektrofix-OL, 2024-10-06:
I solved this for myself with ioBroker (like Homeassistens, NodeRed,...). He knows when the electricity tariff is cheap.
Normally the SmartEVSE is in solar mode, when the price is below a certain value the ioBroker sends an API command to switch to smart mode, if the value increases there is an API command for solar mode.

This way I can see when I get home whether the electricity is cheap or not.

closed 2024-10-14 by jonesPD