dingo35/SmartEVSE-3.5 #130: current_max_sum_mains = 0 but the EV is stil charging

devdems, 2024-10-01:
There seems to be an issue with current_max_sum_mains. Not sure in which version this was introduced. As you see that current_max_sum_mains is 0 but the charger is still charging.

{"version":"v3.6.10","serialnr":6946,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"MATRIX","rssi":-81,"bssid":"02:0C:42:8C:F7:A7"},"evse":{"temp":29,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":230,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":135,"override_current":0,"current_min":6,"current_max":20,"current_main":25,"current_max_circuit":20,"**current_max_sum_mains**":0,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"10.0.0.10","port":1883,"topic_prefix":"SmartEVSE-45332","username":"","password_set":false,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":653,"L1":207,"L2":249,"L3":197,"last_data_update":1727806020,"original_data":{"TOTAL":653,"L1":207,"L2":249,"L3":197}},"backlight":{"timer":120,"status":"ON"}}

[image]

dingo35, 2024-10-01:
I think that 0 = disabled, what happens if you set it to 1?

devdems, 2024-10-01:
My bad. I checked later and saw that 0 means that it is disabled. If I set it to 1 it looks like it stops charging so I will just change the payload in HA from 0 to 1. Tnx.

closed 2024-10-01 by devdems

dingo35, 2024-10-01:
Did you have that v3.6.10plus version in github, or was that someone else?

devdems, 2024-10-01:
No, i have your normal latest release v3.6.10.