dingo35/SmartEVSE-3.5 #133: No error display for illegal statuses / status E faulty function

Elektrofix-OL, 2024-10-07:
**Describe the bug**
I had the opportunity to test my SmartEVSE with a test device and found the following issues:
Status C+E: -12V on the CP pin becomes smaller (diode error detection) Charging does not stop

**Upload your config**
`{"version":"v3.6.10","serialnr":6622,"mode":"SMART","mode_id":3,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"iot","rssi":-52,"bssid":"6A:D7:9A:24:A1:8D"},"evse":{"temp":28,"temp_max":65,"connected":false,"access":true,"mode":1,"loadbl":0,"pwm":1024,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":20,"current_main":50,"current_max_circuit":25,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":1,"enable_C2":"Solar Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.5.2","port":1883,"topic_prefix":"SmartEVSE","username":"Lehm-WG","password_set":true,"status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":11,"import_active_power":0,"total_kwh":734.0999756,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":734.0999756,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":6,"L1":-13,"L2":6,"L3":13,"last_data_update":1728319380,"original_data":{"TOTAL":6,"L1":-13,"L2":6,"L3":13}},"backlight":{"timer":113,"status":"ON"}}`

**Upload a debug log**
(I) (EVSEStates)(C1) Cable limit: 20A  Max: 20A
(D) (IsCurrentAvailable)(C1) Current available checkpoint D. ActiveEVSE increased by one=1, TotalCurrent=0.0A, StartCurrent=6A, Isum=1.0A, ImportCurrent=0A.
(C1) 19:20:46 STATE A -> B
(C1) Switching Contactor1 OFF.
(C1) Switching Contactor2 OFF.
(C1) Diode OK
(D) (Timer100ms)(C1) ModbusRequest 4: Request Energy Node 0
(D) (Timer100ms)(C1) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(D) (CalcBalancedCurrent)(C1) Checkpoint c: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=25.0A, saveActiveEVSE=0.
(I) (printStatus)(C1) STATE: B Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 1.7 A IsetBalanced: 25.0 A, MainsMeter.Timeout=3, [0mEVMeter.Timeout=64.
(I) (printStatus)(C1) L1: -1.3 A L2: 0.6 A L3: 1.7 A Isum: 1.0 A
(D) (IsCurrentAvailable)(C1) Current available checkpoint D. ActiveEVSE increased by one=1, TotalCurrent=0.0A, StartCurrent=6A, Isum=1.2A, ImportCurrent=0A.
(D) (CalcBalancedCurrent)(C1) Checkpoint c: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=25.0A, saveActiveEVSE=0.
(C1) 19:20:48 STATE B -> C
(C1) Switching Contactor1 ON.
(C1) Switching Contactor2 ON.
(D) (Timer100ms)(C1) ModbusRequest 4: Request Energy Node 0
(D) (Timer100ms)(C1) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.4 A IsetBalanced: 25.0 A, MainsMeter.Timeout=10,[0m EVMeter.Timeout=64.
(I) (printStatus)(C1) L1: -1.8 A L2: 0.6 A L3: 2.4 A Isum: 1.2 A
(D) (Timer100ms)(C1) ModbusRequest 4: Request Energy Node 0
(D) (Timer100ms)(C1) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(D) (Timer100ms)(C1) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.4 A IsetBalanced: 25.0 A, MainsMeter.Timeout=8, [0mEVMeter.Timeout=64.
(I) (printStatus)(C1) L1: -1.8 A L2: 0.6 A L3: 2.4 A Isum: 1.2 A
(D) (Timer100ms)(C1) ModbusRequest 4: Request Energy Node 0
(D) (Timer100ms)(C1) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(D) (Timer100ms)(C1) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(I) (printStatus)(C1) STATE: C Error: 0 StartCurrent: -6 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 2.4 A IsetBalanced: 25.0 A, MainsMeter.Timeout=6, [0mEVMeter.Timeout=64.
(I) (printStatus)(C1) L1: -1.8 A L2: 0.6 A L3: 2.4 A Isum: 1.2 A
(C1) 19:20:53 STATE C -> A
(C1) Switching Contactor1 OFF.
(C1) Switching Contactor2 OFF.
**Status E was triggert 19:20:50**

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in?  *Smart*
2. What steps do you take so the problem arises?
*Simulation with the tester:
Switch 1: Vehicle plugged in = status B 
Switch 2: Request charge = status C
Switch 3: Short circuit of the diode = status E
===> charging process is not interrupted!*

**Expected behavior**
If a status E occurs, I expect the charging to be switched off, as if there was an RCM error.

Elektrofix-OL, 2024-10-15:
Yes, it seems to be twice as much, but we Germans seem to interpret the examination guidelines more strictly. Please give me until the weekend to check whether the DGUV complies with the IEC 61851-1 standard or whether something has changed.
If everything is OK, I will then close the issue.

Elektrofix-OL, 2024-10-20:
I bought the EN IEC 61851-1: The current one is from 2019 and has been corrected twice (last correction april 2024).
In Table A.4, _Detected by the electric vehicle power supply device
System states_  the footnote in state C2/D2 is:
```
The electric vehicle power supply device shall support the low level of the PWM signal of −12 V, the presence
the diode, before closing the supply switching device on the power supply device for electric vehicles at least
check once.

```
There is more detail in the A4 _test procedure area_
```
A4.9 Pilot Voltage Short Circuit Test states:
This test verifies compliance with the timing requirements in Table A.6, Procedure 12
checked.
The test begins with the nominal values ​​of the resistors R2test and R3test. If condition C or D
lasted at least 5 s, an additional resistor of 80 Ω is used as a connection between the
Pilot conductor and protective conductor are connected.
```
Diagramm from Picture of Tabel A6. Procedure 12
Curve 1: PWM signal; Curve 2: Signal at the contactor; Curve 3: Auto charging requirements; Curve 4: Power consumption from the car
(Sorry picture in German)
[image]

If there is a disturbance in the PWM signal, the SmartEVSE must switch off the contactor c1 within 3 seconds

In my opinion, there is an issue with SmartEVSE

mstegen, 2024-10-20:
Hello @Elektrofix-OL 
The SmartEVSE checks before closing the contactor, once if the low level of the PWM signal is there. (so it checks for the diode) As required by the standard. 

But you are right, once it's charging it should also detect if the signal is disturbed somehow (shorted to ground for example).
I will add the check for a shorted or otherwise disturbed signal to the code.

Thanks for bringing this to our attention!

Elektrofix-OL, 2024-10-21:
I tested it: failed

Test 1: 
Switch 1: Vehicle plugged in = status B
Switch 2: Request charge = status C
Switch 3: Diode short circuit = status E
===> Charging process is not interrupted!
Switch 2: OFF = status B&E
Delay 2 sec.
Switch 2: Request charge
===> No Charging = OK

Test 2: 
Switch 3: Diode short circuit = status E
Switch 1: Vehicle plugged in = status B
Switch 2: Request charge = status C
===> No Charging = OK

Test 3:
Switch 1: Vehicle plugged in = status B
Switch 3: Diode short circuit = status E
Switch 2: Request charge = status C
===> Charging starts even though diode is short-circuited! 
Short circuit 0 ohms between CP and PE
==> Loading stops
Short circuit removed
===> Charging starts automatically! Dangerus!!

If such an error occurs, it would be good if this error were displayed and had to be confirmed as with the RCM error. Automatic connection is dangerous.
I only had a normal multimeter for this test, so unfortunately I can't see what the voltages were. I checked my tester after the test, diode and resistance values ​​were OK.
I hope my results help you improve your software.

mstegen, 2024-10-21:
@Elektrofix-OL 
Diode short circuit is not the same as state E.
State E is 0V so the pilot signal at GND level or floating (not connected).

Test 1:
The diode test is performed immediately after entering state B, so if you short the diode while in State C it will not be able to detect it. This is however still according to spec, as the test was performed before entering State C.

Test 2:
As your second test shows, if you start with a shorted diode it will not be possible to switch to state C.

Test 3:
See above, the diode test was already performed before you short circuited it. So it was able to enter State C.
When you place a short between CP and PE, it will stop, and switch back to State B 
When you remove the short, it **will** immediately do a diode test again, and if the diode is detected switch to State C.
But if you leave the diode shorted, it will not switch to State C, but remain in State B.

The short circuit test mentioned in A.4.9 was not detected previously. My IEC_61851-1 uses a resistor of 120 Ohm between PE and CP, i have tested 0 / 80 / 120 Ohm and all of these values now successfully interrupt charging.

I don't think we should stop with an error that requires user interaction.
As the diode test is always performed, it's safe to resume charging.

Elektrofix-OL, 2024-10-21:
> Diode short circuit is not the same as state E. State E is 0V so the pilot signal at GND level or floating (not connected).

Yes, I understand status E = CP -1V to 1V and status F -13V to -11V, they are different. It is not stated in EN 61851 that it has anything to do with the diode. I just realized that.

> Test 1: The diode test is performed immediately after entering state B, so if you short the diode while in State C it will not be able to detect it. This is however still according to spec, as the test was performed before entering State C.

@mstegen  I'm not entirely convinced. It would be better to test the diode immediately before changing from status B to C and not so much "minutes" before. A car that balances the cells stays in B and later switches back to C to recharge.

> Test 2: As your second test shows, if you start with a shorted diode it will not be possible to switch to state C.

That's how it should be

> Test 3: See above, the diode test was already performed before you short circuited it. So it was able to enter State C. When you place a short between CP and PE, it will stop, and switch back to [...]

I did my test 3: three times in a row this afternoon and the error was always there: switching on despite a diode short circuit.  Then I briefly installed V3.6.10 and then again the one from yesterday. Now I can't recreate the error in test 3.
Confirmed: After the short circuit the diode check is there and only when I remove the diode short circuit does it switch on again.

> I don't think we should stop with an error that requires user interaction. As the diode test is always performed, it's safe to resume charging.

I agree with you that the diode check shouldn't necessarily lead to a user action, but the short circuit between CP and PE should. In addition, a short circuit between CP and PE always indicates a defective charging cable or defective electronics in the car. Ideally, this shouldn't happen, but my experience shows that people are not careful with cables and tend to drive over them and damage the wires. Forklift drivers are particularly pain-free.

closed 2024-10-21 by Elektrofix-OL

mstegen, 2024-10-21:
I'll see if i can move the diode check to just before switching to State C. or add a second diode check. 

Again thanks for bringing the short circuit detection between CP and PE to my attention.