dingo35/SmartEVSE-3.5 #139: enable OCCP +EvMeter

XsniferX, 2024-10-25:
hi please how to enable OCCP ?

i have an wem3080TR from iammeter.

i read on a previous release note that occp has been removed from the lcd and available in the web ui.

but i cannot found it. it is necessary to have the rfid module ? i wanted to buy it but its unavailable (some one know a rfid reader alternative ?)

also i do not realy understand the "/POST /evmeter" i have an esp32 with a jSY-MK-194T that update this api and also the mainsmeter but i do not quite understand the purpose of "import_active_energy, export_active_energy and import_active_power" those value should be what the car is importing back to house ? or it should be data from my solar panel for me the evmeter is a ct clamp on the phase of my type 2 cable so i do not understand the three energy and power.

Regards,

dingo35, 2024-10-25:
For OCPP you should compile the spurce code yourself with the -e ocpp option added the compile command line. The RFID reader is not necessay for OCPP.

If you are measuring your currents at the connection of the grid to your house, you should configure this device as your MainsMeter.
If the device is measuring the currents of your EV charging, you should configure it as an EVMeter.

import_active_energy, export_active_energy and import_active_power are just relays of the respective values you posted through the API, if any.

Your use of language is unclear. Please use proper punctuation and Google Translate to improve our communication.

closed 2024-10-25 by dingo35

XsniferX, 2024-10-25:
Hi thank you for your quick reply.

sorry, i'll try to be more clear/precise.

So bassically i have an iammeter WEM3080TR (https://www.iammeter.com/products/three-phase-meter)

I have a contract with12 KVA single phase.

Clamp A is monitoring the phase from the grid.
Clamp B is monitoring a 32A EV wallbox.
Clamp C is monitoring the AC phase coming from my solar panels.

So with an esp32 running esphome i'm doing http_request.GET for the clamp A (Grid phase).

then i do http_request.POST of the values of the Grid clamp directly to Smart EVSE via the api (POST: /currents) so those values feed the MAINSMET every 1s.

Now for EV METER i have a jSY-MK-194T with one clamp (lets call it clamp D.) wrap around the phase of my Type 2 cable.

I then do http_request.POST (/ev_meter) to feed the SMART EVSE every 1s.

But furthermore in the api for the ev_meter we can populate "import_active_energy, export_active_energy and import_active_power "

what are those values ? import_active_energy should be the energy taken from the grid (clamp A) in Wh ?

and what is the definition of the two other please ?

should those value need to come from the car (Clamp D)  ?

or a combination of clamp A (grid) and clamp C (solar grid) ?

regards,

dingo35, 2024-10-25:
Clamp A is MainsMeter,  clamp D would be EVMeter. Usually on the EVMeter there will be no energy exported to the grid, so you can ignore that variable.

Active Power is the power in W (if your meter provides that).

XsniferX, 2024-10-26:
thanks your for your time !!