dingo35/SmartEVSE-3.5 #142: Mqtt discovery warning on 3.7.0

hmmbob, 2024-10-26:
**Describe the bug**
HA warns on an invalid discovery configuration. 

**Upload your config**
.

**Upload a debug log**
```
Error 'The option `options` can only be used together with device class `enum`, got `device_class` 'None'' when processing MQTT discovery message topic: 'homeassistant/sensor/SmartEVSE-6360-OCPP/config', message: '{'object_id': 'SmartEVSE-6360-OCPP', 'unique_id': 'SmartEVSE-6360-OCPP', 'state_topic': 'SmartEVSE-6360/CustomButton', 'availability_topic': 'SmartEVSE-6360/connected', 'device': {'model': 'SmartEVSE v3', 'identifiers': 'SmartEVSE-6360', 'manufacturer': 'Stegen', 'configuration_url': 'http://192.168.88.216', 'sw_version': 'v3.7.0', 'name': 'SmartEVSE-6360'}, 'command_topic': 'SmartEVSE-6360/Set/CustomButton', 'options': ['On', 'Off'], 'name': 'OCPP', 'platform': 'mqtt'}'
Error 'The option `options` can only be used together with device class `enum`, got `device_class` 'None'' when processing MQTT discovery message topic: 'homeassistant/sensor/SmartEVSE-6360-OCPPConnection/config', message: '{'object_id': 'SmartEVSE-6360-OCPPConnection', 'unique_id': 'SmartEVSE-6360-OCPPConnection', 'state_topic': 'SmartEVSE-6360/CustomButton', 'availability_topic': 'SmartEVSE-6360/connected', 'device': {'model': 'SmartEVSE v3', 'identifiers': 'SmartEVSE-6360', 'manufacturer': 'Stegen', 'configuration_url': 'http://192.168.88.216', 'sw_version': 'v3.7.0', 'name': 'SmartEVSE-6360'}, 'command_topic': 'SmartEVSE-6360/Set/CustomButton', 'options': ['On', 'Off'], 'name': 'OCPPConnection', 'platform': 'mqtt'}'
```

**To Reproduce**
Reboot the SmartEVSE 

**Expected behavior**
no error

dingo35, 2024-10-26:
Thx for your report, I am currently not able to test with HomeAssistant but this should solve it:
[0816bd2ac00.zip](https://github.com/user-attachments/files/17531505/0816bd2ac00.zip)

hmmbob, 2024-10-27:
It does!

closed 2024-10-28 by dingo35