dingo35/SmartEVSE-3.5 #144: Bricked after update to 3.7.1

Paefi, 2024-10-28:
**Describe the bug**
After updating from version 3.7.0 to version 3.7.1 the EVSE seems bricked. There is not any response at all 

**Upload your config**
Impossible as i cannot access the EVSE

**Upload a debug log**
Flash the debug version of the firmware (see https://github.com/dingo35/SmartEVSE-3.5/blob/master/SmartEVSE-3/HowToFlash.txt), telnet to your device, capture the debug log, cut it so that it shows JUST BEFORE and JUST AFTER the problem arises, and upload it here.

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar?
2. What steps do you take so the problem arises?

**Expected behavior**
Normal reboot
**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

Paefi, 2024-10-28:
To be complete: There seems to be no way to access the EVSE anymore. Display stays empty, pressing buttons has no effect. To my assurance i checked the current on the EVSE, that's OK, Powering down for some time and powering up also has no effect.

cow77, 2024-10-28:
Same here. 
But access to "http://evse-ip/update" still works: gives the old update page and flashing 3.7.1 is possible
So it's up again!

Paefi, 2024-10-28:
#Cow77
I tried that, but did not work. However your post made me persist in another way. I accessed the EVSE trough the micro USB bus (did not try that before but works wonderwell...) and in that way i could perform the update and have the EVSE up and running again.
So still many thanks!

closed 2024-10-28 by Paefi