dingo35/SmartEVSE-3.5 #145: EVMeter power not inverted to positive/absolute when using InvEastron.

rjmgmt, 2024-10-28:
**Describe the bug**
When using the InvEastron, the power value is shown as positive/absolute value on the SmartEVSE display, however within the WebUI/API/HA-addon it's shown as a negative value. All other related values (energy, current) are positive.

**Upload your config**
{"version":"v3.7.1","serialnr":7556,"mode":"SMART","mode_id":3,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"HKHQ Home","rssi":-35,"bssid":"BA:FB:E4:C1:E4:54"},"evse":{"temp":29,"temp_max":65,"connected":false,"access":true,"mode":1,"loadbl":0,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":0,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":15,"enable_C2":"Auto","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"","port":1883,"topic_prefix":"SmartEVSE-7556","username":"","password_set":false,"status":"Disconnected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"InvEastrn","address":16,"import_active_power":0,"total_kwh":28.79999924,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":28.79999924,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":12,"L1":2,"L2":19,"L3":-9,"last_data_update":1730124410,"original_data":{"TOTAL":12,"L1":2,"L2":19,"L3":-9}},"backlight":{"timer":0,"status":"OFF"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

**Upload a debug log**
N/A

**To Reproduce**
Use an InvEastron as EVMeter, check power value.

**Expected behavior**
It should transform the value to a positive/absolute value.

**Screenshots**
[image]

dingo35, 2024-10-28:
I am currently in an area without 230V/50Hz, so I cannot test Eastron devices for the next 5 weeks; please be patient.

PimDoos, 2024-12-19:
Just encountered the same behavior. Energy is displayed correctly, charge power is negative.

dingo35, 2024-12-20:
I cannot reproduce this:
[image]

(Note that this is a meter that is normally configured as Eastron, so it is correct that the registered totals are negative when InvEastron is configured).

You are aware that if you are feeding the Eastron from the top, you should configure "Eastron" , and ONLY if you feed the Eastron from below, you should configure InvEastron?

rjmgmt, 2024-12-20:
It's doing the same thing as we speak:
[image]

Raw data:
"ev_meter":{"description":"InvEastrn","address":16,"import_active_power":-10.30000019,"total_kwh":552.7000122,"charged_kwh":9.600000381,"currents":{"TOTAL":446,"L1":148,"L2":150,"L3":148},"import_active_energy":552.7000122,"export_active_energy":0},

And yes, i'm aware, it's fed from the bottom.

PimDoos, 2024-12-20:
Yes, I am feeding the Eastron meter from below.
[image]

When selecting 'Eastron' as the EV Meter, the value for charged energy and EV Power are displayed as negative when charging.
When selecting 'Inverted Eastron', only the charged energy value is inverted to positive. The power value stays negative.
I suspect the power value is not inverted when using the Inverted Eastron mode, as the meter itself also reads a negative value. I would expect this power value to be inverted by SmartEVSE.

rjmgmt, 2024-12-20:
Similar set-up to Pim here and also the very same SDM630 modbus v2. 

[image]

PimDoos, 2024-12-20:
I think this is the issue. Inverted Eastron meters power values are not inverted if the value reported is negative.
https://github.com/dingo35/SmartEVSE-3.5/blob/94ca08ef4b211b9c34717bbf46912cc2d5a77706/SmartEVSE-3/src/meter.cpp#L266

e.g. for read value of -2400 W this statement works out as follows:
-2400 is not greated than 0, so do not invert the power value.

dingo35, 2024-12-20:
Ok Im testing at SDM72D-M v2 here, so that might explain the different outcomes...

Could you guys test v3.6.6,  there were some major changes the code between  v3.6.6 and v3.6.7 ?

rjmgmt, 2024-12-20:
Just tried, but I can't seem to be able to downgrade to 3.6.6 through the UI from 3.7.2. The upload of firmware.bin just stops. 

I'll try again tomorrow.

PimDoos, 2024-12-21:
Found the bug and got it working locally with the fix implemented. I'll open a PR to merge.
[image]

rjmgmt, 2024-12-21:
Thanks for the fix @PimDoos. I just tried to test with the old firmware again, but i guess the downgrade is too much of a step.

PR #162 (2024-12-22): Invert power measurement when receiving currents for Inverted Eastron
Fixes #145 
Fixes #146 

When receiving currents for some meters, SmartEVSE also calculates the power measurement. This power measurement was not properly inverted for Inverted Eastron meters, causing a negative power readout.

Before (Firmware 3.7.2)
[image]

After this change:
[image]

Currents are also still measured correctly after this change.

I verified this with an Eastron SDM630-Modbus V2.
commit 5edfa67b (2024-12-22): Invert power measurement when receiving currents for Inverted Eastron (#162)

* Only invert if power measured is negative (Fix #145)

* Invert power for Inverted Eastron while receiving currents

* Change LCD formatting to match datatype for EVMeter measurements
closed 2024-12-22 by dingo35