dingo35/SmartEVSE-3.5 #147: bricked when flashed to 3.7.1

XsniferX, 2024-10-28:
``hello, 

i think i got the smart evse bricked.

i was recently on 3.7.0(debug version)  and then via the web ui and the update button i chose the latest 3.7.1 debug.

at the botom of the page it was stuck for more than 10 min on the bytes (like 99993443/99993443)

the lcd of the Smart evse where frozen and not responding. i did a power cycle but the lcd was briefly flashing like ok it is powered.  but then goes dark (like the power were off). i try the unbrick topic (post #79).  the lcd is now always on but empty. just the backlight is on.

i then did the serial connection and it always showing that :

`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
`

Regards,

dingo35, 2024-10-28:
Just surf to http://your-smartevse/update and update via "choose file".

dingo35, 2024-10-28:
Also see #144

XsniferX, 2024-10-28:
hi, sorry to reopen the issue but i have no way of accesss of the smart evse. it was on 192.168.32.79 but the ip is not responding anymore so nothing happen. 

[image]

dingo35, 2024-10-29:
https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/configuration.md#i-think-i-bricked-my-smartevse

dingo35, 2024-10-29:
Although the picture you send is showing a responding /update page..... DO NOT PUSH the flash buttons but pick "choose file" (bottom of page) and select a firmware.bin file you extracted from a downloaded .zip file from the releases page of this repo

dingo35, 2024-10-29:
If you uploaded an irrelevant page, use the flash tool that is mentioned in the doc link above...

XsniferX, 2024-10-29:
i it possible to have a proper instructions/how to  upload the firmware to the device plz ? 

i tried those 3 : 

[image]

[image]

[image]

all finish without error but the screen its still empty and not responding to a ping.

Regards,

dingo35, 2024-10-29:
I cant see any errors in the screens you uploaded.

What always works:
1 install platformio on your windows, 
2 git clone <myrepo>
3 goto directory where platformio.ini resides
4 connect via USB cable
5 pio run -t upload

This wil flash the bootloader, partition table and firmware to your device, so 100% result guaranteed...

dingo35, 2024-10-29:
Ad 1:
https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#local-download-macos-linux-windows

XsniferX, 2024-10-29:
> I cant see any errors in the screens you uploaded.
> 
> What always works: 1 install platformio on your windows, 2 git clone 3 goto directory where platformio.ini resides 4 connect via USB cable 5 pio run -t upload
> 
> This wil flash the bootloader, partition table and firmware to your device, so 100% result guaranteed...

AWESOME MATE !!!

thanks you ! that bring back the evse to life.

dingo35, 2024-10-29:
Pleasure!

closed 2024-10-29 by dingo35