dingo35/SmartEVSE-3.5 #153: While charging on mode=solar/1phase/C2=auto, SmartEVSE wrongly detects its charging on 3Ph

rjmgmt, 2024-11-11:
**Describe the bug**
While charging on mode=solar/1phase/C2=auto, SmartEVSE wrongly detects its charging on 3phase while its obviously not and tries to switch back from 3Ph>1Ph, creating a loop where it keeps starting on 1Ph on solar, charges properly for about 5 mins, then wrongly detects its on 3Ph, even though the C2 contactor is and has always been OFF and EVmeter reports nothing but 0.0A on both L2 and L3, the evse then switches both contactors off, restarts on 1Ph on solar availability, and repeats this cycle forever (until my car (eGolf '19) decides its been enough, reports a charger error and stops charging)

**Upload your config**
{"version":"v3.7.1","serialnr":7556,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"xxxx","rssi":-30,"bssid":"xxxx"},"evse":{"temp":30,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":0,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Charging Stopped - Waiting for Solar","state_id":9,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":0,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":20,"enable_C2":"Auto","mains_meter":"Eastron3P","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"","port":1883,"topic_prefix":"SmartEVSE-7556","username":"","password_set":false,"status":"Disconnected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"InvEastrn","address":16,"import_active_power":0,"total_kwh":74.40000153,"charged_kwh":4.300000191,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":74.40000153,"export_active_energy":0},"mains_meter":{"import_active_energy":265.2000122,"export_active_energy":43.5},"phase_currents":{"TOTAL":45,"L1":17,"L2":17,"L3":11,"last_data_update":1731335098,"original_data":{"TOTAL":45,"L1":17,"L2":17,"L3":11}},"backlight":{"timer":120,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

**Upload a debug log**
[verbose_log_C2_20241111-1406.txt](https://github.com/user-attachments/files/17703699/verbose_log_C2_20241111-1406.txt)

**To Reproduce**
See describe the bug.

**Expected behavior**
It should not detect its on 3Phase when its not. And -if- its on 3Ph, it should only switch back to 1Ph once.

**Screenshots**
N/a

PimDoos, 2025-03-19:
This might be related, but I experienced something else is up with C2=Auto mode. 
When I set C2=Auto and start a solar charge with Grid = 0;0;0 and home battery at 12A, the charge starts with C2 enabled.
But: With minimum current set to 6A, 3-phase charging should mean at least 18A should be available before switching C2 on.
Furthermore, the SolarStopTimer does not run while C2=Auto, so in solar mode the EVSE keeps charging at 4 kW (3x6A) and does not stop or turn off C2.
Also, when changing C2 to SolarOff briefly to disable the contactor, then switching to C2=Auto, the SolarStopTimer also does not run and keeps charging at 1.4 kW (1x6A).

I think it should work like this when starting a Session

1. if AvailableCurrent > SolarStartCurrent: Enable C1 and start charging
1. and if AvailableCurrent > 3xMinCurrent: also Enable C2

And during the session
1. If AvailableCurrent < 3xMinCurrent: Start timer to switch C2 off
1. If AvailableCurrent < MinCurrent: Start timer to stop charging and switch off C1 as well (like the behavior in C2=SolarOff)

Basically add a SolarStopTimer to switch between single and three-phase charging when C2=Auto, and make sure SolarStopTimer works when using three-phase charging.

dingo35, 2025-03-19:
The C2=Auto setting is experimental, and it is known to have bugs.
If you want to charge reliably on 1 phase set C2 to "Always_Off".

PimDoos, 2025-03-19:
Is this in the works in the newer codebase? Otherwise I might take a crack at improving it when I have the time, as being able to switch to 3-phase when my Solar power delivers over 4 kW (and back again) would be a nice to have for me :)

dingo35, 2025-03-19:
Not really, it would be greatly appreciated if you would improve this part of the code!! The main4 branch is concentrating on the SmartEVSE v4 code.

Please aim your Pull Requests also at the main4.cpp branch, so they will arrive in the future releases without extra coding!

dingo35, 2025-03-19:
Current problems are: 
-there are two party overlapping detection/guessing routines
-in Auto config only switching from 3f -> 1f, never switching back.

We think you can only reliably detect phases with an EVmeter installed, and that if no reliable detection is there it should default to 3 phase charging, because all newer houses and EVs seem to default to that...

PimDoos, 2025-03-19:
Yeah I noticed the code for this is a bit scattered and needs some cleanup and refactoring. I'll look into it!

rjmgmt, 2025-03-20:
Thanks for looking into this @PimDoos. I can't code to save my life but I'd by happy to help test any fixes, I have both a 2-phase and a 3-phase EV. I'd love to have the option to "switch up" to 3F too.

eavyon, 2025-03-28:
Adding myself because I just noticed this issue on my setup. I can test this if you want. @PimDoos  My SmartEVSE is hooked to a local evmeter for measuring. (I have also some Sessy's as battery storage which already do work well with the current logic)

Hermarcel, 2025-04-13:
Same issue here. Charging falls back to 6A after PV-power drops at the end of the day. But it does not pause charging after "STOP"-minutes.

`
{
  "version": "v3.8.0",
  "serialnr": 7532,
  "mode": "SOLAR",
  "mode_id": 2,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "****",
    "rssi": -29,
    "bssid": "D6:01:C3:73:79:F7"
  },
  "evse": {
    "temp": 34,
    "temp_max": 65,
    "connected": true,
    "access": 1,
    "mode": 2,
    "loadbl": 0,
    "pwm": 102,
    "custombutton": false,
    "solar_stop_timer": 0,
    "state": "Charging",
    "state_id": 2,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 60,
    "override_current": 0,
    "current_min": 6,
    "current_max": 32,
    "current_main": 20,
    "current_max_circuit": 32,
    "current_max_sum_mains": 0,
    "max_sum_mains_time": 0,
    "solar_max_import": 0,
    "solar_start_current": 6,
    "solar_stop_time": 1,
    "enable_C2": "Auto",
    "mains_meter": "API",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0,
    "lcdlock": 0
  },
  "mqtt": {
    "host": "***********",
    "port": 1883,
    "topic_prefix": "SmartEVSE-7532",
    "username": "******",
    "password_set": true,
    "status": "Connected"
  },
  "ocpp": {
    "mode": "Disabled",
    "backend_url": "",
    "cb_id": "",
    "auth_key": "",
    "auto_auth": "Disabled",
    "auto_auth_idtag": "",
    "status": "Disconnected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "API",
    "address": 12,
    "import_active_power": 1.399999976,
    "total_kwh": 48.40000153,
    "charged_kwh": 4.800000191,
    "currents": {
      "TOTAL": 60,
      "L1": 60,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 48.40000153,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 44,
    "L1": 54,
    "L2": -8,
    "L3": -2,
    "last_data_update": 1744568511,
    "original_data": {
      "TOTAL": 44,
      "L1": 54,
      "L2": -8,
      "L3": -2
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  },
  "color": {
    "off": {
      "R": 0,
      "G": 0,
      "B": 0
    },
    "normal": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "smart": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "solar": {
      "R": 255,
      "G": 170,
      "B": 0
    },
    "custom": {
      "R": 0,
      "G": 0,
      "B": 255
    }
  }
}
`

dingo35, 2025-05-05:
So does this also happen after a reboot? Changing the C2 configuration only takes effect at the start of a session, so if you change it during the session it has no effect....

fbloemhof, 2025-05-05:
I did not try rebooting, will try this ASAP.

fbloemhof, 2025-05-09:
Indeed, after rebooting the C2 is enabled again and it started charging at 3F on solar.

rjmgmt, 2025-05-09:
Was this fixed? Or is it deferred to a later release or something?

dingo35, 2025-05-09:
Youre right, inadvertently closed.

Im awaiting a PR from @PimDoos , so no progress.

kozmoz, 2025-05-20:
I'm trying to get a clear understanding of how everything is supposed to work. I've selected the solar option and added a second contractor. In the configuration, I've experimented with both C2 = Auto and C2 = Solar Off.

Could someone clarify how Auto mode is intended to function? I understand there are still some issues, but I'm happy to help work through them.

Also, what exactly does Solar mode mean in this context? As I understand it, it usually operates on a single phase in solar mode and switches to three phases in smart mode (C2 = Solar Off). However, I did encounter a situation where it was charging on three phases while still in solar mode. I haven’t yet tried to reproduce this behavior.

Regarding the "Stop" setting — does it mean the system continues charging for n minutes when there isn't enough solar power, or does it stop and retry after n minutes? Or is it doing both?
Additionally, why does setting it to "disabled" result in charging continuing indefinitely, rather than stopping immediately?

I'd really appreciate access to any documentation or examples that explain common scenarios and configurations.
I'm sure this information exists, but currently it seems too scattered to piece together effectively.

dingo35, 2025-05-20:
Solar mode means the charger starts charging when more then StartCurrent is delivered to the grid; it charges as much as it can as long as it doesnt exceed ImportCurrent; if the current to be drawn is less then MinCurrent the Stoptimer starts, it keeps charging at MinCurrent until the timer ends (then it stops charging) or more then MinCurrent is available (then the timer is reset and it continues charging).

All currents in SmartEVSE are PER PHASE, except StartCurrent and ImportCurrent, they are summed over all phases.
In Normal and Smart mode it means there is no difference between 1P and 3P charging, but in Solar, to know how much "room" is available, you need to know hoe many phases are used.

When C2 is set to "SolarOff" or "AlwaysOff", we know we are 1P charging; in other settings we can never be sure, so we try to estimate when an EVmeter is available. Even then it is not sure because an EV is allowed to add/remove phases when charging.
When we are not sure we are careful and assume 3P.

In C2=Auto, when we are in 3P, and there is not enough power, we end the charging session, switch C2 Off, and start 1P charging. 
The logic to switch back to 3P when enough power is available is not there.

kozmoz, 2025-06-13:
> Ok guys, I have a version here that is supposed to be able to switch back from 1P -> 3P, on top of the "normal" 3P -> 1P .
> 
> I don't have test capabilities currently so anxious to learn from your test results!
> 
> [e49ac7caa45d.zip](https://github.com/user-attachments/files/20707945/e49ac7caa45d.zip)

Is it in a public branch somewhere? I'd like to merge it with my personal branch, where I've added a large amount of data to the MQTT for easier analysis during testing.

dingo35, 2025-06-13:
My source code is still unstable so not public yet; here is a version that adds some nice extra controls on your web dashboard when in Solar mode:

[63466b88.zip](https://github.com/user-attachments/files/20725453/63466b88.zip)

My first tests show good switch from 3P -> 1P, and it keeps it nice around Isum=0A, also stops when not enough sun. I cannot test the switching from 1P -> 3P because I don't have enough solar for that ....

kozmoz, 2025-06-13:
Was only able to test for 10 minutes (someone needed the car unexpectedly). Initially, charging started with 3-phase 3P. However, it soon stopped, and the total current displayed was exactly 0A on all three phases. The system then attempted to restart charging, but it never resumed. Before it was able to start again, the grid load was 0,0,0 again for a while. It seems like the currents are somehow being reset? 

Unfortunately, it was only a very short test. I can't do any more testing today.

dingo35, 2025-06-13:
Next time, please run a telnet debug log, and post it here with your /settings, so we can trace what is happening.

I guess Ill have to up my ImportCurrent drastically to test the 3P charging code...

kozmoz, 2025-06-13:
When updating the existing master branch, I keep running into merge conflicts even though I haven't made any local changes to it. Could this be caused by another branch that was renamed to master or is there another reason this might happen, maybe force-pushed changes?

fbloemhof, 2025-06-14:
Interesting first finding: my circuit breaker tripped on connecting the car. 

After switching it back on, charging started at 3 fases (what also is weird, because one of the 3 fases wasn't working for a while...).

dingo35, 2025-06-14:
Ok first warning: I noticed that on versions newer then v3.7.4, the "choose file" update does not work (reliably); so you might be testing an old version instead of the new one you thought you just flashed!!! Working on this bug.

As to the breaker tripping, did you experience this before or is this a first?

At the moment of reflashing, was your EV connected, if yes, was it charging?

fbloemhof, 2025-06-14:
It is running: `Version: 12:58:58 @Jun 12 2025`

This is the 4th or 5th time the breaker tripped, still not clear what this is causing. At first I blamed the C2 / auto setting, but it also tripped once without charghing. 

The EV was connected when flashing but not charging.

fbloemhof, 2025-06-14:
Not a specific version, I keep it uptodate with your releases.

I know of one other that reported the same: https://gathering.tweakers.net/forum/list_message/82586536#82586536

I was planning to replace the breaker but the person on GoT reported that it did not had result. I'm also not convinced this is solely a firmware issue, probably a combination of things.

dingo35, 2025-06-14:
I suggest you run factory stable (currently v3.7.4) until you have your breaker problem solved .... It prevents us from being distracted from firmware development ...

dingo35, 2025-06-14:
Please note that if you use a switching breaker (as opposed to an old-fashioned melting fuse), "installatie-automaat", that your are supposed to limit the currents to 80% of the nominal value. So for a continuous 16A charging current you need a 20A switching fuse.

dingo35, 2025-06-14:
It also helps to buy a 10kA version instead of the normal 6kA ....

dingo35, 2025-06-14:
[17656cb02.zip](https://github.com/user-attachments/files/20737663/17656cb02.zip)

This version solves some switching bugs and display problems on the LCD screen; work in progress, alpha version, for test only!!!

kozmoz, 2025-06-14:
> I sometimes rebase my commits and then force-push them; I will try to avoid it as much as possible in the future, but it will lead to slowing me down pushing my commits :-(

Yes, it's a trade-off. I have some experimental branches and need to manually move the code to the new branch now. It's not critical, but it would be nice if the standard merge worked as expected.

kozmoz, 2025-06-14:
I did some testing today using the build from the master branch: 15:01:39 @ Jun 14 2025 (Yes I double-checked the version!)
I've attached a logfile. Since the logfile had been running for a while before the test, it's best to start reviewing it from the end.

The system started solar charging on 3 phases. However, it did not switch back to 1 phase, nor did it stop charging when the grid started importing. I was expecting it to switch to 1 phase in this situation.

The device I used for testing was configured as slave device.
 
[smartevse.txt](https://github.com/user-attachments/files/20738070/smartevse.txt)

dingo35, 2025-06-15:
Ok I should have been clearer on the limitations. I usually have 3 build/test phases:
1 power shared diabled
2 power share master
3 power share slave

I am currently at the end of 1, stsrting 2.

For this particular change, C2=Auto for a Slave, will probably not work, because it is very complex for the master and the communication protocol might need adaptation.

Also it has low prio, since current master/slave behaviour has a bigger regulation problem: if a master cannot provide enough current for itself and all charging slaves, it turns off _all_ charging, instead of switching off one or more EVs and continue charging on others. This happens in both Solar and Smart.

I have some code in calcbalancedcurrent5 branch, but it had bugs in it and the community did not show much interest in testing it, so it never made it to production...

In that light, "fine tuning" switching off separate phases of a slave seems rather ridiculous.

fbloemhof, 2025-06-15:
Charging session from this morning:

[image]

Started ~10:50 at 1F, charged +/- 5 minutes, stopped, started again at 3F for a while but stopped (clouds...) and after ~12:00 a small period on 1F but after that the EV was fully charged. 

Unsure what te dip in charge power around 11:35 is, maybe connection issues with the car.

My assumption is that you re-use the solar wait timer in seconds, is this correct. Quit clever, but I think it might be too short in some situations. But time will tell I think.

dingo35, 2025-06-15:
Thanks for your reports! Chatgpt also developed a more elaborate testscript for solar mode (see test/feed.sh) for us so nice work to be done!

PimDoos, 2025-06-15:
Can also confirm switching from 1P to 3P works, have not been able to test 3P to 1P yet.
[image]

dingo35, 2025-06-16:
@kozmoz If I would bring in the calcbalancedcurrent5 logic, which can switch off different nodes separtely when not enough power to feed them all, AND include the EnableC2 logic , would you be willing and able to test this?

The only thing not (yet) working would be the C2 = AUTO setting on slaves, all other settings would work. The AUTO setting on slaves could be the next step then...

kozmoz, 2025-06-16:
Yes, I would be willing to test it. However, it’s a bit challenging to get all the environmental variables right, as I have no control over how much sunlight there is at any given moment 😅. Could you let me know exactly what scenario you would like to see tested?

Does "C2 = AUTO" on a slave for now mean that the slave would then be set to "Solar Off"?

fbloemhof, 2025-06-17:
The visual counter, but the configuration not afaik. I had it configured at 30, what resulted in 1800 seconds.

fbloemhof, 2025-06-18:
As far as I understand the code, it is correct what I say. You confiugre it in minutes, it calculates in seconds. 

There are a couple of hard coded timers which seems fine, but this one I'm not sure of: https://github.com/dingo35/SmartEVSE-3.5/blob/6e7c99a7827bd09ffd7bd5c98a760d2aa2cb5150/SmartEVSE-3/src/main.cpp#L1355 

Today was a good day for testing, EV needed 30% and the weather is partly cloudy. What I notice is that, because of the very short timer(s), the switching is a little to fast in my opinion. Maybe it is needed to introduce a new seperate wait timer for phase switching? 

I am now running `Version: 13:38:12 @Jun 14 2025`.

PimDoos, 2025-06-18:
I have seen successful switching from 1P > 3P > 1P after 5 minutes (as set in the SmartEVSE settings)

[image]

dingo35, 2025-06-19:
Ok Im working on a new version so I suggest you revert to v3.8.3 or v3.7.4;  Ill be back!

dingo35, 2025-07-11:
Your Isum is around 1.3A, and you are charging 3P 6.0A (sometimes 5.5A); this means if it would stop charging, it would free up 3*6 = 18.0 A, so Isum would be 1.3A-18.0A = -16.7A, which is much more delivery then your Startcurrent of -6A ; to prevent unnecessary start/stops it just continues charging:

So this would be expected behaviour.

dingo35, 2025-07-12:
@kozmoz @Hermarcel @rjmgmt @fbloemhof Great day for testing this today, lots of clouds and sunny periods !!

Switching 1P -> 3P -> 1P seems to go flawless now; notice that MQTT now publishes NrOfPhases so you can monitor what is happening.

kozmoz, 2025-07-12:
For debugging purposes, I’m sharing my log file on GitHub. The sequence was as follows: started with 3-phase (3P) charging → switched to 1-phase (1P) → turned off → back to 1P → and finally returned to 3P (at least this was the LCD screen telling me). I did not modify any charger settings during the test. The only change was turning on an 11 kW water heater to simulate reduced available power.

[smartevse-20250712-4.txt](https://github.com/user-attachments/files/21195198/smartevse-20250712-4.txt)

dingo35, 2025-07-12:
So any unexpected behaviour?

kozmoz, 2025-07-12:
> So any unexpected behaviour?

I didn’t notice anything unusual. The only thing that stood out is that when it says it's about to switch phases, it appears to trigger a second countdown timer.

I’ll need to reconfigure my monitoring setup to confirm whether it actually performed the phase switch as indicated, but I won’t be able to do that until next month. Due to vacation, I won’t have access to my 3-phase charger after today.

fbloemhof, 2025-07-12:
@dingo35 seems to work good today indeed! I noticed you released 3.9.3 with this code, great work.

kozmoz, 2025-07-12:
> NrOfPhases

That's a nice addition. For consistency, can this also be added to the settings?

dingo35, 2025-07-12:
@fbloemhof you mean v3.9.0 instead of 3.9.3 ?

dingo35, 2025-07-12:
@kozmoz good point, just added:
        doc["evse"]["nrofphases"] = Nr_Of_Phases_Charging;

to master branch.

rjmgmt, 2025-07-12:
Thanks for the clarification @dingo35 I have done a full charging session over two days on solar now (with a 3-phase car), and everything seems to work as it should. It switches both up and down now, which really helps to maximize solar self-consumption. I will try with the 2-phase car this afternoon as well, since that is where the issue occured with originally.