dingo35/SmartEVSE-3.5 #154: MQTT Issues - reconnects and status

stomko11, 2024-11-14:
**Describe the bug**
I have configured my MQTT details. It connects fine to the server, I verified it via mqtt explorer, device is correctly reporting all the values. However there are 2 issues:
1) webinterface of EVSE always shows "disconnected"
[image]

2) I see lot of reconnects on the mqtt server, like every minute. I found topic https://github.com/dingo35/SmartEVSE-3.5/issues/115 which seems similar to what people were seeing, but it suggested fix would be incorporated in 3.7.0. I am on 3.7.2

```
2024-11-14 04:09:53: New connection from 192.168.x.x:51563 on port 1883.
2024-11-14 04:09:53: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:09:53: New client connected from 192.168.x.x:51563 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:09:56: New connection from 192.168.x.x:51564 on port 1883.
2024-11-14 04:09:56: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:09:56: New client connected from 192.168.x.x:51564 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:05: New connection from 192.168.x.x:51565 on port 1883.
2024-11-14 04:10:05: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:05: New client connected from 192.168.x.x:51565 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:06: New connection from 192.168.x.x:51566 on port 1883.
2024-11-14 04:10:06: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:06: New client connected from 192.168.x.x:51566 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:08: New connection from 192.168.x.x:51567 on port 1883.
2024-11-14 04:10:08: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:08: New client connected from 192.168.x.x:51567 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:15: New connection from 192.168.x.x:51568 on port 1883.
2024-11-14 04:10:15: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:15: New client connected from 192.168.x.x:51568 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:18: New connection from 192.168.x.x:51569 on port 1883.
2024-11-14 04:10:18: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:18: New client connected from 192.168.x.x:51569 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:21: New connection from 192.168.x.x:51570 on port 1883.
2024-11-14 04:10:21: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:21: New client connected from 192.168.x.x:51570 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:30: New connection from 192.168.x.x:51571 on port 1883.
2024-11-14 04:10:30: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:30: New client connected from 192.168.x.x:51571 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:31: New connection from 192.168.x.x:51572 on port 1883.
2024-11-14 04:10:31: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:31: New client connected from 192.168.x.x:51572 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:33: New connection from 192.168.x.x:51573 on port 1883.
2024-11-14 04:10:33: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:33: New client connected from 192.168.x.x:51573 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:43: New connection from 192.168.x.x:51574 on port 1883.
2024-11-14 04:10:43: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:43: New client connected from 192.168.x.x:51574 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:45: New connection from 192.168.x.x:51575 on port 1883.
2024-11-14 04:10:45: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:45: New client connected from 192.168.x.x:51575 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:46: New connection from 192.168.x.x:51576 on port 1883.
2024-11-14 04:10:46: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:46: New client connected from 192.168.x.x:51576 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:55: New connection from 192.168.x.x:51577 on port 1883.
2024-11-14 04:10:55: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:55: New client connected from 192.168.x.x:51577 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:10:58: New connection from 192.168.x.x:51578 on port 1883.
2024-11-14 04:10:58: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:10:58: New client connected from 192.168.x.x:51578 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:11:01: New connection from 192.168.x.x:51579 on port 1883.
2024-11-14 04:11:01: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:11:01: New client connected from 192.168.x.x:51579 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:11:08: New connection from 192.168.x.x:51580 on port 1883.
2024-11-14 04:11:08: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:11:08: New client connected from 192.168.x.x:51580 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:11:10: New connection from 192.168.x.x:51581 on port 1883.
2024-11-14 04:11:10: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:11:10: New client connected from 192.168.x.x:51581 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:11:16: New connection from 192.168.x.x:51582 on port 1883.
2024-11-14 04:11:16: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:11:16: New client connected from 192.168.x.x:51582 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:11:20: New connection from 192.168.x.x:51583 on port 1883.
2024-11-14 04:11:20: Client SmartEVSE-xxxx already connected, closing old connection.
[... 41 lines cut ...]
2024-11-14 04:12:21: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:12:21: New client connected from 192.168.x.x:51597 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:12:25: New connection from 192.168.x.x:51598 on port 1883.
2024-11-14 04:12:25: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:12:25: New client connected from 192.168.x.x:51598 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
2024-11-14 04:12:28: New connection from 192.168.x.x:51599 on port 1883.
2024-11-14 04:12:28: Client SmartEVSE-xxxx already connected, closing old connection.
2024-11-14 04:12:28: New client connected from 192.168.x.x:51599 as SmartEVSE-xxxx (p2, c1, k15, u'evsemqtt').
```

**Upload your config**
Press the "raw" button in the SmartEVSE webserver and upload the json settings file, so we can look at your configuration.
-attached

**To Reproduce**
Steps to reproduce the behavior:
1. Normal
2. Configure MQTT

**Expected behavior**
I would expect no reconnects on MQTT server. This is the only client that does this. Also I would expect "connected" status in the webUI

**Screenshots**
See above
[settings.json](https://github.com/user-attachments/files/17742415/settings.json)

dingo35, 2024-11-14:
You are right, that fix is implemented in v3.7.2, and was tested by the people having the issue.
Im going to try to reproduce this!

dingo35, 2024-11-14:
Which MQTT server on which OS are you using?

stomko11, 2024-11-14:
Mosquitto as official addon on Home Assistant OS

dingo35, 2024-11-14:
Do you have the possibillity to install a standalone mosquitto server on Linux to see of this behaviour persists with your SmartEVSE?

stomko11, 2024-11-14:
I should be able to try in docker container on a server, outside of HA. Will try later today and let you know.

stomko11, 2024-11-14:
I have tested by installing official docker image of mosquitto, running on unraid server.
Result is the same, this is running for about a minute. What I find strange - as I found out correct settings to allow mqtt connections in to the docker, I have been messing with mqtt setup on evse - changing usernames from none, to anonymous, evsemqtt and as well test1. What I find strange is that after I apply config, instead of just trying with the last entered username onwards, it actually keeps trying with all usernames entered before. As if each "save" on evse started additional instance of mqtt client with its own settings? I have not tried rebooting evse controller.
```
1731590788: Client SmartEVSE-7560 already connected, closing old connection.
1731590788: New client connected from 192.168.89.67:60043 as SmartEVSE-7560 (p2, c1, k15, u'evsemqtt').
1731590791: New connection from 192.168.89.67:60045 on port 1883.
1731590791: Client SmartEVSE-7560 already connected, closing old connection.
1731590791: New client connected from 192.168.89.67:60045 as SmartEVSE-7560 (p2, c1, k15, u'evsemqtt').
1731590791: New connection from 192.168.89.67:60046 on port 1883.
1731590791: Client SmartEVSE-7560 already connected, closing old connection.
1731590791: New client connected from 192.168.89.67:60046 as SmartEVSE-7560 (p2, c1, k15, u'evsemqtt').
1731590798: New connection from 192.168.89.67:60048 on port 1883.
1731590798: Client SmartEVSE-7560 already connected, closing old connection.
1731590798: New client connected from 192.168.89.67:60048 as SmartEVSE-7560 (p2, c1, k15).
1731590803: New connection from 192.168.89.67:60050 on port 1883.
1731590803: Client SmartEVSE-7560 already connected, closing old connection.
1731590803: New client connected from 192.168.89.67:60050 as SmartEVSE-7560 (p2, c1, k15, u'anonymous').
1731590806: New connection from 192.168.89.67:60052 on port 1883.
1731590806: Client SmartEVSE-7560 already connected, closing old connection.
1731590806: New client connected from 192.168.89.67:60052 as SmartEVSE-7560 (p2, c1, k15, u'evsemqtt').
1731590806: New connection from 192.168.89.67:60053 on port 1883.
1731590806: Client SmartEVSE-7560 already connected, closing old connection.
1731590806: New client connected from 192.168.89.67:60053 as SmartEVSE-7560 (p2, c1, k15, u'evsemqtt').
1731590813: New connection from 192.168.89.67:60056 on port 1883.
1731590813: Client SmartEVSE-7560 already connected, closing old connection.
1731590813: New client connected from 192.168.89.67:60056 as SmartEVSE-7560 (p2, c1, k15, u'evsemqtt').
1731590813: New connection from 192.168.89.67:60057 on port 1883.
1731590813: Client SmartEVSE-7560 already connected, closing old connection.
1731590813: New client connected from 192.168.89.67:60057 as SmartEVSE-7560 (p2, c1, k15).
1731590816: New connection from 192.168.89.67:60059 on port 1883.
1731590816: Client SmartEVSE-7560 already connected, closing old connection.
1731590816: New client connected from 192.168.89.67:60059 as SmartEVSE-7560 (p2, c1, k15, u'evsemqtt').
1731590820: New connection from 192.168.89.67:60061 on port 1883.
1731590820: Client SmartEVSE-7560 already connected, closing old connection.
1731590820: New client connected from 192.168.89.67:60061 as SmartEVSE-7560 (p2, c1, k15, u'test1').
1731590821: New connection from 192.168.89.67:60062 on port 1883.
1731590821: Client SmartEVSE-7560 already connected, closing old connection.
1731590821: New client connected from 192.168.89.67:60062 as SmartEVSE-7560 (p2, c1, k15, u'anonymous').
1731590823: New connection from 192.168.89.67:60063 on port 1883.
1731590823: Client SmartEVSE-7560 already connected, closing old connection.
1731590823: New client connected from 192.168.89.67:60063 as SmartEVSE-7560 (p2, c1, k15, u'evsemqtt').
1731590824: New connection from 192.168.89.67:60064 on port 1883.
1731590824: Client SmartEVSE-7560 already connected, closing old connection.
1731590824: New client connected from 192.168.89.67:60064 as SmartEVSE-7560 (p2, c1, k15, u'evsemqtt').
1731590831: New connection from 192.168.89.67:60067 on port 1883.
1731590831: Client SmartEVSE-7560 already connected, closing old connection.
1731590831: New client connected from 192.168.89.67:60067 as SmartEVSE-7560 (p2, c1, k15, u'test1').
1731590831: New connection from 192.168.89.67:60068 on port 1883.
1731590831: Client SmartEVSE-7560 already connected, closing old connection.
1731590831: New client connected from 192.168.89.67:60068 as SmartEVSE-7560 (p2, c1, k15).
```

dingo35, 2024-11-14:
Rebooting the controller seems the most logical thing to do in this test situation...

stomko11, 2024-11-14:
Strange - rebooted EVSE (did that yesterday too), and I no longer see that behavior neither on mosquitto docker not in HA addon

dingo35, 2024-11-18:
This strange behaviour should be solved in this version:
[20ccbee72.zip](https://github.com/user-attachments/files/17800365/20ccbee72.zip)

Please test!

zulufoxtrot, 2024-11-23:
I had the exact same issue (HA, mosquitto, same symtoms)
I tested your firmware and the issue seems to be fixed.
The status page shows MQTT as connected, and I no longer see connects/disconnects in mosquitto logs.

dingo35, 2024-11-23:
Thx for testing!

closed 2024-11-23 by dingo35

EmanuelStassar, 2024-12-22:
I was having issues connecting to HA with MQTT as well. Your test version solved the issue! Thanks