dingo35/SmartEVSE-3.5 #155: Charging with Smart4four EQ is no longer possible from version 3.6.0

Elektrofix-OL, 2024-11-16:
**Describe the bug**
I built a wallbox for a colleague. The Smart4Four can't be charged with 3 phases.

**Upload your config**
`{"version":"v3.7.2","serialnr":7508,"mode":"SMART","mode_id":3,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"LohmannCity","rssi":-13,"bssid":"74:42:7F:B7:15:35"},"evse":{"temp":26,"temp_max":65,"connected":false,"access":true,"mode":1,"loadbl":0,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":8,"current_max":10,"current_main":50,"current_max_circuit":32,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0},"mqtt":{"host":"192.168.100.2","port":1883,"topic_prefix":"SmartEVSE","username":"Lohmann","password_set":true,"status":"Connected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":12,"import_active_power":0,"total_kwh":1836.199951,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":1836.199951,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":44,"L1":15,"L2":14,"L3":15,"last_data_update":1731771512,"original_data":{"TOTAL":44,"L1":15,"L2":14,"L3":15}},"backlight":{"timer":68,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}`
**Upload a debug log**
[putty27327.log](https://github.com/user-attachments/files/17787050/putty27327.log)

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the plug to the vehicle, no charge
2. Firmware to 3.5.1
3. Connect the plug to the vehicle, charge
4. Firmware to 3.6.0 
5. Connect the plug to the vehicle, no charge

**Expected behavior**
That the Smart4four charges on the wallbox in the same way as a Smart4two

**continue exams**
Before we changed the software, we swapped the hardware from a SmartEVSE 3.7.2 to a SmartEVSE 2.30, because the Smart4four charges easily on another wallbox of mine with a SmartEVSE 2.0x.

It seems as if the SmartEVSE from 3.6 has changed the modulation on the CP, so that the Smart4four cannot cope with it.

dingo35, 2024-12-27:
I cannot find any changes between 3.5.1 and 3.6.0 that could cause this, this is the list:
```
48c5684 (tag: v3.6.0) glcd.cpp: fix Switch menu hanging when mode=OFF and switch toggling to Access B
7aa8aff evse.cpp: fix reboot on wifi disconnect
5e55919 tests.sh: make sure no other tests are running in background
37867cb update2.html: small clarification in text
da5539c tests.sh: make test script mongoose compatible
c8d7d14 evse.cpp: do not register mainsmeter worker for slave because we get the currents already from BroadcastCurrent from the master
28ca1c1 evse.cpp: fix mainsmeter timeout on slave on solar mode when waiting for power
0834e37 evse.h: small improvement on comments
1375b75 (tag: v3.6.0-RC2) update2.html, evse.cpp: allow direct upload of firmware.debug.bin
4886937 evse.cpp: lower mongoose priority and increase delay even more
a702b9d evse.cpp: lower mongoose priority and increase delay
71c853c evse.cpp: move mongoose poll to separate task
081319a docs: remove old remark on contactor2
1547209 evse.cpp: lock cable when delayed charging AND not disconnected 
35733dc coredumping.txt: add instructions for advanced developers howto analyze core dump
85804e9 evse.cpp: enable config of static ip and dns
01cb6f1 evse.cpp: make mongoose using dhcp generated dns instead of 8.8.8.8
7f3d9bb (tag: v3.6.0-RC1) pio-build.yaml: no longer build spiffs.bin 
a006f89 evse.cpp: overrule delayed starttime when POSTing new mode
4d26d3d app.js: show closing message after upload completes
92cbdc0 platformio.ini: remove upload_port since it is automatically detected 
342467e packfs.py: avoid unnecessary file copying
503f16a packfs.py: code out bash wildcard expansion because Win doesnt expand it
5d99287 packfs.py: fix compile error on Windows
165b7f1 evse.cpp: fix rfid.txt upload if no proper eof/eol
3303b38 Fix incorrect EnergyCharged value when going into STATE_C
985be4b evse.cpp: Lock actuator when delayed charging
d1af652 evse.cpp: get rid of superfluous [Un]lockCable semaphores
0612d29 evse.h: log locking actuator
6f58e50 OneWire.h, OneWire.cpp, evse.cpp: fix rfid.txt upload
680e9cf glcd.cpp: fix display of top line messages of RFID reader submenu
d4b1006 evse.cpp: remove DELETE constraint on erasesettings
0157493 packfs.py: remove OS specific commands
c39f267 pack.c: remove it since we dont use it
de0ce75 packfs.py: make build process independent of os gzip
e9ee81d pack.py, packfs.py: make build process independent of cc or gcc
77d6f3e platformio.ini: move to emodbus 1.7.1
c825d49 evse.cpp: improve mqtt url code
b687802 add partitions_custom.csv.coredump for developers
98ca481 eModbus-fix: remove it since we don't use it anymore
171c89c packfs.py: make sure necessary firmware flash files are there
ab49e51 evse.h, evse.cpp: move some variables from int32 to int16
cf26659 platformio.ini: upgrade to espressif 6.6.0
1f08534 evse.cpp: improve API response when feeding EVmeter
e6efc51 upload rfid.txt
83f1c70 move mqtt to mongoose
1676de6 delete spiffs
5749611 all files: add mongoose except mqtt except rfid.txt
```

The only way I can think of pinpointing this problem is if you go to the middle of the list, e.g. :
git checkout 5d99287
pio run -t upload
.... and see if the problem still exists. If it does, go down on the list, if not, go up, until you find the commit that causes the problem.

Then we probably understand what is causing the problem and we can probably solve it easily.

I understood you had problems in the past compiling from source, are you wiling and able to do this?

dingo35, 2025-02-06:
Try setting MinCurrent to 7A, this helped with a Kona:
https://github.com/dingo35/SmartEVSE-3.5/issues/40#issuecomment-2640681624

mbuijs, 2025-02-13:
FWIW, I updated my SmartEVSE v3 from serkri's v1.7.2 to v3.6.10 in 1 step a few weeks ago. My Zoe ZE40 has been charging properly on a daily basis.

MinCurrent is set to 6A in my config.

closed 2025-04-01 by dingo35