dingo35/SmartEVSE-3.5 #159: When CurrentMaxSumMains (or capacity) is reached, charging continues

fhbonte, 2024-12-09:
**Describe the bug**
Lowering "current_max_sum_mains" does not lead to state 'charging stopped' in Smart mode (3 phase).
Charging goes on with a value of "current_min" on each phase.

**Upload your config**
{"version":"12:02:19 @Nov 17 2024","serialnr":7149,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{ XXX },"evse":{"temp":25,"temp_max":65,"connected":true,"access":true,"mode":1,"loadbl":1,"pwm":102,"custombutton":false,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":16,"current_main":32,"current_max_circuit":20,"current_max_sum_mains":10,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":10,"enable_C2":"Always On","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{ YYYY },"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"API","address":12,"import_active_power":4.099999905,"total_kwh":1039,"charged_kwh":4.199999809,"currents":{"TOTAL":178,"L1":59,"L2":60,"L3":59},"import_active_energy":1039,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":245,"L1":72,"L2":106,"L3":67,"last_data_update":1733757448,"original_data":{"TOTAL":245,"L1":72,"L2":106,"L3":67}},"backlight":{"timer":120,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

**Upload a debug log**
use version 12:02:19 that solves the issue #151

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Smart
2. What steps do you take so the problem arises?  changing via /Set/CurrentMaxSumMains value of value MaxSumMains

**Expected behavior**
Expected is as soon MaxSumMains (or capacity) is reached (sum of all phases), smartevse state changes from 'charging' to 'charging stopped' in Smart mode

dingo35, 2024-12-09:
Thx for your report; I thought this bug was solved in 3.7.2, didnt want to put you on an unnecessary update!

Its on my list!

fhbonte, 2024-12-09:
[like]  Frederick Bonte reacted to your message:

fhbonte, 2024-12-09:
If changing mode manual to OFF and then back to smart, the behaviour is
- State: Charging Stopped
- Error: No Power Available
[= correct at my opinion]

The issue seems thus to occur when changing "current_max_sum_mains" 'on the fly' when in Smart mode. 
This is in line with description of the bug above.

fhbonte, 2024-12-10:
Hallo, 

As written above, in smart mode, it should be expected that the smartevse stops charging. However, in log it is not noticed that power is short.

Looking at the code (line 1290 and following), the following seems peculiar:

if (MainsMeter.Type && Mode != MODE_NORMAL)
                if (IsetBalanced > (MaxMains * 10) - Baseload) 

The IsetBalanced seems to be per phase, while MaxMains is total [in my mind: MaxSumMains = sum (L1 L2 L3)]

dingo35, 2024-12-10:
MaxMains !=MaxSumMains

dingo35, 2024-12-27:
This should solve the problem, please test:
[b235c16a85d.zip](https://github.com/user-attachments/files/18258993/b235c16a85d.zip)

closed 2025-01-18 by fhbonte

dingo35, 2025-01-18:
So test is ok?