dingo35/SmartEVSE-3.5 #164: Failed upgrade v3.7.0 -> v3.7.3. Froze device

VincentSC, 2025-01-01:
**Describe the bug**
When upgrading via the web-interface from v3.7.0 to v3.7.3, the last screen I saw was this:

[image]

Then nothing happened. After about 10 minutes of nothing, I tried to open a second tab to the ip-address, but then found out that the device was not connected to wifi anymore. 

Opening up the charging pole, the LCD shows "Ready to charge", but none of the buttons work. Also, when connecting the charging cable, nothing happens. So it looks very frozen...

Got the pre-built V3 from Stegen. I used stock Safari, as Chrome and Firefox have this "you must use https everywhere, also when using local ip-addresses, and also when you turned it off" now.

I *assume* I need to re-upload V3.7.3 via the USB-C cable (when it's not raining)? What can I do to find out what went wrong, such that we can prevent this for others? Should we mark V3.7.3 unstable, just to be sure?

dingo35, 2025-01-01:
First try to coldboot your device (so get the power off and on again).

If that doesnt work, try http://ip/update; try reflashing v3.7.3, it should be a stable version with only a few modifications.
If all fails, follow the manual: https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/building_flashing.md#i-think-i-bricked-my-smartevse

VincentSC, 2025-01-01:
Forgot to mention that I tried cold-booting already. Just tried a second time, and again nothing

It does not get to the stage where it connects to wifi, so USB it is.

dingo35, 2025-01-01:
v3.7.0 is the version with wrong update links in it, it actually loads Sensorbox firmware; its both ESP32 so although the LCD screen is blank, the webserver still should be running... Most users reported http://ip/update is still available and updated like that, some users had to do the USB procedure...

VincentSC, 2025-01-02:
Fixed it via USB. Was rather easy via MacOS.

Strangely enough the wifi was disabled. This is what caused the additional problems.

closed 2025-01-02 by VincentSC