dingo35/SmartEVSE-3.5 #165: After power outage both my SmartEVSE are bricked. PLS help

Adze1502, 2025-01-08:
**Describe the bug**
While i was charging both EVs power in our house went out. After power restored both SmartEVSE's were unoperable. White flashing screen with no text on it. Tried to do a fimware debrick (see debug log), but to no avail. Can you please help me into right direction to get both SmartEVSE up again?

**Upload your config**
At this moment i'm unable to supply you with config as both SmartEVSE's are not working (blanc screen)

**Upload a debug log**
```
C:\Users\jeroe>esptool --port com4 write_flash 0x10000 firmware.bin
esptool.py v4.8.1
Serial port com4
Connecting...
Failed to get PID of a device on com4, using standard reset sequence.
......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Failed to get PID of a device on com4, using standard reset sequence.
...
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:4c:ab:3b:01:28
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00010000 to 0x00189fff...
Compressed 1545040 bytes to 981504...
Wrote 1545040 bytes (981504 compressed) at 0x00010000 in 86.8 seconds (effective 142.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

C:\Users\jeroe>esptool --port com4 write_flash 0x1c0000 firmware.bin
esptool.py v4.8.1
Serial port com4
Connecting...
Failed to get PID of a device on com4, using standard reset sequence.
.
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Failed to get PID of a device on com4, using standard reset sequence.
...
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:4c:ab:3b:01:28
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x001c0000 to 0x00339fff...
Compressed 1545040 bytes to 981504...
Wrote 1545040 bytes (981504 compressed) at 0x001c0000 in 87.0 seconds (effective 142.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

**Screenshots**
A video of what i see and hear: https://youtube.com/shorts/bMbPOz_I98g

dingo35, 2025-01-08:
It looks like the bootloader and/or other partitions are damaged; no worries, if you compile the latest version 3.7.3 yourself and flash it with "pio run -t upload" all partitions are flashed.

Detailed instructions you find here:
https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/building_flashing.md#building-the-firmware

Adze1502, 2025-01-08:
Thanks for the insight. Managed to bring them back to life with your help. Strange how a powercut could cause such a breakage. Nevertheless, problem resolved; case closed.

closed 2025-01-08 by Adze1502