dingo35/SmartEVSE-3.5 #166: Reboot button doesn't work after update

Imaginous, 2025-01-10:
**Describe the bug**
I was using a self compiled firmware version.
Selected update and the latest bleeding edge V3.7.3 firmware.
The update proceeds and ends with the message that the SmartEVSE will reboot once the car is disconnected.

I wanted to force the update and went to the main screen and pressed "reboot", this doesn't work after a firmware update.

**To Reproduce**

1.  Update firmware via the "standard" button.
2. After the update is finished and a car is connected it will wait for a disconnect.
3. Go back to the main page and press the "reboot" button.

**Expected behavior**
I expected the device to reboot.

**Note:**
The car was only connected, not charging.

When you upload a firmware file manually the system will directly reboot afterwards.

dingo35, 2025-01-10:
"The update proceeds and ends with the message that the SmartEVSE will reboot once the car is disconnected."
"The update proceeds and ends with the message that the SmartEVSE will reboot once the car is disconnected."

Solution: disconnect the car.

This option is chosen because rebooting with a connected EV can lead to unpredictable results, especially in configurations with multiple EVSEs connected ("Power Sharing").

Imaginous, 2025-01-10:
Point accepted ;)

Maybe disable / hide the reset button.

closed 2025-01-10 by Imaginous