dingo35/SmartEVSE-3.5 #167: EVmeter doesn't open contractor when mode is switched to OFF

ps-nl, 2025-01-12:
**Describe the bug**
When switching the Mode from Smart (or norma) to OFF, the contactor does not open. I have double checked the contactor to see if the contacts are welded, but all is fine. Contactor opens again when power cycling the EVSE-meter.

**Upload your config**
{"version":"v3.7.3","serialnr":7216,"mode":"OFF","mode_id":0,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"###","rssi":-72,"bssid":"F0:9F:C2:D7:49:BE"},"evse":{"temp":24,"temp_max":65,"connected":true,"access":false,"mode":1,"loadbl":0,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":145,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":6,"enable_C2":"Always Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{"host":"10.0.133.20","port":1883,"topic_prefix":"SmartEVSE-7216","username":"smartevse","password_set":true,"status":"Connected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"WAGO","address":20,"import_active_power":0,"total_kwh":14496.7998,"charged_kwh":0,"currents":{"TOTAL":5,"L1":2,"L2":1,"L3":2},"import_active_energy":14496.7998,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":36,"L1":24,"L2":12,"L3":0,"last_data_update":1736692290,"original_data":{"TOTAL":36,"L1":24,"L2":12,"L3":0}},"backlight":{"timer":0,"status":"OFF"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

**Upload a debug log**
[Log.log](https://github.com/user-attachments/files/18389736/Log.log)

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar?
2. What steps do you take so the problem arises?

**Expected behavior**
A clear and concise description of what you expected to happen; expected Charging Current?

**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

dingo35, 2025-01-17:
In your log you can see the firmware is actually switching Contactor1 off:
```(I) (printStatus)(C1) L1: 7.3 A L2: 8.7 A L3: 0.0 A Isum: 16.0 A
(setState)(C1) 15:22:21 STATE C -> C1
(setState)(C0) 15:22:22 STATE C1 -> B1
(setState)(C0) Pilot Disconnected
(setState)(C0) Switching Contactor1 OFF.
(setState)(C0) Switching Contactor2 OFF.
```   
So if your Contactor 1 is correctly wired (between C1 and N, C2 is for a second contactor), you probably have a hardware problem?

dingo35, 2025-01-17:
Checked it on my test bench, cannot reproduce this problem.

What type of contactor are you using? You know Solid State Contactors are giving problems with SmartEVSE? 
Also you could measure the voltage between C1 and N when switching to B1 with a multimeter, to localize the problem...

ps-nl, 2025-01-18:
Thanks, I'll check the hardware. The contactor type should be okay: it's an ABB ESB 40-40 mechanical contactor.

ps-nl, 2025-01-19:
I have checked the wiring, and it is all hooked up correctly. Even though the log says the contractors are being switched off, there remains a 230V across terminals C1 and N. 

I am baffled as to what this might be. Other than software updates of the Tesla Model 3, I can't think of anything that changed in the system. It would also surprise me if there is a hardware defect in the internals (relay?) of the EVSE, because after switching power on and off again, the C1-N voltage drops to 0 again. 

How to go about troubleshooting this? Perhaps a full wipe of the storage and settings of the EVSE?

dingo35, 2025-01-19:
Yes, I would say http://ipaddress/erasesettings , if that doesnt help Im afraid you'll have to call in Michael Stegen's help....

ps-nl, 2025-01-24:
Unfortunately unable to resolve this by resetting and re-uploading firmware. I will contact Michael as suggested.

mstegen, 2025-01-24:
It's most likely the contactor that is incompatible.
The SmartEVSE v3 does not support hum free/silent contactors.
They will switch on fine, but will not switch off anymore.
If you use a normal contactor, the problem will be fixed. For example the IKA432-40.
The IKD series from Iskra also does -not- work.

Another thing to check is the isolation between the poles, this should be >3mm, It's unclear what the ABB contactors value is. (cant find it online)
And finally, it should switch off as quickly as possible, the IKA series does this within 30ms. Total switch off time (including reaction time of the SmartEVSE) should be <100ms. It it takes longer then some EV's will display a charger error message.

closed 2025-01-27 by ps-nl