dingo35/SmartEVSE-3.5 #168: Introduce pause function in NORMAL or SMART mode. (Override current to zero)

Adze1502, 2025-01-20:
**Is your feature request related to a problem? Please describe.**
I have two SmartEVSE chargers on one 25A circuit. OCPP is configured and correctly working, using TAP Electric service. While charging in NORMAL or SMART mode, I am able to override charge current within configured MIN and MAX values, but I'm unable to temporary pause charging. When I press "STOP", charging is stopped, but then the OCPP session is also stopped.

**Describe the solution you'd like**
I like to have the option to stop/pause charging, without dropping the active OCPP charge session. In SMART mode, when MAINS is saturated, charging is temporary halted, so I know it is feasible somehow... If this pause option was available I could control charging rate from Home Assistant with more granularity and more personal preferences.

**Describe alternatives you've considered**
... One user suggested to switch between NORMAL and SOLAR to work around this issue. While this might work I think it ugly and introduces new issues during summer. 

**Additional context**
I will make a € 50,- donation for your effort if this function ends up in a new release of SmartEVSE. Thank you for your time.

dingo35, 2025-04-03:
Ok please test this version:

[08f08a766.zip](https://github.com/user-attachments/files/19588687/08f08a766.zip)

You can enter PAUSE:
-by pressing the PAUSE button on the webserver
-by sending mode=4 via the REST API
-by sending topic /Set/Mode value "Pause"

Adze1502, 2025-04-08:
Fixed in 3.8.0

closed 2025-04-08 by Adze1502