dingo35/SmartEVSE-3.5 #169: MQTT Connected, but EvMeter and MainsMeter not updating

RonaldPhilipsen, 2025-01-26:
**Describe the bug**
I'm publishing data to both:
SmartEVSE-7277/Set/MainsMeter  `19.03:7.29:0.0`
SmartEVSE-7277/Set/EVMeter `0.0:0.0:0.0:0.0:1970373.0`

This data is visible (and retained) on my EMQX broker(s)
however, the data does not seem to update on the web UI and a communication error stops any charging

When updating the data over the rest API there's no issue 

[settings.json](https://github.com/user-attachments/files/18550754/settings.json)

**Upload a debug log**
Flash the debug version of the firmware (see https://github.com/dingo35/SmartEVSE-3.5/blob/master/SmartEVSE-3/HowToFlash.txt), telnet to your device, capture the debug log, cut it so that it shows JUST BEFORE and JUST AFTER the problem arises, and upload it here.

Tried both firmware-debug.bin and firmware-debug-signed.bin 
Telnetting to 
```
✦ ❯ telnet 172.16.10.2 80
Trying 172.16.10.2...
Connected to 172.16.10.2.
Escape character is '^]'.

Connection closed by foreign host.
```

**To Reproduce**

Not quite sure, sending data over rest works just fine, and the MQTT packets seem to arrive at the EVSE (172.16.10.2 in the picture below)

[image]

**Expected behavior**
Data is read from the MQTT packets

maart84, 2025-01-26:
Your values are wrong i guesss. Not sure of decimals are allowed but of you want to post 19.1 A you should post 191

dingo35, 2025-01-26:
@maart84 is right, you're not supposed to post decimals, just RTFM:
https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/configuration.md#mqtt-api

RonaldPhilipsen, 2025-01-30:
Thanks for the clarification, Reading through the documentation linked it's not quite clear that this is the case.

perhaps something like

> ...where L1 - L3 are integers representing the currents in deci-Ampères. So 100 means 10.0A.

instead of the current

> ...where L1 - L3 are the currents in deci-Ampères. So 100 means 10.0A.

would clear up some of the confusion.

closed 2025-01-30 by dingo35