dingo35/SmartEVSE-3.5 #172: EVmeter kWh incorrect or not updating

AlexT60, 2025-01-30:
**Describe the bug**
After updating from 3.6.10 to 3.7.3 the kWh values are incorrect or remain at 0. The EVMeter currents are correct and realtime updated.

A warm boot does not make any difference. After a cold boot values of kWh will be "0" and are not updated during charging.

**Upload your config**
{"version":"v3.7.3","serialnr":7173,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Tenderifi","rssi":-49,"bssid":"0C:80:63:EB:8A:12"},"evse":{"temp":21,"temp_max":75,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":425,"custombutton":false,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":250,"override_current":0,"current_min":6,"current_max":25,"current_main":25,"current_max_circuit":25,"current_max_sum_mains":75,"max_sum_mains_time":0,"solar_max_import":9,"solar_start_current":3,"solar_stop_time":30,"enable_C2":"Not present","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{"host":"192.168.178.19","port":1883,"topic_prefix":"SmartEVSE-7173","username":"AlexT","password_set":false,"status":"Disconnected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"ABB","address":15,"import_active_power":11,"total_kwh":15.89999962,"charged_kwh":-15320.09961,"currents":{"TOTAL":468,"L1":157,"L2":157,"L3":154},"import_active_energy":15.89999962,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":455,"L1":135,"L2":146,"L3":174,"last_data_update":1738246499,"original_data":{"TOTAL":455,"L1":135,"L2":146,"L3":174}},"backlight":{"timer":120,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

After a cold boot:
{"version":"v3.7.3","serialnr":7173,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Tenderifi","rssi":-65,"bssid":"0C:80:63:EB:8A:40"},"evse":{"temp":22,"temp_max":75,"connected":true,"access":true,"mode":1,"loadbl":0,"pwm":425,"custombutton":false,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":250,"override_current":0,"current_min":6,"current_max":25,"current_main":25,"current_max_circuit":25,"current_max_sum_mains":75,"max_sum_mains_time":0,"solar_max_import":9,"solar_start_current":3,"solar_stop_time":30,"enable_C2":"Not present","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{"host":"192.168.178.19","port":1883,"topic_prefix":"SmartEVSE-7173","username":"AlexT","password_set":false,"status":"Disconnected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"ABB","address":15,"import_active_power":11,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":469,"L1":156,"L2":157,"L3":156},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":464,"L1":138,"L2":147,"L3":179,"last_data_update":1738246944,"original_data":{"TOTAL":464,"L1":138,"L2":147,"L3":179}},"backlight":{"timer":120,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

**Upload a debug log**
No debug files made.

**To Reproduce**
Steps to reproduce the behavior:
1. In Smart mode (any mode will have the same behavior)

**Expected behavior**
Charging is working fine, except EVmeter kWh values

**Screenshots**
[image]

[image]

AlexT60, 2025-02-06:
In the Assets of v3.7.0 the distribution files are missing, I took v3.7.1 and can confirm same problem already exists.

I have added the two log files

[Puttylog 3.6.10.txt](https://github.com/user-attachments/files/18698136/Puttylog.3.6.10.txt)
[Puttylog 3.7.3.txt](https://github.com/user-attachments/files/18698135/Puttylog.3.7.3.txt)

dingo35, 2025-02-06:
That is correct, v.3.7.0 had some wrong update-links in them, sorry I forgot about that!

Ill lool into the log files!

dingo35, 2025-02-13:
Modbus traffic seems correct ...

AlexT60, 2025-02-13:
The file add7f40 shows challenging differences in data formats of the kWh-meters. The comments in the file explains the signed and unsigned variables and different dimensions (16, 32 and 64 bits).
Any chance the mistake can be found there?

dingo35, 2025-02-13:
It might, but I couldnt find it; my problem is that I have no kWh meters at my current location, so hard to debug...

AlexT60, 2025-02-13:
If I am the only user with a ABB kWh-meter forget it.
Else if I can support you by sending my meter (unless you live on the Pitcairn Islands😄) let me know.
Unfortunately I am from a different era (PL/M iRMX86 iDCX96) to support you in debugging the code.

dingo35, 2025-02-13:
You are not the only one with an ABB meter, so we'll fix it, but it might take a while; I'm in 120V / 60Hz territory until end of March...

AlexT60, 2025-02-13:
Ok, I can live with v3.6.10.
In case I can be of any help let me know.

mstegen, 2025-03-20:
I finally tested this on my ABB B23 meter, and with a Lamp as load, it took a while, but the Charged kWh counter is increasing.

[image]

Could it be that you feed the power in from the bottom of the meter, and out on the top?

AlexT60, 2025-03-20:
If have openend the panel and checked the power connections. I feed the meter from the top, below the meter is the outlet to my external box.
[image]

For the type of ABB meter I have attached another picture.
[image]

The meter is manufactured in 2018 but works flawlessly with v3.6.10. I use it every day to be able to keep track of the consumption and to be able to bill back.

I will double check the A/B wiring, for that I need to open the external box (an old Tesla housing). I doubt if that is the case, v3.6.10 should fail too.

mstegen, 2025-03-21:
Thanks for the pictures. By comparing the A/B connections on the Sensorbox and meter, i think they are connected correctly.
Will try to investigate this further.

mstegen, 2025-03-21:
Can you check if the latest compiled firmware shows any improvements? I have attached it.
I noticed that it has a ~12ms longer delay between modbus messages, then v3.7.3

[firmware.zip](https://github.com/user-attachments/files/19390895/firmware.zip)

mstegen, 2025-03-21:
Great! Then your issue should be solved when we release v3.7.4

mstegen, 2025-03-22:
Can you please test this new firmware, i noticed that the extra delay of the previous firmware was caused by me leaving some debugging data in the code.

This latest firmware has 50ms silent time between messages, and reworked modbus code. should help with the ABB meter.

[firmware.zip](https://github.com/user-attachments/files/19404437/firmware.zip)

AlexT60, 2025-03-22:
Installed and tested the attached version. Did also a cold boot.
Monitored for a while the readings of the ABB EV-meter, still working perfect.

[image]

Thanks again for the effort to solve this issue!

AlexT60, 2025-03-22:
That was the wrong picture attached!
This is the correct one
[image]

dingo35, 2025-04-01:
Should be fixed in v3.7.4

closed 2025-04-01 by dingo35

AlexT60, 2025-04-01:
Installed 3.7.4, issue is solved!
Thanks again