dingo35/SmartEVSE-3.5 #174: PRIORITY mode not found when configuring a Master/Slave

PhilippeRigaux, 2025-02-02:
Hello,

Did you include this option in 3.7.3? I can't find it and I really want to charge the Master first and then the Slave.

Thx

Philippe

> I think I have a better solution for you. Mind you, this is only alpha version, but you are one of our advanced users:
> 
> [4db803fa50.zip](https://github.com/user-attachments/files/16176423/4db803fa50.zip)
> 
> 
> 
> With this version, if you select "PWR SHARE = Master", you get a new option:
> 
> [image]
> 
> 
> 
> I think if you select NodeNr as priority strategy, this is exactly what you want? 

 _Originally posted by @dingo35 in [#84](https://github.com/dingo35/SmartEVSE-3.5/issues/84#issuecomment-2222741335)_

dingo35, 2025-02-02:
No it is postponed till after v4, because there were too many problems integrating it; you can find it in the calcbalancedcurrent5 branch.

PhilippeRigaux, 2025-02-02:
OK, just to be sure I make the things right, I connected A,B and GND to both evse modules. Meter values are send via API (Mqtt) should I send all the values to each modules or only to the master?

It would be great to get this functionality in the Master branch, what will I loose in the calcbalanced one?

dingo35, 2025-02-03:
Meter values should only be sent to the Master; git log will show you the differences between the branches.

PhilippeRigaux, 2025-02-03:
OK, I have to send EVmeter to both or I got a communication error.. 

To charge one EV before the other I turn one OFF and let the other charge and when it is done I put the second in Solar mode too.

Thx for your support.

dingo35, 2025-02-03:
You should have one MainsMeter(feed) at the Master, and optional one EVMeter(feed) per SmartEVSE. It does not make sense to feed both SmartEVSEs the same EVMeter values...

PhilippeRigaux, 2025-02-03:
Hum I was thinking that the EV meter is used to protect the sub panel?

Like in https://github.com/dingo35/SmartEVSE-3.5/blob/master/docs/installation.md

My installation is Mainbreaker 25A
SubPanel 40A
EVs breaker 32A both EVSE are on the same 32A and as such get the same EV Meter data.

PhilippeRigaux, 2025-02-03:
Or should I disable the EVmeter on the Slave and send it only to the Master?

dingo35, 2025-02-03:
If you use the subpanel configuration you can guard it by the MaxCircuit parameter, and you should send that info to the master: the slave should not have and EVmeter configured.

PhilippeRigaux, 2025-02-03:
OK, EVMeter to disabled on the slave and the master receive all Mqtt informations, this make sense. 

Again. thanks a lot for all these informations, I did read the code but it is pretty a big task to understand every bit of this ;)

I'm closing the issue as it basically do what I need..

closed 2025-02-03 by PhilippeRigaux