dingo35/SmartEVSE-3.5 #175: possibility to set a start time in the past

XsniferX, 2025-02-05:
Hi

is there a possibility to set a progammable start time in the past ?

for example i usually allow charging between 22h and 6h (off peak hours in france)

but via the web ui if it is past 22h i can only set a start time lets say 22h01 to next day 06h00 and check the daily box.

it would be great to be able to set a start time of 22h for the current day event if 22h is passed. 

Regards,

dingo35, 2025-02-05:
Just set the timers once before 2200 with the daily checkbox on, and it should charge only between 2200 and 0600 

Is it too much to ask to do this one time action at one time before 2200?

XsniferX, 2025-02-05:
**Hi / Hello**,

Yes thank you i know the procedure to set the times. Many thanks.

And to answer you, yeah some days, mostly every odd days it is indeed too much to ask...

Now going back to the **ORIGINAL FEATURE REQUEST** i use the switch to toggle between solar mode and smart mode so i need to set it again and again especially when some days of the week i get back home after 2200.

Obviously, it is not a difficult task to do.

And since we can post feature request i gave it a shot. 

But if it too much to ask... no issue at all.

There is no need to be condescending.

closed 2025-02-06 by dingo35