dingo35/SmartEVSE-3.5 #177: Charging session never stops in Solar mode

RemixQ, 2025-02-10:
**Describe the bug**
I encountered an issue during several charging sessions: it never stops even after 80 minutes at the minimum current (6A) and grid import. The solar stop timer always stays at 0.

Context:
- SmartEVSE with firmware 3.7.3
- Single phase
- Solar mode
- Main meter pushed from Home assistant to MQTT topic each 3s.
- Solar stop time: 10m

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar?

SOLAR mode

3. What steps do you take so the problem arises?

Settings:
- mode SOLAR
- Solar max import at 0
- Solar start current at 6
- Solar stop time at 10

I stopped manually the charge at 16:23:

[image]

[image]

[image]

[image]

4. A clear and concise description of what you expected to happen; expected Charging Current?

My understanding is that stopping the session in solar mode should be related to the time where charge current is at the minimum current value (which is not working in my case), but I would expect it to be also (or only) related to the time when the mains current value is higher than the solar max import value.
> (if "Charge_current" = "Current_min" and "Total_mains" is >= "Solar_max_import") for more than "Solar_stop_time"

dingo35, 2025-02-10:
Could you upload your /settings please?

dingo35, 2025-02-10:
Also: it might be that single phase is not detected; could you change Contactor2 (C2) setting to "Always Off" and see if that changes behaviour?

RemixQ, 2025-02-10:
Yes, here is the configuration (after setting C2 to "Always Off", let's see today or tomorrow if it works):
```
{
  "version": "v3.7.3",
  "serialnr": 5261,
  "mode": "SOLAR",
  "mode_id": 2,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "XXXX",
    "rssi": -79,
    "bssid": "FF:FF:FF:FF:FF:FF"
  },
  "evse": {
    "temp": 20,
    "temp_max": 65,
    "connected": true,
    "access": true,
    "mode": 2,
    "loadbl": 0,
    "pwm": 1024,
    "custombutton": false,
    "solar_stop_timer": 0,
    "state": "Charging Stopped - Waiting for Solar",
    "state_id": 9,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 60,
    "override_current": 0,
    "current_min": 6,
    "current_max": 32,
    "current_main": 32,
    "current_max_circuit": 16,
    "current_max_sum_mains": 600,
    "max_sum_mains_time": 0,
    "solar_max_import": 0,
    "solar_start_current": 6,
    "solar_stop_time": 10,
    "enable_C2": "Always Off",
    "mains_meter": "API",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0,
    "lcdlock": 0
  },
  "mqtt": {
    "host": "192.168.1.68",
    "port": 1883,
    "topic_prefix": "SmartEVSE-5261",
    "username": "smartevse",
    "password_set": true,
    "status": "Connected"
  },
  "ocpp": {
    "mode": "Disabled",
    "backend_url": "",
    "cb_id": "",
    "auth_key": "",
    "auto_auth": "Disabled",
    "auto_auth_idtag": "",
    "status": "Disconnected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Disabled",
    "address": 12,
    "import_active_power": 0,
    "total_kwh": 0,
    "charged_kwh": 0,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 7,
    "L1": 7,
    "L2": 0,
    "L3": 0,
    "last_data_update": 1739173473,
    "original_data": {
      "TOTAL": 7,
      "L1": 7,
      "L2": 0,
      "L3": 0
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  },
  "color": {
    "off": {
      "R": 0,
      "G": 0,
      "B": 0
    },
    "normal": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "smart": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "solar": {
      "R": 255,
      "G": 170,
      "B": 0
    },
    "custom": {
      "R": 0,
      "G": 0,
      "B": 255
    }
  }
}```

RemixQ, 2025-02-14:
I confirm changing Contactor2 (C2) setting to "Always Off" solve the issue, the solar stop time counter start normally (counting down from 600).

Thank you for your quick help.

closed 2025-02-14 by dingo35

eavyon, 2025-03-28:
Today I encountered the same issue with the same settings as @RemixQ . I have a 3-phase system so I put the C2 contactor on AUTO so it should change from 3F to 1F when charging mode is SOLAR and not enough current for 3F. While that does work, the displays keeps saying 3F and when current drops below 6A it doesnt turn off by set timer. Only when C2 is set to "Solar Off" or "Always Off" it behaves as it should.

So it looks like there is an exception that it doesnt calculate when C2 is on AUTO and never shuts off. @dingo35 

Tested another thing: Minimum solar: 0 and C2 on "Always On" Makes it charge in in 1F but display shows 3F.(Well now it always charges on 1F also in NORMAL and SMART.) And yes, minimum solar 0 will ofcourse never shuts of solar charging.

Another test: I rebooted smartevse and setup SMART and C2 "Always on" from the device itself. It started charging in 3F but the display said 1F.

Wrapping up, there is something up with the logic for choosing different charging modes, using C2 contactor and calculating current. Possibly its better to make a new issue after testing more and or me looking into the current logic for this. Is there a flow diagram of this logic?

dingo35, 2025-03-28:
This should be taken care of when handling issue #153 

> 

> [](url)