dingo35/SmartEVSE-3.5 #188: Feeding currents from esphome no longer works on 3.7. 4 (from 3.5.1)

HA-TB303, 2025-04-04:
**Describe the bug**
Not sure if it's even a bug....

**Upload your config**
```
{
    "version": "v3.7.4",
    "serialnr": XXXX,
    "mode": "NORMAL",
    "mode_id": 1,
    "car_connected": true,
    "wifi": {
        "status": "WL_CONNECTED",
        "ssid": "XXXXXXXXXXXX",
        "rssi": -25,
        "bssid": "XXXXXXXXXXXXX"
    },
    "evse": {
        "temp": 34,
        "temp_max": 65,
        "connected": true,
        "access": true,
        "mode": 0,
        "loadbl": 0,
        "pwm": 169,
        "custombutton": false,
        "solar_stop_timer": 0,
        "state": "Charging",
        "state_id": 2,
        "error": "None",
        "error_id": 0,
        "rfid": "Not Installed"
    },
    "settings": {
        "charge_current": 100,
        "override_current": 0,
        "current_min": 6,
        "current_max": 10,
        "current_main": 80,
        "current_max_circuit": 10,
        "current_max_sum_mains": 600,
        "max_sum_mains_time": 0,
        "solar_max_import": 0,
        "solar_start_current": 10,
        "solar_stop_time": 10,
        "enable_C2": "Not present",
        "mains_meter": "API",
        "starttime": 0,
        "stoptime": 0,
        "repeat": 0,
        "lcdlock": 0
    },
    "mqtt": {
        "host": "192.168.228.200",
        "port": 1883,
        "topic_prefix": "SmartEVSE-XXXX",
        "username": "smartevse2",
        "password_set": true,
        "status": "Connected"
    },
    "ocpp": {
        "mode": "Disabled",
        "backend_url": "",
        "cb_id": "",
        "auth_key": "",
        "auto_auth": "Disabled",
        "auto_auth_idtag": "",
        "status": "Disconnected"
    },
    "home_battery": {
        "current": 0,
        "last_update": 0
    },
    "ev_meter": {
        "description": "Eastron3P",
        "address": 100,
        "import_active_power": 6.800000191,
        "total_kwh": 4362.100098,
        "charged_kwh": 6.5,
        "currents": {
            "TOTAL": 299,
            "L1": 99,
            "L2": 100,
            "L3": 100
        },
        "import_active_energy": 4362.100098,
        "export_active_energy": 0
    },
    "mains_meter": {
        "import_active_energy": 0,
        "export_active_energy": 0
    },
    "phase_currents": {
        "TOTAL": 300,
        "L1": 100,
        "L2": 100,
        "L3": 100,
        "last_data_update": 1743756490,
        "original_data": {
            "TOTAL": 300,
            "L1": 100,
            "L2": 100,
            "L3": 100
        }
    },
    "backlight": {
        "timer": 120,
        "status": "ON"
    },
    "color": {
        "off": {
            "R": 0,
            "G": 0,
            "B": 0
        },
        "normal": {
            "R": 0,
            "G": 255,
            "B": 0
        },
        "smart": {
            "R": 0,
            "G": 255,
            "B": 0
        },
        "solar": {
            "R": 255,
            "G": 170,
            "B": 0
        },
        "custom": {
            "R": 0,
            "G": 0,
            "B": 255
        }
    }
}
```

**Expected behavior**
I used to feed the currents from esphome like this:

```
interval:
  - interval: 4sec
    then:
       - http_request.post:
           headers:
             Content-Length: 0
           url: !lambda |-
               std::string url;
               float l1 = id(returnedl1).state > 0 ? - id(currentl1).state : id(currentl1).state;
               float l2 = id(returnedl2).state > 0 ? - id(currentl2).state : id(currentl2).state;
               float l3 = id(returnedl3).state > 0 ? - id(currentl3).state : id(currentl3).state;
               url.append("http://192.168.228.111/currents?L1=");
               url.append(to_string(l1*10));
               url.append("&L2=");
               url.append(to_string(l2*10));
               url.append("&L3=");
               url.append(to_string(l3*10));
               return url.c_str();
```
This still works for v3.5.1 but on 3.7.4 it fails with this error in esphome:

```
[11:02:13][E][http_request.arduino:119]: HTTP Request failed; URL: http://192.168.228.112/currents?L1=350.000000&L2=340.000000&L3=280.000000; Code: 411
[11:02:13][E][component:164]: Component http_request set Error flag: unspecified
[11:02:13][D][http_request.arduino:125]: Content-Length: 0
[11:02:15][W][component:237]: Component interval took a long time for an operation (612 ms).
[11:02:15][W][http_request.arduino:112]: HTTP Request failed; URL: http://192.168.228.113/currents?L1=350.000000&L2=340.000000&L3=280.000000; Error: connection refused
[11:02:15][W][component:237]: Component interval took a long time for an operation (1017 ms).
[11:02:15][W][component:238]: Components should block for at most 30 ms.
[11:02:16][W][http_request.arduino:112]: HTTP Request failed; URL: http://192.168.228.114/currents?L1=350.000000&L2=340.000000&L3=280.000000; Error: read Timeout
[11:02:16][W][component:237]: Component interval took a long time for an operation (1133 ms).
[11:02:16][W][component:238]: Components should block for at most 30 ms.
[11:02:16][D][http_request.arduino:125]: Content-Length: 80
[11:02:16][W][component:237]: Component interval took a long time for an operation (127 ms).
[11:02:16][W][component:238]: Components should block for at most 30 ms.
[11:02:17][E][http_request.arduino:119]: HTTP Request failed; URL: http://192.168.228.112/currents?L1=350.000000&L2=340.000000&L3=280.000000; Code: 411
[11:02:17][D][http_request.arduino:125]: Content-Length: 0
[11:02:17][W][component:237]: Component interval took a long time for an operation (219 ms).
[11:02:17][W][component:238]: Components should block for at most 30 ms.
[11:02:18][D][http_request.arduino:125]: Content-Length: 80
[11:02:18][W][component:237]: Component interval took a long time for an operation (206 ms).
[11:02:18][W][component:238]: Components should block for at most 30 ms.
[11:02:18][E][component:176]: Component http_request cleared Error flag
[11:02:19][W][http_request.arduino:112]: HTTP Request failed; URL: http://192.168.228.114/currents?L1=350.000000&L2=340.000000&L3=280.000000; Error: read Timeout
[11:02:19][E][component:164]: Component http_request set Error flag: unspecified
[11:02:19][W][component:237]: Component interval took a long time for an operation (1108 ms).
[11:02:19][W][component:238]: Components should block for at most 30 ms.
[11:02:20][D][http_request.arduino:125]: Content-Length: 80
```

CURL does work:

```
% curl -X POST "http://192.168.228.112/currents?L1=100&L2=100&L3=100" -d ''
{"original":{"L":100,"":100,"DIMMED":100},"L":100,"":100,"DIMMED":100,"TOTAL":300}
```

Am I doing something wrong?

dingo35, 2025-04-04:
I think specifiying your L1-3 currents with a 1 microAmpère significancy might cause the problem?

If curl works, the problem is caused by the curl-building script....

HA-TB303, 2025-04-04:
Ok, but that's how I get the data from the P1 reader (also esphome)

HA-TB303, 2025-04-04:
```
~% curl -X POST "http://192.168.228.112/currents?L1=350.000000&L2=350.000000&L3=350.000000" -d ''
{"original":{"L":350,"":350,"DIMMED":350},"L":350,"":350,"DIMMED":350,"TOTAL":1050}
```
Works...
Strange thing is that the 3.5.1 version has no issues. My other 3 chargers running 3.5.1. still work, receiving the exact same post.

HA-TB303, 2025-04-04:
esphome config:
```
interval:
  - interval: 4sec
    then:
       - http_request.post:
           headers:
             Content-Length: 0
           url: !lambda |-
               std::string url;
               float l1 = id(returnedl1).state > 0 ? - id(currentl1).state : id(currentl1).state;
               float l2 = id(returnedl2).state > 0 ? - id(currentl2).state : id(currentl2).state;
               float l3 = id(returnedl3).state > 0 ? - id(currentl3).state : id(currentl3).state;
               url.append("http://192.168.228.111/currents?L1=");
               url.append(to_string(l1*10));
               url.append("&L2=");
               url.append(to_string(l2*10));
               url.append("&L3=");
               url.append(to_string(l3*10));
               return url.c_str();
       - delay: 0.5s
       - http_request.post:
           url: !lambda |-
               std::string url;
               float l1 = id(returnedl1).state > 0 ? - id(currentl1).state : id(currentl1).state;
               float l2 = id(returnedl2).state > 0 ? - id(currentl2).state : id(currentl2).state;
               float l3 = id(returnedl3).state > 0 ? - id(currentl3).state : id(currentl3).state;
               url.append("http://192.168.228.112/currents?L1=");
               url.append(to_string(l1*10));
               url.append("&L2=");
               url.append(to_string(l2*10));
               url.append("&L3=");
               url.append(to_string(l3*10));
               return url.c_str();
       - delay: 0.5s
       - http_request.post:
           url: !lambda |-
               std::string url;
               float l1 = id(returnedl1).state > 0 ? - id(currentl1).state : id(currentl1).state;
               float l2 = id(returnedl2).state > 0 ? - id(currentl2).state : id(currentl2).state;
               float l3 = id(returnedl3).state > 0 ? - id(currentl3).state : id(currentl3).state;
               url.append("http://192.168.228.113/currents?L1=");
               url.append(to_string(l1*10));
               url.append("&L2=");
               url.append(to_string(l2*10));
               url.append("&L3=");
               url.append(to_string(l3*10));
               return url.c_str();
       - delay: 0.5s
       - http_request.post:
           url: !lambda |-
               std::string url;
               float l1 = id(returnedl1).state > 0 ? - id(currentl1).state : id(currentl1).state;
               float l2 = id(returnedl2).state > 0 ? - id(currentl2).state : id(currentl2).state;
               float l3 = id(returnedl3).state > 0 ? - id(currentl3).state : id(currentl3).state;
               url.append("http://192.168.228.114/currents?L1=");
               url.append(to_string(l1*10));
               url.append("&L2=");
               url.append(to_string(l2*10));
               url.append("&L3=");
               url.append(to_string(l3*10));
               return url.c_str();

```
Charger 192.168.228.111 = v3.5.1 -> works
Charger 192.168.228.112 = v3.7.4 -> fails
Charger 192.168.228.113 = v3.51 -> works
Charger 192.168.228.114 = v3.51 -> works

dingo35, 2025-04-04:
You are absolutely right, this is your problem; from the documentation REST_API.md:

"NOTE:
In the http world, GET parameters are passed like this:
curl -X GET http://ipaddress/endpoint?param1=value1&param2=value2
and POST parameters are passed like this:
curl -X POST http://ipaddress/endpoint -d 'param1=value1' -d 'param2=value2' -d ''

Now in the ESP world, we all have picked up the habit of using the GET way of passing parameters also for POST commands. SmartEVSE development not excluded....
From version v3.6.0 on, instead of using the Arduino Core webserver libraries, we are now using the Mongoose webserver, which is broadly used. This webserver however sticks to the "normal" http standards.

This means that if you POST a request to SmartEVSE > 3.6.0, the webserver will be waiting for the -d data until it times out (or you ctrl-C your curl command). -d ''
You can prevent this by adding
'''
-d ''
'''

to your curl POST command. -d ''

dingo35, 2025-04-04:
....so just add a url.append(" -d ' ' "); to your script; untested, so you have to fiddle yourself with the quoting :-)

HA-TB303, 2025-04-04:
Thanks, but I'm using esphome. Not a script. I shared this config with you: https://github.com/dingo35/SmartEVSE-3.5/blob/master/integrations/home-assistant/esphome/ET-SM01%20smart%20meter%20module.yaml

But this no longer works. I'm limited to: https://esphome.io/components/http_request.html

I am not sure if and how I could work around this with esphome.

dingo35, 2025-04-04:
The 411 error code is a complaint about the content length; so you only have to convince esphome.io to send either and empty data header (e.g. an empty json parameter) or add a Content Length header to the post request, setting it to 0.

I cannot change mongoose's behaviour, so you will have to change this on the input side Im afraid...

HA-TB303, 2025-04-04:
Thanks, following your advise, I added:
```
       - http_request.post:
           headers:
             Content-Length: 0
           url: !lambda |-
          ....
```

And it now works again.

dingo35, 2025-04-04:
Great! Do I need to update anything in the repo?

HA-TB303, 2025-04-04:
No oddly enough the version in your repo already has this header parameter, even though the original file I shared back then did not. I guess someone must have added that ad some point.

dingo35, 2025-04-04:
None of us is as smart as all of us!