dingo35/SmartEVSE-3.5 #189: Homewizard P1 meter not working

Imaginous, 2025-04-05:
**Describe the bug**
I set the meter to Homewizard P1, but it doesn't work.

**Upload your config**
`{"version":"v3.8.0","serialnr":6972,"mode":"OFF","mode_id":0,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"Imaginous","rssi":-50,"bssid":"7C:10:C9:CD:FE:A8"},"evse":{"temp":25,"temp_max":55,"connected":false,"access":0,"mode":1,"loadbl":0,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"Communication Error","error_id":2,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":27,"current_main":33,"current_max_circuit":32,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":7,"solar_stop_time":31,"enable_C2":"Always Off","mains_meter":"HmWzrd P1","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{"host":"mqtt.imaginous.nl","port":1883,"topic_prefix":"SmartEVSE","username":"mqtt","password_set":true,"status":"Connected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0,"host":"p1meter-04636A.local"},"phase_currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0,"last_data_update":0,"original_data":{"TOTAL":0,"L1":0,"L2":0,"L3":0}},"backlight":{"timer":119,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}`

**Expected behavior**
It to work ;)

I can ping *p1meter-04636A.local* and it responds with the correct IP.

The version of th SmartEVSE is V3.8.0
The Home Wizard P1 meter (HWE-P1-G1) V5.19

dingo35, 2025-04-05:
Could you flash the debug version and make a telnet log? 

Look for entries containing the string "HWP1"

Imaginous, 2025-04-05:
This is want I found in the debug log:

`(homewizard_loop)(C1) homewizard_loop(): start HomeWizrd P1 reading.(getMainsFromHomeWizardP1)(C1) getMainsFromHWP1(): invocation
(discoverHomeWizardP1)(C1) discoverHWP1(): Using cached host 'p1meter-04636A.local'.
(getMainsFromHomeWizardP1)(C1) getMainsFromHWP1(): connect to URL http://p1meter-04636A.local/api/v1/data
(getMainsFromHomeWizardP1)(C1) getMainsFromHomeWizardP1(): required JSON fields 'active_current_l[1..3]_a' not found`

This is the data at *http://p1meter-04636A.local/api/v1/data*:
`{"wifi_ssid":"Imaginous","wifi_strength":86,"smr_version":50,"meter_model":"Kaifa AIFA-METER","unique_id":"0000303539303030303032393936380000","active_tariff":1,"total_power_import_kwh":7412.085,"total_power_import_t1_kwh":4283.482,"total_power_import_t2_kwh":3128.603,"total_power_export_kwh":6551.330,"total_power_export_t1_kwh":1930.678,"total_power_export_t2_kwh":4620.652,"active_power_w":-2725.000,"active_power_l1_w":-2725.000,"active_voltage_l1_v":238.400,"active_current_a":11.430,"active_current_l1_a":-11.430,"voltage_sag_l1_count":8.000,"voltage_swell_l1_count":0.000,"any_power_fail_count":0.000,"long_power_fail_count":0.000,"total_gas_m3":1795.627,"gas_timestamp":250405135009,"gas_unique_id":"0000303738353635363033323134350000","external":[{"unique_id":"0000303738353635363033323134350000","type":"gas_meter","timestamp":250405135009,"value":1795.627,"unit":"m3"}]}`

I have obfuscated the meter IDs.

dingo35, 2025-04-05:
Ok that could be!

What does the JSON data look like when you surf to:
http://p1meter-04636A.local/api/v1/data

dingo35, 2025-04-05:
Im sorry the screen of my phone is too small :-( 

Just to be sure: So the Homewizard does NOT have a special 1 phase model, but  it leaves the L2 and L3 data out if it doesnt receive those telegrams from P1?

What brand/model is your electricity meter (that provides the P1 data)? Is it 1 phase or is it 3phase that is wired to only 1 ?

(Not really familiar with the Homewizard, nor with single phase installations; I always have them replaced by 3x25A feeds ...)

Imaginous, 2025-04-05:
It's all the same model. It's just my mains meter is 1 phase.

dingo35, 2025-04-05:
Could you please test this version:

[b5eb79e.zip](https://github.com/user-attachments/files/19615675/b5eb79e.zip)

Imaginous, 2025-04-05:
Will try it tomorrow, not at home now.

Keep you posted.

Imaginous, 2025-04-06:
Tested this morning, still no success.

Telnet is disabled in this version?
Couldn't connect a session.

Another minor issue with the virtual buttons on the display in a chrome browser on a phone.
They register a long or multi press and scrolls trough several menu options. Using an Android phone.

dingo35, 2025-04-06:
Telnet is ENABLED, sometimes the ESP32 needs some time to establish all the connections after reboot, perhaps you tried to soon after reboot?

Please don't mix issues, the display-in-the-browser issue is addressed PR #178 .

Imaginous, 2025-04-06:
Sorry for mixing up bugs.

I powered off en restarted the SmartEvse. The P1 meter is now working. But it doesn't use the first decimal, so it shows 4.0A while the current in the data was 4.3A or 4.6A. So it's rounded down which is better then rounding away of 0.5.

Also telnet works now again.

But visually it would be better to take the first digit in account.

That's my 2 cents.

dingo35, 2025-04-06:
Could you test this version, should have 1 decimal accuracy:

[b43892c5e5.zip](https://github.com/user-attachments/files/19620370/b43892c5e5.zip)

Also: do you mind if your JSON api data is included in our source code as test data?

Imaginous, 2025-04-06:
This version shows the first decimal, super!

I don't mind using my data snippet, maybe set the meter number to all 0's.

dingo35, 2025-04-06:
Set all the meter nrs to 0's ! Thx for your input!

closed 2025-04-06 by dingo35

kozmoz, 2025-04-07:
Never took into account that in the case of a single-phase meter, the keys for the other phases are not present in the JSON at all. Great that this was resolved so quickly. And I can confirm that it still works correctly with three-phase as well.

Imaginous, 2025-04-07:
We have to help each other out as developers :)