dingo35/SmartEVSE-3.5 #192: HomeWizard P1 meter not reconnecting

Imaginous, 2025-04-10:
**Describe the bug**
If the WiFi drops, I updated the firmware of my accesspoint, the SmartEvse reconnects, but the P1 connections stays offline.

**Upload your config**
`{"version":"10:10:54 @Apr  6 2025","serialnr":6972,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Imaginous","rssi":-55,"bssid":"7C:10:C9:CD:FE:A8"},"evse":{"temp":25,"temp_max":55,"connected":true,"access":1,"mode":1,"loadbl":0,"pwm":102,"custombutton":false,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":27,"current_main":33,"current_max_circuit":32,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":7,"solar_stop_time":31,"enable_C2":"Always Off","mains_meter":"HmWzrd P1","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{"host":"mqtt.imaginous.nl","port":1883,"topic_prefix":"SmartEVSE","username":"mqtt","password_set":true,"status":"Connected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0,"host":"p1meter-04636A.local"},"phase_currents":{"TOTAL":-51,"L1":-51,"L2":0,"L3":0,"last_data_update":1744299252,"original_data":{"TOTAL":-51,"L1":-51,"L2":0,"L3":0}},"backlight":{"timer":0,"status":"OFF"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}`

**Upload a debug log**
`(homewizard_loop)(C1) homewizard_loop(): start HomeWizrd P1 reading.(getMainsFromHomeWizardP1)(C1) getMainsFromHWP1(): invocation
(discoverHomeWizardP1)(C1) discoverHWP1(): Using cached host 'p1meter-04636A.local'.
(getMainsFromHomeWizardP1)(C1) getMainsFromHWP1(): connect to URL http://p1meter-04636A.local/api/v1/data
(getMainsFromHomeWizardP1)(C1) getMainsFromHWP1(): Error on HTTP request (httpCode=-1), url=http://p1meter-04636A.local/api/v1/data.
`

**To Reproduce**
When using HomeWizard P1 as mainsmeter reboot your WiFi accesspoint/router.
SmartEvse reconnects, the P1 meter in the SmartEvse doesn't.

I can request the data via *http://p1meter-04636A.local/api/v1/data*, so the meter is online.

Reboot the SmartEvse and it directly works again.

**Expected behavior**
Restoring connection on WiFi failure.

dingo35, 2025-04-12:
I'm happy to pull this into the repo, but its not very handy to edit everything by hand. Why don't you just 
zip the few files you changed and upload them here; if you don't wait too long (the repo will change), and assuming you used a recently synced version of my repo, I can just unzip those files over my local repo and see what the diff is ...

Imaginous, 2025-04-12:
I will do that, will make another small change.
I will lower the poll time to 4000ms, so it can try 3 times within 10 seconds.

Had one failure this morning, but it reconnected automatically as planned.

Will take the latest version and redo my changes and upload the zip within an hour.

Imaginous, 2025-04-12:
Here is the latest version: [network.zip](https://github.com/user-attachments/files/19718455/network.zip)

Changed the scan time to 4000ms and the timeout to 3500.

dingo35, 2025-04-12:
Ok Rob I have the commit ready to push, I only removed some old lines you commented out because we think it is bad practice to keep old code lines in; we have git if we ever want to go back to the old code :-) 

I can see you changed the timeout to 3500ms, but I must be overlooking the scan time modification? Is it in network.cpp ?

Imaginous, 2025-04-12:
I'm Michel btw :)

Will remove old comment next time, sorry ;)

dingo35, 2025-04-12:
Sorry Michel working on multiple issues simultaneously, and as we all know, men cannot multitask :-)

Do I need to change "interval" in esp32.cpp from 5000 to 4000 , or is that not the scan time you mentioned?

Imaginous, 2025-04-12:
I forgot to include the file I changed `constexpr unsigned long interval = 4000; // 4 seconds` in network.cpp.

[network.zip](https://github.com/user-attachments/files/19718908/network.zip)

dingo35, 2025-04-12:
Ok committed as #73eec4d63c8

Imaginous, 2025-04-14:
@dingo35 I made some more optimizations...

Now my "stupid" question. I pulled the GIT repository made some changes, but obviously I can't push them to your repository.
How can I push these changes, never done this before.

I changed the comments and lowered the scan time to 1950ms, so it can scan 5 times within the 10 second Mains Meter timeout.
Lowered the request timeout to 1500 so it fits within the 1950ms scan time.

Added the changed files here, pulled 20 minutes ago. 
But it would be great to learn how I can contribute directly. I'm an old school programmer... almost 50... and still learning :)

[src.zip](https://github.com/user-attachments/files/19730738/src.zip)

dingo35, 2025-04-14:
Not a stupid question, it is IMHO not very intuitive; you have to create a Pull Request (PR).
When you push your changes from your local repo to your Github repo ("git push"), on your Github webpage you'll see a green button "compare & pull request".
If you press this button you will create a PR in MY repo, which I then can review, accept, decline.

 It is a great tool but takes some learning; for details Google is your friend...

dingo35, 2025-05-01:
Fixed by 73eec4d63c863946aad0a66b14ba705a39fd8116

closed 2025-05-01 by dingo35