dingo35/SmartEVSE-3.5 #199: 3.7.4 solar charging does not start count down

BarryH120, 2025-04-15:
**Describe the bug**
Solar charging does nog start count down when solar power too low

**Upload your config**
{"version":"v3.7.4","serialnr":7416,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"B2","rssi":-60,"bssid":"F8:0D:A9:09:B9:19"},"evse":{"temp":35,"temp_max":65,"connected":true,"access":true,"mode":2,"loadbl":0,"pwm":102,"custombutton":false,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":25,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":0,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":10,"enable_C2":"Always On","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{"host":"","port":1883,"topic_prefix":"SmartEVSE-7416","username":"","password_set":false,"status":"Disconnected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":20,"L1":-30,"L2":0,"L3":50,"last_data_update":1744737184,"original_data":{"TOTAL":20,"L1":-30,"L2":0,"L3":50}},"backlight":{"timer":119,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Solar
2. What steps do you take so the problem arises? solar charging - 1F, starts at 6A, with no import. starts as soon as enough feed in, but does not start the 10 minute countdown once the solar feed in drops. Seems to occur since update to 3.8.0, but after downgrade to 3.7.4 still occurs
3. 

**Expected behavior**
I expect the count down to start once solar power below 6A

**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

dingo35, 2025-04-15:
Since you seem to be charging in single phase, could you please set EnableC2 to Always_Off, and see if that solves the problem?

BarryH120, 2025-04-15:
Just did.
set smartevse to normal to start charging, then to solar and it immediately starts with the countdown

dingo35, 2025-04-16:
So problem solved?

The single phase detection is a known problem , see issue #153

BarryH120, 2025-04-16:
Yes

closed 2025-04-16 by BarryH120

dingo35, 2025-04-16:
Done, even added an extra topic in the docs to explain. Thx for the sugggestion!