dingo35/SmartEVSE-3.5 #200: v3.8.0 Mode switches to N/A

vdweegen, 2025-04-15:
**Describe the bug**
I have two SmartEVSE's running v.3.8.0 running in Master/Slave mode. I have a custom integration with Domoticz (triggers based on shell scripts and feedback based on MQTT). The following schedule is used while settings modes:
00:00 - 07:00 - SmartEVSE1 (master): Off, SmartEVSE2 (slave): SMART
07:00 - 00:00 - both SmartEVSE's: SOLAR

As of update v.3.8.0 sometimes one (or both) SmartEVSE's expectantly change mode to N/A. The display shows Waiting for Solar but the behaviour is more like mode NORMAL. I've attached some screenshots, two from the SmartEVSE homepages after a mode N/A just happened and a screenshot from my Pushover app which notifies me of mode changes of my SmartEVSE's.

**Upload your config**
SmartEVSE 1 (master)
`
{
  "version": "v3.8.0",
  "serialnr": 6678,
  "mode": "SOLAR",
  "mode_id": 2,
  "car_connected": false,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "redacted",
    "rssi": -61,
    "bssid": "redacted"
  },
  "evse": {
    "temp": 28,
    "temp_max": 65,
    "connected": false,
    "access": 1,
    "mode": 2,
    "loadbl": 1,
    "pwm": 1024,
    "custombutton": false,
    "solar_stop_timer": 0,
    "state": "Ready to Charge",
    "state_id": 0,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 0,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 25,
    "current_max_circuit": 16,
    "current_max_sum_mains": 600,
    "max_sum_mains_time": 0,
    "solar_max_import": 2,
    "solar_start_current": 3,
    "solar_stop_time": 10,
    "enable_C2": "Always Off",
    "mains_meter": "Sensorbox",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0,
    "lcdlock": 1
  },
  "mqtt": {
    "host": "redacted",
    "port": 1883,
    "topic_prefix": "SmartEVSE-15494",
    "username": "",
    "password_set": false,
    "status": "Connected"
  },
  "ocpp": {
    "mode": "Disabled",
    "backend_url": "",
    "cb_id": "",
    "auth_key": "",
    "auto_auth": "Disabled",
    "auto_auth_idtag": "",
    "status": "Disconnected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Eastron3P",
    "address": 101,
    "import_active_power": 0,
    "total_kwh": 3049,
    "charged_kwh": 0,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 3049,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": -2,
    "L1": 14,
    "L2": -25,
    "L3": 9,
    "last_data_update": 1744739024,
    "original_data": {
      "TOTAL": -2,
      "L1": 14,
      "L2": -25,
      "L3": 9
    }
  },
  "backlight": {
    "timer": 0,
    "status": "OFF"
  },
  "color": {
    "off": {
      "R": 0,
      "G": 0,
      "B": 0
    },
    "normal": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "smart": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "solar": {
      "R": 255,
      "G": 170,
      "B": 0
    },
    "custom": {
      "R": 0,
      "G": 0,
      "B": 255
    }
  }
}
`

SmartEVSE2 (slave)
`
{
  "version": "v3.8.0",
  "serialnr": 7130,
  "mode": "SOLAR",
  "mode_id": 2,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "redacted",
    "rssi": -75,
    "bssid": "redacted"
  },
  "evse": {
    "temp": 32,
    "temp_max": 65,
    "connected": true,
    "access": 1,
    "mode": 2,
    "loadbl": 2,
    "pwm": 1024,
    "custombutton": false,
    "solar_stop_timer": 0,
    "state": "Charging Stopped - Waiting for Solar",
    "state_id": 9,
    "error": "None",
    "error_id": 0,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 0,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 25,
    "current_max_circuit": 16,
    "current_max_sum_mains": 600,
    "max_sum_mains_time": 0,
    "solar_max_import": 2,
    "solar_start_current": 3,
    "solar_stop_time": 10,
    "enable_C2": "Always Off",
    "mains_meter": "Sensorbox",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0,
    "lcdlock": 1
  },
  "mqtt": {
    "host": "redacted",
    "port": 1883,
    "topic_prefix": "SmartEVSE-46809",
    "username": "",
    "password_set": false,
    "status": "Connected"
  },
  "ocpp": {
    "mode": "Disabled",
    "backend_url": "",
    "cb_id": "",
    "auth_key": "",
    "auto_auth": "Disabled",
    "auto_auth_idtag": "",
    "status": "Disconnected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Eastron3P",
    "address": 110,
    "import_active_power": 0,
    "total_kwh": 1316.599976,
    "charged_kwh": 3.400000095,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 1316.599976,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": -1,
    "L1": 15,
    "L2": -25,
    "L3": 9,
    "last_data_update": 0,
    "original_data": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  },
  "color": {
    "off": {
      "R": 0,
      "G": 0,
      "B": 0
    },
    "normal": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "smart": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "solar": {
      "R": 255,
      "G": 170,
      "B": 0
    },
    "custom": {
      "R": 0,
      "G": 0,
      "B": 255
    }
  }
}
`

**Upload a debug log**
Unable to produce a debug log on short notice, will try to grab a telnet log if I have some free time

**To Reproduce**
Issue appears to arise at random moments.

I'm currently only semi-reliable way to trigger the issue the following way:
1. Switch both SmartEVSE's to OFF
2. Switch SmartEVSE2 (slave) to SMART (SmartEVSE1 remains OFF)
3. Switch SmartEVSE2 (slave) to SOLAR

I'm unable to consistently trigger the issue. Also, it appears that when the issue is NOT triggered both SmartEVSE's start charging in solar mode regardless of reaching the solar_start_current value.

**Expected behavior**
Both SmartEVSE's change to expected modes and behave accordingly.

**Screenshots**
[image]
[image]

- Pushover messages when issues was triggered
[image]

- SmartEVSE2 (slave) mode SOLAR but charging when there is no generated power
[image]

dingo35, 2025-04-16:
Ok I can not reproduce this; to find out what is going on, I need to know the values of both "mode" and "mode_id" in your settings whenever the N/A appears on your dashboard.

Your current settings seem to be ok; could you, whenever the N/A appears in your dashboard, IMMEDIATELY (before the 5s refresh page kicks in) press the "Raw" button and post the output here?

vdweegen, 2025-04-16:
Here we go. This morning at 07:00 an automation script set smartevse2 (slave) to SOLAR, which in master/slave configuration also changed smartevse1 (master) to SOLAR. Both devices immediately switched to mode N/A. I just made a screenshot and dumped the raw settings (both devices are still in mode N/A almost 6 hours later). You'll find the screenshot and the settinga attached.

`
{"version":"v3.8.0","serialnr":6678,"mode":"N/A","mode_id":-1,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"redacted","rssi":-65,"bssid":"redacted"},"evse":{"temp":23,"temp_max":65,"connected":false,"access":1,"mode":186,"loadbl":1,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":2,"solar_start_current":3,"solar_stop_time":10,"enable_C2":"Always Off","mains_meter":"Sensorbox","starttime":0,"stoptime":0,"repeat":0,"lcdlock":1},"mqtt":{"host":"redacted","port":1883,"topic_prefix":"SmartEVSE-15494","username":"","password_set":false,"status":"Connected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":101,"import_active_power":0,"total_kwh":3049,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":3049,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":0,"L1":57,"L2":-46,"L3":-11,"last_data_update":1744800465,"original_data":{"TOTAL":0,"L1":57,"L2":-46,"L3":-11}},"backlight":{"timer":0,"status":"OFF"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}
`

[image]

dingo35, 2025-04-16:
Exactly what I needed, thx!

vdweegen, 2025-04-25:
Meanwhile I've written a script that monitors the MQTT output and when it detects an unsupported mode it triggers a modeswitch to the previous mode. While this (rather dirty) fix seems to at least make sure the nightly modeswitch for SmartEVSE2 to SMART works, I regularly notice that the same device does not charge in SOLAR mode when there is more than enough surpluss (just noticed -14A and Waiting for Solar). Another switch to SOLAR fixes this behaviour.

vdweegen, 2025-04-25:
Accidentally closed the issue....

dingo35, 2025-05-01:
I fixed a modbus communication bug that might or might not cause this; could you test this please:

[77d1880b7cd.zip](https://github.com/user-attachments/files/19997221/77d1880b7cd.zip)

vdweegen, 2025-05-01:
Just flashed it to both SmartEVSEs, will report back.

vdweegen, 2025-05-04:
Haven't seen the bug since updating to this version. Will keep monitoring.

vdweegen, 2025-05-05:
Just had another occurrence of the behaviour.

dingo35, 2025-05-08:
Ok I built a version that prevents change mode to an illegal value, and also added some error messages:
_LOG_A("ERROR: setMode tries to set Mode to %u.\n", NewMode);

and

  _LOG_A("ERROR: mode=%s, Mode=%u, modeId=%d, AccessStatus=%u.\n", mode.c_str(), Mode, modeId, AccessStatus);

Could you please flash this version:

[323268c06.zip](https://github.com/user-attachments/files/20097593/323268c06.zip)

...and run telnet logs on both master and slave; if the error happens, grep for the error messages above and post about 20 lines before the message until the message itself.....

THANKS!

vdweegen, 2025-05-10:
> Ok I built a version that prevents change mode to an illegal value, and also added some error messages: _LOG_A("ERROR: setMode tries to set Mode to %u.\n", NewMode);
> 
> and
> 
> _LOG_A("ERROR: mode=%s, Mode=%u, modeId=%d, AccessStatus=%u.\n", mode.c_str(), Mode, modeId, AccessStatus);
> 
> Could you please flash this version:
> 
> ...and run telnet logs on both master and slave; if the error happens, grep for the error messages above and post about 20 lines before the message until the message itself.....
> 
> THANKS!
> 
> EDIT: this version should be even better:
> 
> [4b36d0969f.zip](https://github.com/user-attachments/files/20098603/4b36d0969f.zip)

Thanks for the quick reply. I'm out of the country until the end of May, will test this version and collect the logs upon returning.

dingo35, 2025-06-02:
Could you please test v3.8.2 ?

vdweegen, 2025-06-04:
Just flashed v3.8.3, will let you know how it goes.

vdweegen, 2025-06-05:
v3.8.3

dingo35, 2025-07-09:
Closed due to inactivity.

closed 2025-07-09 by dingo35