dingo35/SmartEVSE-3.5 #205: Enhanced cable locking option (solution attached)

jimflbar, 2025-04-26:
**Is your feature request related to a problem? Please describe.**
Currently, the functionality of automatically locking the cable upon insertion into the EV (as described since v3.6.1) was not active in all charging states (it did not work in STATE A). Additionaly, there was no functionality to keep the charging cable always locked in the actuator except in status OFF. This last thing is something that users with semi-permanently attached charging cables could use.

**Describe the solution you'd like**
Based on v3.8.0, I have implemented a new configuration option called "CABLELOCK" in the menu. When this option is Enabled, disconnecting the charging cable from the charging station is restricted to the OFF mode and is only possible after the cable has been removed from the EV first. In all other modes (including PAUSE), the cable remains locked in the charging station.

If the CABLELOCK option is changed to Disabled in the menu, while the cable is still connected to both the EV and the charging station (in PAUSE or OFF), the user should be able to disconnect the cable directly from the charging station without the need to unplug from the EV first.

Furthermore, it is now intended that locking the charging cable in the charging station as soon as it is also plugged into the EV works consistently, regardless of whether CABLELOCK is Enabled or Disabled. This corrects the behavior where this functionality was not active in STATE A.

A possible additional improvement could be a lock symbol somewhere on screen when the CABLELOCK option has been activated. This is not part of the solution provided in the attached diff file collection.

**Describe alternatives you've considered**
Alternative solutions considered include:
Context-dependent locking that automatically unlocks in certain situations (e.g., after a certain time in PAUSE). However, this could be more complex to implement and potentially less predictable for the user.
Displaying a notification or warning when the user attempts to disconnect the cable in a locked mode. This does not solve the fundamental issue of the unwanted locking.
The chosen solution with an explicit CABLELOCK option gives the user direct control over this behavior and makes the system more predictable.

**Additional context**
This adjustment has been tested and works correctly in my setup. It also provides a practical solution for users who want to leave a charging cable semi-permanently attached to their charging station with a socket, allowing them to control when the cable can be directly disconnected. The implementation also corrects an inconsistency with the previously described functionality of automatic cable locking (since v3.6.1) by ensuring it now also works in STATE A.

[CableLock.zip](https://github.com/user-attachments/files/19922240/CableLock.zip)

dingo35, 2025-04-27:
Appreciate your work, thanks for that! Some remarks:
1. State A means "EV Disconnected"; in most use cases people who disconnect their EV also want to disconnect the cable to take it with them. So State A disconnects the cable on the EVSE side by design. 
2. By setting this variable you are sort of turning a socket-cable into a fixed cable; so either something went wrong in the design process (why not put in a fixed cable in at the first place?), or something changed (an opinion, or somebody only wants to disconnect the cable EVSE side on weekdays, or whatever). There are a few users who asked for this feature, asking to make it dynamically switchable by the API's, because of their "dynamic" use cases.
3. We try to confine the LCD menu for "static" parameters that are not/hardly changing, e.g. MaxMains and MainsMeter etc.; more dynamic parameters go into the dashboard.
4. So I propose to include this feature, BUT get it out of the LCD menu and move it into the dashboard; next to LCD_LOCK would be a logical place, we could rename "LCD:" to "Locks:" and add a cable-lock checkbox, that would only appear if there is a locking device enabled (solenoid or motor) in the LCD menu.
5. That way it would already be into the REST API, I would suggest to also add it to the MQTT API to keep things consistent (and I mean not the current "read only" topic, but move it to some /Set topic....
6. You made some poor choices by diffing your changes file-by-file from .cpp.orig to .cpp files (that forces me to setup a special directory with orig files and patch file by file). It would have been easier (both for you and for me) to git clone the current master to /tmp, and then do a simple diff -Naur /tmp/SmartEVSE-3.5 ...path...to...your...changed....dir/SmartEVSE-3.5 >changes.diff .  But preferable is to make a PULL REQUEST (PR); fork my repo to your own account, make the changes in there, and it automatically will ask you if you want to make a PR in my repo; that PR is then supported by a great review process, also makes it possible for me to accept your commits with a press of a button, and also keeps your name attached to these commits so the credit is yours :-) .

Hope I didn't discourage you, great work, hope you will join us in our ever-improving community!

jimflbar, 2025-04-27:
I appreciate your feedback. Let me respond to the points raised:

1. I understand now that State A needs to remain “EV disconnected” and whenever an EV is connected, the State needs to change to B, “EV connected”.

The problem I had, which triggered me to dive into the code is that in my situation, the EVSE never switches from state A to state B (and locking the cable) just by plugging in the cable in both the EV and the EVSE, unless I activate charging mode (it then goes from State A to B and then to C).

My setup: I only have one EVSE v3 with a mains API. The AccessStatus I start with is either OFF or PAUSE.

I see in the current code that switching from State A to State B requires AccessStatus == ON (and of course pilot == PILOT_9V). So to my understanding, only when a charge mode is selected (e.g. NORMAL, SMART or SOLAR), the EVSE will lock the cable in the EVSE, because only then the AccessStatus is ON. My code works around this problem, but I understand that this is not how it is supposed to work. 

Maybe adjusting the PILOT_9V related line 2647 in main.cpp to at least also include AccessStatus PAUSE would fix my initial problem. However, skipping the AccessStatus check completely in this line would I think be in line with the functionality described in v3.6.1 (e.g. always lock the cable when plugged into an EV). 

Can you please check this observation and advice?

2. Indeed for some people, this would be a design issue. I had to deal with the wife approval factor in creating a clean looking charge pole, thus without fixed cable. Also, I did not know at the time that my car just releases its lock on the cable when it has finished charging. That makes it possible to steal the cable, because unplugging it from the car will release the cable from the EVSE as well. That was my second reason for working on this enhancement. I wanted an option to keep the cable locked in the EVSE, unless I switch to AccessStatus OFF.

3. Clear!

4. I will rework the code.

5. Even better!

6. Sorry for that. I never really worked with github before. I will do as suggested, so no need for you to manually add my lines of code. Happy to learn.

dingo35, 2025-04-28:
Ad 1.:
You are right, I think an improvement would be to change line 2648 from "AccessStatus == ON" to "AccessStatus != OFF"; the PAUSE status should be closer to ON then to OFF....

I don't want to mess in your development/testing process so I will leave this change to you?

Ad 2: Ok, the WAF I understand :-)

Ad 6: No worries; I started out just like you; and here we are, almost a 1000 commits later on the original code :-) .

jimflbar, 2025-04-28:
Can you invite me as a collaborator in your project in github, so I can create a PR? I got the error "must be a collaborator" when trying to create a PR.

hmmbob, 2025-05-02:
> 2. (...) Also, I did not know at the time that my car just releases its lock on the cable when it has finished charging. That makes it possible to steal the cable, because unplugging it from the car will release the cable from the EVSE as well.

Just jumping in on this comment not related to the code: this behavior is usually a setting in your car itself. You might want to verify the manual and/or car onboard menu.

jimflbar, 2025-05-02:
> Just jumping in on this comment not related to the code: this behavior is usually a setting in your car itself. You might want to verify the manual and/or car onboard menu.

Usually yes, but unfortunately I have no setting available for this in my car.

jimflbar, 2025-05-02:
The CableLock solution is ready.

Please note: I did include this change: "Ad 1.: You are right, I think an improvement would be to change line 2648 from "AccessStatus == ON" to "AccessStatus != OFF"; the PAUSE status should be closer to ON then to OFF....", as this change resulted in some unexpected behavior. I am still testing this.

closed 2025-05-08 by jimflbar