dingo35/SmartEVSE-3.5 #207: solar_stop_time value ignored in Solar mode

zulufoxtrot, 2025-05-01:
**Describe the bug**

No matter what I set solar_stop_time to, when necessary current becomes unavailable, the EVSE stops charging almost immediately (about 20 seconds).

**Upload your config**
```json
{"version":"10:28:51 @May  1 2025","serialnr":7581,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Airbox","rssi":-40,"bssid":"10:7C:61:2A:4B:50"},"evse":{"temp":37,"temp_max":65,"connected":true,"access":1,"mode":2,"loadbl":0,"pwm":144,"custombutton":false,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":85,"override_current":0,"current_min":6,"current_max":16,"current_main":30,"current_max_circuit":16,"current_max_sum_mains":26,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":0,"solar_stop_time":2,"enable_C2":"Not present","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{"host":"gap.local","port":1883,"topic_prefix":"smart-evse","username":"mqtt","password_set":true,"status":"Connected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"API","address":12,"import_active_power":1.700000048,"total_kwh":1750.400024,"charged_kwh":2.200000048,"currents":{"TOTAL":74,"L1":74,"L2":0,"L3":0},"import_active_energy":1750.400024,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0,"last_data_update":1746093660,"original_data":{"TOTAL":0,"L1":0,"L2":0,"L3":0}},"backlight":{"timer":120,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}
```
**Upload a debug log**
```
(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1746093424.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=8.4 A, Idifference=8.6, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 3 Isetbalanced=8.4 A, IsumImport=0.1, Isum=0.1, ImportCurrent=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=8.4 A.
(D) (CalcBalancedCurrent)(C1) Checkpoint b: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=8.4A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C1) [H]Node 0 = 8.4 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=8.4 A.
(I) (write_settings)(C1) settings saved
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.1 A IsetBalanced: 8.4 A, MainsMeter.Timeout=11, EVMeter.Timeout=11.
(I) (loop)(C1) L1: 0.1 A L2: 0.0 A L3: 0.0 A Isum: 0.1 A
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.1 A IsetBalanced: 8.4 A, MainsMeter.Timeout=10, EVMeter.Timeout=10.
(I) (loop)(C1) L1: 0.1 A L2: 0.0 A L3: 0.0 A Isum: 0.1 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=8.4 A Imeasured=0.1 A MaxCircuit=16 Imeasured_EV=7.4 A, Battery Current = 0.0 A, mode=2.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=8.4 A, Idifference=8.6, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 3 Isetbalanced=8.4 A, IsumImport=0.1, Isum=0.1, ImportCurrent=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=8.4 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=8.4 A.
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.1 A IsetBalanced: 8.4 A, MainsMeter.Timeout=9, EVMeter.Timeout=9.
(I) (loop)(C1) L1: 0.1 A L2: 0.0 A L3: 0.0 A Isum: 0.1 A
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.1 A IsetBalanced: 8.4 A, MainsMeter.Timeout=11, EVMeter.Timeout=8.
(I) (loop)(C1) L1: 0.1 A L2: 0.0 A L3: 0.0 A Isum: 0.1 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=8.4 A Imeasured=0.1 A MaxCircuit=16 Imeasured_EV=7.4 A, Battery Current = 0.0 A, mode=2.
(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1746093428.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=8.4 A, Idifference=8.6, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 3 Isetbalanced=8.4 A, IsumImport=0.1, Isum=0.1, ImportCurrent=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=8.4 A.
(D) (CalcBalancedCurrent)(C1) Checkpoint b: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=8.4A, ActiveEVSE=1.
(V) (CalcBalancedCurrent)(C1) [H]Node 0 = 8.4 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=8.4 A.
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.1 A IsetBalanced: 8.4 A, MainsMeter.Timeout=10, EVMeter.Timeout=63.
(I) (loop)(C1) L1: 0.1 A L2: 0.0 A L3: 0.0 A Isum: 0.1 A
(I) (write_settings)(C1) settings saved
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.1 A IsetBalanced: 8.4 A, MainsMeter.Timeout=11, EVMeter.Timeout=11.
(I) (loop)(C1) L1: 0.1 A L2: 0.0 A L3: 0.0 A Isum: 0.1 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=8.4 A Imeasured=0.1 A MaxCircuit=16 Imeasured_EV=7.4 A, Battery Current = 0.0 A, mode=2.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=8.4 A, Idifference=8.6, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 3 Isetbalanced=8.4 A, IsumImport=0.1, Isum=0.1, ImportCurrent=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=8.4 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=8.4 A.
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.1 A IsetBalanced: 8.4 A, MainsMeter.Timeout=10, EVMeter.Timeout=10.
(I) (loop)(C1) L1: 0.1 A L2: 0.0 A L3: 0.0 A Isum: 0.1 A
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 8.2 A IsetBalanced: 8.4 A, MainsMeter.Timeout=11, EVMeter.Timeout=64.
(I) (loop)(C1) L1: 8.2 A L2: 0.0 A L3: 0.0 A Isum: 8.2 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=8.4 A Imeasured=8.2 A MaxCircuit=16 Imeasured_EV=7.4 A, Battery Current = 0.0 A, mode=2.
(V) (CalcBalancedCurrent)(C1) Current is limited by MaxSumMains: MaxSumMains=26A, Isum=8.2A, Temp_Phases=3.
(V) (CalcBalancedCurrent)(C1) phaseLastUpdate=1746093432.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=8.4 A, Idifference=5.9, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 3 Isetbalanced=4.3 A, IsumImport=8.2, Isum=8.2, ImportCurrent=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=4.3 A.
(I) (CalcBalancedCurrent)(C1) No Current!!
(V) (CalcBalancedCurrent)(C1) [H]Node 0 = 6.0 A.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=6.0 A.
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 120 NoCurrent: 1 Imeasured: 8.2 A IsetBalanced: 6.0 A, MainsMeter.Timeout=10, EVMeter.Timeout=63.
(I) (loop)(C1) L1: 8.2 A L2: 0.0 A L3: 0.0 A Isum: 8.2 A
(I) (loop)(C1) STATE: C Error: 0 StartCurrent: -0 ChargeDelay: 0 SolarStopTimer: 119 NoCurrent: 1 Imeasured: 8.2 A IsetBalanced: 6.0 A, MainsMeter.Timeout=9, EVMeter.Timeout=62.
(I) (loop)(C1) L1: 8.2 A L2: 0.0 A L3: 0.0 A Isum: 8.2 A
(V) (CalcBalancedCurrent)(C1) Checkpoint 1 Isetbalanced=6.0 A Imeasured=8.2 A MaxCircuit=16 Imeasured_EV=7.4 A, Battery Current = 0.0 A, mode=2.
(V) (CalcBalancedCurrent)(C1) Current is limited by MaxSumMains: MaxSumMains=26A, Isum=8.2A, Temp_Phases=3.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=6.0 A, Idifference=5.9, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 3 Isetbalanced=6.0 A, IsumImport=8.2, Isum=8.2, ImportCurrent=0.
[... 87 lines cut ...]
(setState)(C1) Switching Contactor1 OFF.
(setState)(C1) Switching Contactor2 OFF.
(I) (loop)(C1) STATE: B1 Error: 1 StartCurrent: -0 ChargeDelay: 60 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.7 A IsetBalanced: 3.1 A, MainsMeter.Timeout=11, EVMeter.Timeout=64.
(I) (loop)(C1) L1: 0.7 A L2: 0.0 A L3: 0.0 A Isum: 0.7 A
(D) (IsCurrentAvailable)(C1) No current available StartCurrent line 1006. ActiveEVSE=0, TotalCurrent=0.0A, StartCurrent=0A, Isum=0.7A, ImportCurrent=0A.
(setState)(C1) Switching Contactor1 OFF.
(setState)(C1) Switching Contactor2 OFF.
```

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar?

Solar

3. What steps do you take so the problem arises?

1. Set mode to Solar
2. Set SolarStopTime to at least 1 minute (but also works with 10 minutes)
3. Start charging
4. Turn a load on in the home / provide a high MQTT value for home power draw

**Expected behavior**

The EVSE keeps charging for the duration defined in solar_stop_time.

dingo35, 2025-05-01:
Why is your MaxSumMains set to 26A? Are you on a capacity rate provider?
Are yoy charging 1phase or 3phase?

zulufoxtrot, 2025-05-01:
I was not sure what to set it to, and assumed it only mattered for power share, so the value is more or less arbitrary.

1phase.

dingo35, 2025-05-01:
-disable maxsummains
-set C2 to "Always_Off" when charging 1 phase, as described in the manual

zulufoxtrot, 2025-05-01:
It solved my issue. Thank you.

closed 2025-05-01 by zulufoxtrot