dingo35/SmartEVSE-3.5 #208: Solar mode jumps to maxcurrent instead of 6A

Imaginous, 2025-05-01:
**Describe the bug**
There was not enough solar current available, instead of the timeout and charging at 6A, it charged at 27A (max current).

**Upload your config**
`{"version":"08:10:16 @Apr 14 2025","serialnr":6972,"mode":"SMART","mode_id":3,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"Imaginous","rssi":-55,"bssid":"7C:10:C9:CD:FE:A8"},"evse":{"temp":39,"temp_max":55,"connected":true,"access":1,"mode":1,"loadbl":0,"pwm":460,"custombutton":false,"solar_stop_timer":0,"state":"Connected to EV","state_id":1,"error":"None","error_id":0,"rfid":"Not Installed"},"settings":{"charge_current":270,"override_current":0,"current_min":6,"current_max":27,"current_main":33,"current_max_circuit":32,"current_max_sum_mains":600,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":7,"solar_stop_time":31,"enable_C2":"Always Off","mains_meter":"HmWzrd P1","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0},"mqtt":{"host":"mqtt.imaginous.nl","port":1883,"topic_prefix":"SmartEVSE","username":"mqtt","password_set":true,"status":"Connected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0,"host":"p1meter-04636A.local"},"phase_currents":{"TOTAL":-41,"L1":-41,"L2":0,"L3":0,"last_data_update":1746115787,"original_data":{"TOTAL":-41,"L1":-41,"L2":0,"L3":0}},"backlight":{"timer":56,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}`

**Upload a debug log**
No debug log when it happened unfortunately.

**To Reproduce**
Solar charge, wait until not enough solar power is available.
It is the first time it happened with this firmware. I charged 2 times before and it worked correctly.

**Expected behavior**
To charge at 6A until the timeout has expired.
**Screenshots**

[image]

[image]

dingo35, 2025-05-01:
Interesting! I need a verbose debug log so I can trace which line of code decided there is enough current available; could you run a telnet session and try to catch this error again?

Imaginous, 2025-05-03:
Build the latest version with debug on.

Charged two times, but it did not happen yet.

Keep you posted.

dingo35, 2025-05-08:
Probably related to #215 Homewizard P1

Imaginous, 2025-05-08:
I will do a pull and fresh debug build tonight.
Keep you posted.

dingo35, 2025-06-23:
Thx for your feedback!