dingo35/SmartEVSE-3.5 #211: Due to bugs in 3.8.0 I'm back to 3.7.4

PhilippeRigaux, 2025-05-06:
**Describe the bug**
Most anoying one is some king of timeout on my ID.4 so when I'm on SOLAR mode, when it pause for the night, the day after I have to unplug and replug the car in the morning for the charge to resume ( is not the case with 3.7.4) Second problem is that the Charge speed in the web interface of the master EVSE is not correct stay at one fixed value can be 6A or anything else when the slave is charging ( I think it is the last value the MASTER EVSE charge )
**Upload your config**
Not related to my config as it works with old versions for months or even years..
**Upload a debug log**
Problème is always there.
**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar? SOLAR
2. What steps do you take so the problem arises? Try to charge

**Expected behavior**
Auto resume charge in the morning when there is enough solar power
**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

dingo35, 2025-05-06:
Could you upload a debug log? That shows us (hopefully) what's going on when your ID4 is charging..

PhilippeRigaux, 2025-05-06:
I flashed the 2 EVSE with 3.8.0 debug, will try to get a debug log of both soon..

PhilippeRigaux, 2025-05-06:
When I want to be sure that no charge occur, I send a discharge of 100A from the Solar battery. When. the Evoke want to charge on Master EVSE, THE SLAVE mode is changed to OFF and go back to SOLAR once the Evoke stop to charge, at that time the ID.4 is allowed to charge.

Here are some debug logs in RTF to have the colors and I added a timestamp on each lines.

[Master during the night (no power available)](https://drive.proton.me/urls/FM7TJEVNFG#mcVeNMi0gJSZ)
[Slave during the night (no power available)](https://drive.proton.me/urls/J88FSN3F1G#o4r2qpARyw29)

Tomorrow I will be able to get the logs when the sun shines again..

dingo35, 2025-05-07:
Ok I just solved a bug that solves a communication problem between master and slave, which might be the cause of your problems. The fix is incorporated in v3.8.1 , just released.

Could you please flash the v3.8.1 debug version and test + log for errors ?

Thank you !!

PhilippeRigaux, 2025-05-07:
OK, just flashed the 3.8.1-debug, the charge did not start automatically this morning, I will do the same as yesterday, charge 1% later today, let it wait for the night and see what's going on tomorrow morning..
Here are the logs, charge should start on the slave..
[Master 3.8.1, has power but charge don't start](https://drive.proton.me/urls/NPDNYPZJ2G#XTaCjOLTEQHK)
[Slave 3.8.1, has power but charge don't start](https://drive.proton.me/urls/W3KST4WVAG#0FxwMCRmrpb2)

Thx

PhilippeRigaux, 2025-05-07:
I simulated plenty of power to start a charge, then let them know that there is no power to stop the charge and let the capture run to get the waiting traces too..
[Master](https://drive.proton.me/urls/R01JQJTSE8#2lgO9PKscKQU)
[Slave](https://drive.proton.me/urls/2G2NDZ96K4#jPBKEjKzvPBG)

dingo35, 2025-05-08:
Could you test this version please:

[4b36d0969f.zip](https://github.com/user-attachments/files/20098585/4b36d0969f.zip)

PhilippeRigaux, 2025-05-08:
OK, will test it overnight, but the charge did correctly start this morning with 3.8.1...

PhilippeRigaux, 2025-05-08:
Test version waiting for power during the night:
[Master](https://drive.proton.me/urls/SMWTK2ZJR0#SK2yK6kiaF4P)
[Slave](https://drive.proton.me/urls/BYQPH78YEW#aouHa7cldfJl)

PhilippeRigaux, 2025-05-09:
OK, the charge did resume correctly this morning ( but we don't have a lot of sun at the moment ) and the charge amp did show correctly, so from my point of view the dev firmware fix both of my issues

[Master](https://drive.proton.me/urls/MR8Q226YD4#8p74RSyfxUXD)
[Slave](https://drive.proton.me/urls/4563Z2EVT8#X8ukrwPBLowZ)

Thx a lot

closed 2025-05-09 by dingo35

PhilippeRigaux, 2025-05-12:
Well I have had to replug this morning...