dingo35/SmartEVSE-3.5 #218: ID.4 still don't resume charges in the morning

PhilippeRigaux, 2025-05-13:
**Describe the bug**
This is a followup for the other bug report I opened about one week ago, charges on the ID.4 don't resume, this problem is only with the slave, on the master charges resume correctly or with the pre 3.8.0 it worked perfectly.
After 2 days of correctly ( simulated usage ) function, and after the first real car usage and recharge, it don’t want to resume the charge on the morning. So I unplugged and it charged currently.
Today after a second normal usage, use the car yesterday afternoon, plug end of yesterday afternoon, pause during the night, it didn’t resume today morning. So it seems to need more work on it..
**Upload your config**
Press the "raw" button in the SmartEVSE webserver and upload the json settings file, so we can look at your configuration.

**Upload a debug log**
Master: https://drive.proton.me/urls/9AA7VFDWNR#WkxMs46xnxz5
Slave: https://drive.proton.me/urls/BEBKYJ0JBM#rfqJFpvDDa7a

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Solar
2. What steps do you take so the problem arises?

**Expected behavior**
Resume charge when there is enough solar power in the morning.
**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

dingo35, 2025-05-13:
Could you please test this version:

[3bc248fa.zip](https://github.com/user-attachments/files/20187029/3bc248fa.zip)

PhilippeRigaux, 2025-05-13:
OK, both are flashed, will report on progress but only on real usage because the simulated don't seems to reproduce the problem..

PhilippeRigaux, 2025-05-15:
Well I used the car yesterday, the charge started yesterday end of afternoon, paused for the night but did not restart this morning. I have to open the car ( just unlocking with the remote is enough) and then the charge start..

[Master](https://drive.proton.me/urls/8MVYX04E9M#apeQ8e7ZO6Pe)
[Slave](https://drive.proton.me/urls/59XN48CYHG#o8rcX3ZIvinf)

Thx

dingo35, 2025-05-15:
OK, this is strange, I thought I found your bug, fixed it, it doesnt happen on my testbench anymore.

First of all, could you check if your devices are actually running the correct test version? I noticed that in rare cases the flash seems to go ok but the device boots on the previous flash version ...

So your test devices should show on the web dashboard:
Version: 13:35:10 @May 13 2025

If your tests/logs were done with THIS version, and the problem STILL persists, the following is the case:

Your Slave log just shows the EV is connected (and detected) by the SmartEVSE correctly; the reason that it is not charging is because the ID.4 is not asking for current, for some reason.
So this might be caused by an OTA software update of the ID.4, or by a bug of the SmartEVSE, so that it left the ID.4 in this non-asking state. The logs don't show that.

The problem is that neither you, nor I can reproduce the problem in a test situation.

The only way I can think of to localize this problem is the following:
-generate a list of commits from v3.7.4 to v3.8.0, top of the list is v3.8.0, bottom of the list is v3.7.4  (I can do that for you)
-check if v3.7.4 still works fine (this rules out ID.4 software update) (you would have to do that)
-if so, then pick a commit halfway of the list, compile it, flash it, and test it (you would have to do that)
-if the problem persists, pick a commit lower on the list; if it is solved, pick a commit higher on the list.
This way we could trace which commit is the culprit, and then solve the problem.

Now between those versions the list would be 292 commits, if you pick your commits halfway it would take you approx. 9 times to localize the problem.

Are you willing and able to do that?

PhilippeRigaux, 2025-05-15:
As the problem came with 3.8.0, is this list of commits right?
$ git log --after "2025-03-30" --before "2025-04-06" --pretty="%h - %s"
16155ad - esp32.cpp, network.cpp: Increase accuracy of Homewizard P1 to 1 decimal
f2f8b03 - network.h, network.cpp: exclude Homewizard logic from sensorbox
ae44284 - esp32.cpp, network.cpp, network.h: make Homewizard P1 interface work with single phase mainsmeters
ba1829f - tests.sh: fix to http
6766c25 - main.cpp: fix SET_ON_RECEIVE for non-int datatypes
b1ece8c - index.html: remove unused id
bff2030 - esp32.cpp: write_settings after all REST API POST settings
d0f1e3e - introduce Pause AccessStatus
1bf16d2 - platformio.ini: upgrade espressif 6.8.1 to 6.10.0
d6170e8 - esp32.cpp, glcd.cpp: remove Melopero RTC code
9c90600 - network.cpp, esp32.cpp: change 'reboot after EV disconnect' to 'reboot after EV stopped charging'
2e8e5fc - HowToFlash.txt: update doc
6013637 - pio-build.yaml: fix workflow v4 so now bin files are in root of .zip file
a57e649 - update docs. List supported kWh meters
f5c8eff - glcd.cpp, main.cpp: only LCDlock will disable LCD menu
8b13cef - Modbus: Wait for response before sending new request. Enforce 50ms quiet time between messages, this improves reading slow modbus meters.
477e384 - main.cpp: timeout on Node if no Mainsmeter is defined
5f42414 - Update configuration.md (#79)
7002b9f - workflow to v4
6dde9f6 - Update url in building_flashing.md
6190ac2 - Update configuration.md (#170)
12ff689 - docs: add warning for Energy Efficient contactors
0a4789a - main.cpp: make sure in Normal Mode, PwrShare enabled, MaxMains is NOT obeyed

dingo35, 2025-05-15:
Great that you are willing to do this, I understand you want to be able to be updating your firmware in the future, and I like to solve bugs :-) .

No your list is a bit too small; v3.7.4 is the same as v3.7.3 with a few backport commits, so you will have to go back to v3.7.3 (not much difference with v3.7.4), I attached the log file with all the commits from v3.8.0 back to v3.7.3 in timely order:

[commits.zip](https://github.com/user-attachments/files/20227394/commits.zip)

PhilippeRigaux, 2025-05-15:
OK, will look at that but as I can't simulate the problem, it can take time. I don't think the problem will be big to solve, I have had the same problem with Wallbox Pulsar that I used before Smatevse.. They released a firmware update that broke the ID.4 and several weeks later a new update of the Pulsar fixed it..

PhilippeRigaux, 2025-05-19:
OK, so I confirmed 3.7.4 working perfectly..
Build a firmware based on commit:
ed754bb - main.cpp: fix compiler warning

commit ddc50f184188605800e1591412a704dc549085e6
Author: dingo35 <spaarder@hotmail.nl>
Date:   Sun Mar 16 11:33:55 2025 -0400

    main.cpp: fix compiler warning

PhilippeRigaux, 2025-05-22:
I confirm that eb754bb - main.cpp: fix compiler warning resume the charge correctly.

I built:
0018e15 - glcd.cpp: workaround to display LCD RFID menu topline messages for v4

PhilippeRigaux, 2025-05-24:
ed754bb didn't resume this morning..

So built and installed:
f5eb58c - esp32.cpp: eliminate CommTimeout

PhilippeRigaux, 2025-05-25:
f5eb58c didn't resume this morning

Built and installed:
516fc14 - esp32.cpp, main.cpp: mark problem areas where ErrorFlags is set in v4 ESP32 code

dingo35, 2025-05-26:
Ok you're closing in, you reduced it to only 20 commits!

PhilippeRigaux, 2025-05-26:
Great! So I flashed on both nodes:

Version: 15:28:28 @May 26 2025

PhilippeRigaux, 2025-05-28:
8a45af4 worked flawlessly 

Built and installed:
9be0317 - main.cpp: remove duplicate code that clears flags

PhilippeRigaux, 2025-05-29:
9be0317 Didn't resume this morning

Built and installed:
b92cd6f - unionize ErrorFlags

dingo35, 2025-05-30:
No I expect 79ba to resume ok, and the culprit is in 9be031.

PhilippeRigaux, 2025-05-30:
OK, will install and test it, It think I will be able to give an answer on Sunday..

PhilippeRigaux, 2025-05-30:
Humm are you sure of the firmware file you sent?

Smart EVSE V3
Version: 15:28:28 @May 26 2025

Not from today?

dingo35, 2025-06-01:
Ok the fix will be in the next release, thx for your efforts!