dingo35/SmartEVSE-3.5 #225: Feature request : (for eGolf) current override in solar mode

royspeed, 2025-05-20:
Solar mode: overshoot to 32 A & long renegotiation loop above 16 A; no way to cap at 16 A

---

**Description:**  
In **Solar** mode, charging follows available PV power smoothly up to 16 A. When the target current rises just above 16 A, the EVSE’s pilot duty quickly goes up to 100 % (32 A), the car finally negotiates to draw > 16 A—but only after several seconds/minutes. As soon as it does, the EVSE’s solar-driven limit immediately drops back to match the new draw. When de current drops below 16 A  cycle repeats.

**Environment:**  
- Firmware version: e.g. **v3.8.0**  
- Hardware: SmartEVSE controller + Modbus kWh meter and sensorbox  
- EV: VW e-Golf 2020 (dual 16 A onboard chargers)  
- Mode: Solar mode (sensor-driven charging), no manual override in this mode

**Steps to reproduce:**  
1. Set SmartEVSE to **Solar** mode with a sensorbox or Modbus meter.  
2. Ensure PV output ramps target current from 0 A past 16 A.  
3. Watch charging:  
   - Up to 16 A, charging is smooth.  
   - As soon as the pilot target exceeds 16 A, the EVSE  increases to 32 A.  
   - The car takes several minutes to switch its second 16 A module on.  
   - Once the car draws > 16 A, the EVSE’s solar limit immediately falls back.  
   - When charging drops below 16 A and the loop starts over.

**Expected behavior:**  
- A user-configurable **max current cap** (e.g. 16 A) should be available in Solar mode—just like the existing override in Smart mode—to prevent overshoot and renegotiation loops.

**Actual behavior:**  
- Solar mode does **not** expose any max-current override.  
- Every time PV-driven target crosses 16 A, EVSE overshoots to 32 A, the car renegotiates slowly, then EVSE pulls back below 16 A, causing a repeating stall.

**Workaround:**  
Switch to **Smart** mode to set an override cap, but lose PV-based dynamic control.

**Proposal / Solution:**  
Add a **max-current** parameter to **Solar** mode (identical in function to the Smart-mode override). This would allow users to:  
1. Prevent EVSE from overshooting above their chosen limit (e.g. 16 A).  
2. Maintain smooth PV-based charging without multi-second renegotiation loops.

---

*Happy to share logs, screenshots, or parameter settings if needed!*

dingo35, 2025-05-20:
By the way could you upgrade to v3.8.1? No use to debug old versions..

royspeed, 2025-05-20:
Have a 1 phase 35A connection. 
The car can charge on 1 or 2 phase. With 1 phase 1x32A and with 2 phase 2x16A. 
From 6A to 16 A everything goes well. 
I have 4500W of peak power from the solar panels. As soon as more than 16A is available, the car needs a long time before more than 16A is charged. 
The smart evse does not notice that the car is stuck at 16A and does see that there is export via the sensor box. The maximum charging current then slowly goes to 32A. There is nothing wrong with the evse or with the firmware, it is a limitation of the car that takes a few minutes and briefly stops charging if more than 16A is charged. This probably does not happen on 2 phases either, but I can't check that.
A nice solution would be a limitation on 16A.

[image]
[image]

PhilippeRigaux, 2025-06-01:
Sometimes I see that too, the car decide to not charge faster than about 16A but SmartEVSE continue to increase allowed charge speed and go to 30A while the car ( EV meter ) stay at about 16A.

Perhaps the solution is to look at the EVmeter readout and don't change the allowed value if the difference between EV meter and allowed power usage is more then 2A?

It would be an elegant way to fix the latency between the EVSE and the car.

dingo35, 2025-06-01:
That is a very interesting idea, let me sleep a night or two over that!

PhilippeRigaux, 2025-06-27:
Hi Dingo,

Was the sleep effective? ;)

dingo35, 2025-06-27:
Not really; the current logic is independent of an EVmeter being present, and I would like to keep it that way; flash ram is valuable, with upcoming changes we're around 99% so I dont want to waste it on this very specific use case.

BUT there is a big change coming up that will change the handout of Current completely, so that might or might not change things ...