dingo35/SmartEVSE-3.5 #229: webui exposes tls private key

HA-TB303, 2025-05-30:
**Describe the bug**
Not sure if this is considered a bug, more of a security issue maybe...

http://[IP]/key.pem shows:

```
-----BEGIN EC PRIVATE KEY-----
MHcC1234ICtA0BJEaHGDHGJFDJGHOA5I7qqm4W/CoFbJSP/osM3poAoGCCqGSM49
AwEHoUQDQg1234SIMmJ+CfEhNy0QVYZA8iwE+nLphi+90Bgfj1234gdJB
rI6b1K7HJcJglS1a12340eD8Pxct6oLw==
-----END EC PRIVATE KEY-----

**Upload your config**
N/A

**Upload a debug log**
N/A

**To Reproduce**
Open in browser: http://[IP]/key.pem

**Expected behavior**
Private key should be private?

**Screenshots**
N/A

dingo35, 2025-05-30:
Thanks for your report!

Im no security expert, but I dont think this will do any harm:
Its a self signed key that anybody could generate; it enables the use of https, but the real security cannot be warranted because a) it is a self signed key b) that therefore cannot be checked against a certificate authority.

AFAIK ts not possible to generate / buy certificates (e.g. lets-encrypt) because there is only a local mDNS url, no global one.

HA-TB303, 2025-05-30:
check, thanks for looking into it.