dingo35/SmartEVSE-3.5 #235: Modbus errors

rhannink, 2025-06-08:
I have 2 smartevse’s both with a Eastron SDM-72 kWh meter. All connected with eachother through the RS-485 bus. kWh meters report to smartevse. Master slave does also work, but I get rs485 bus error (ev meter not communicating en red flashing butto).

Question, does it matter how I build the bus? Should the smartevse’s be at the ends or the kWh meters. Do the smartevse’s terminate with 120 ohm or should I add resistors?

Right now the bus is kWh meter 1 - smartevse’s 1 - smartevse 2 - kWh meter 2. I use cat6 cable (one twisted pair and gnd)

Gr. Remco

dingo35, 2025-06-19:
Which software version are you on?
Are you using twisted pair?
Do you have the SDM72DM-v2 version that supports all registers (currents, power, energy)?

dingo35, 2025-06-19:
Then the only thing I can think of is: are your meter addresses configured correctly? No hex-decimal conversion mistake there? 

Since you are getting EV meter error, try disconnecting the EV meter, and see if the MainsMeter is working correctly (check your web-dashboard).
If that works, switch your EV Meter with your MainsMeter; hopefully you can localize your problem then...

Another last thing: in the past I have read reports of users who had problems on addresses 0x0b, 0x0c (decimal 11, 12), and they were succesfull on addresses 100, 101 (decimal) ....

rhannink, 2025-06-19:
I have the Mains meter connected via web API from esphome P1 reader.
I only have the modbus EV meter error few times per week, for the rest (after reboot) it works fine.
I use adress 12 (dec) for EV meter 1 and 13 (dec) for EV meter 2 and I have the two Smartevse's interconnected with RS485 modbus, although I am not running them in master/Slave mode, but standalone.

Ik will try to change the modbus addresses and I will remove the RS485 interconnect bewteen the two SmartEVSE's.

Gr. Remco

closed 2025-07-21 by dingo35