dingo35/SmartEVSE-3.5 #239: SetChargingProfile command for OCPP not supported

Apflkuacha, 2025-06-16:
**Describe the bug**
With my OCPP backend I do get a WebSocket disconnect from SmartEVSE 20 seconds after setting the charging profile. The OCPP backend works with several stations from different manufacturers. 
The charging profile is used to set the charging current and pause charging by setting the current to zero. 

**debug log**
if i do send following OCPP message to SmartEVSE I do get a TimeoutException. And 20 seconds later the connection closes
    11:52:25 Send message: [2,"07258e03-3fb0-43e0-9b88-72f5aa4e643a","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":null,"stackLevel":5,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Absolute","recurrencyKind":null,"validFrom":null,"validTo":null,"chargingSchedule":{"duration":null,"startSchedule":null,"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0,"numberPhases":3}],"minChargingRate":null}}]
    11:52:30 TimeoutException: Station did not answer back to "07258e03-3fb0-43e0-9b88-72f5aa4e643a" within 5000ms
    11:52:44 System.Net.WebSockets.WebSocketException (0x80004005): The remote party closed the WebSocket connection without completing the close handshake.

**Expected behavior**
The station should support charging profile that the charging current could be set like on other stations

Apflkuacha, 2025-06-17:
That is currently v3.8.1

dingo35, 2025-06-17:
Why arent you on the most recent version? No use of wasting our time trying to debug old versions ...

On v3.8.3 it works fine here:
[MO] Recv: [2,"f3da7201-6931-4b7f-9e27-92d2fcd5bcef","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":6,"stackLevel":5,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Absolute","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0,"numberPhases":3}]}}}]
[MO] Send: [3,"f3da7201-6931-4b7f-9e27-92d2fcd5bcef",{"status":"Accepted"}]
@MSG: OCPP Smart Charging suspends EVSE

closed 2025-07-09 by dingo35

Apflkuacha, 2025-07-28:
It took a while as I am developing a backend and a costumer uses a EVSE Station. But now a Update on this one: it also didn't work with the latest version 3.9.0
But I changed that the null-Values on JSON are ignored and not written any more on the Json message, then it works with SmartEVSE. Other stations didn't have a problem with this null-values.

So instead of 

    "chargingProfileId":1,"transactionId":null,"stackLevel":5,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Absolute","recurrencyKind":null,"validFrom":null,"validTo":null, ...

it results in 

    "chargingProfileId":1,"stackLevel":5,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Absolute", ...

dingo35, 2025-07-29:
What does it say about the null values?

Apflkuacha, 2025-07-29:
I say that SmartEVSE doesn't work if the null values are provided and it works when they are omitted. So I omit them now.

dingo35, 2025-07-29:
Glad you could solve it, thanks for the info!