dingo35/SmartEVSE-3.5 #240: Add mqtt ssl connection to broker

Hermarcel, 2025-06-23:
I want to configure smartEVSE to communicate via the mqtt-broker of the Victron Cerbo GX evice. But this broker needs an SSL/TLS connection with exchange of the Certificate Authority. As far as I can see, this is not implemented into the current fimware. Would be nice if this was added.

dingo35, 2025-06-23:
We are currently at 99.x % of Flash Ram usage, and I am not sure if this wish is going to make it towards other wishes.

I do not know the device very well, but I dont think it has a MQTT broker built in; usually you need a broker running to connect devices like SmartEVSE and Victron.
That broker you can configure any way you want, with or without SSL, per device.
You probably need to translate the different topics anyway...

dingo35, 2025-06-23:
Understood; I ll leave it open with low prio, perhaps the mongoose code can enable this without extra resources....